dusty-nv/jetson-inference #1707: How to use --resume command with Checkpoint parameter

4boris, 2023-07-27:
I do not know where to find the correct path to use as an argument to the --resume flag. 

I followed the Re-training on the PlantCLEF Dataset instructions and quit (ctrl C) after 1 epoch was completed. I then attempted to resume using:

python3 train.py --model-dir=models/plants data/PlantCLEF_Subset --resume=CHECKPOINT 

but I do not know what the checkpoint path is.

I attempted using       /root/.cache/torch/hub/checkpoints/resnet18-f37072fd.pth        which was given when the classification began in the following line:
Downloading: "https://download.pytorch.org/models/resnet18-f37072fd.pth" to /root/.cache/torch/hub/checkpoints/resnet18-f37072fd.pth 100.0%

I then received the following error message:
File "train.py", line 195, in main
args.start_epoch = checkpoint['epoch'] + 1
KeyError: 'epoch'

4boris, 2023-07-27:
when I do the list command I get the following
data/PlantCLEF_Subset:
labels.txt  test   train   val

models/plants:
chackpoint.pth.tar    labels.txt    model_best.pth.tar    tensorboard

But nothing seems to give me the correct path

dusty-nv, 2023-07-27:
@4boris try using `--resume=models/plants/checkpoint.pth.tar` (for last checkpoint) or `--resume=models/plants/model_best.pth.tar` (for highest-accuracy checkpoint)

4boris, 2023-07-27:
i get the following error message:
File "train.py", line 196, in main
best_accuracy= checkpoint['best_accuracy']
KeyError: 'best_accuracy'

dusty-nv, 2023-07-27:
@4boris sorry about that, this should be fixed in https://github.com/dusty-nv/pytorch-classification/commit/3e9cf8c4003311009539a6c101d156c919fe2250

Either comment out those lines in your local copy of train.py, or pull the latest.   If you are running the container, the container won't have been updated, so make the change inside container.   Or just restart training from the beginning.

4boris, 2023-07-27:
Thank you for the help!

4boris, 2023-07-28:
I am training object detection with a dataset which I created using the camera-capture. after the completion of every epoch in the training set i see the checkpoint path print out and I can see all the checkpoints in my directory but when I run the command with the --resume flag and the checkpoint path, it loads the checkpoint but then begins from epoch 0 again.

dusty-nv, 2023-07-28:
I believe its normal for the training script to start the epoch count from 0 when --resume is used

Also, I would use a different model dir when you --resume , otherwise it will overwrite your previous checkpoints

4boris, 2023-07-28:
if it starts again from epoch 0 does it mean all the training progress was lost or is that just a restart on the epoch counter?

dusty-nv, 2023-07-28:
Its just the counter, the training process is still kept by virtue of the model weights that you resumed from