dusty-nv/jetson-inference #1708: CudaCrop reducing effectivness of my model

manicmojo, 2023-07-31:
I've trained my first model, low accuracy at the moment. But runs okay enough for now on one camera full video stream. 

However, in my code I am using CudaCrop so detectnet does run on things outside my ROI. However, this dramatically drops my detection rate. 

I suppose this is due to the BACKGROUND not learning about the black side bars as these were not in my training images.

Can I artificially add these to training images? Repeat images but with black side bars? Do they have to be exact? How many?

Or is there please a simpler solution? 

Thankyou.
Jetson Orin 64gb, Python3.8

dusty-nv, 2023-07-31:
@manicmojo I can't predict the exact pre-processing or image augmentation that you need to perform to get your model to perform well on your real-world data, but my best recommendation is to train your model on data that is representative of what it will be run on during inferencing.   So if the images are cropped, at least some of the images in the training set should be cropped too.   I believe the train_ssd.py does some cropping during it's image augmentations but you may need to experiment with this to get the desired results.

manicmojo, 2023-07-31:
Thankyou, dont suppose you know of any nifty tools to edit data, images/xml. I could photoshop each one, and edit them manually. 

If not, thankyou for your help!