dusty-nv/jetson-inference #1709: How do I change Gstreamer resolution profile - v4l2

manicmojo, 2023-07-31:
My 4k v4l2 m.264 cameras are set at different resolutions. None at the 3840x2160 16fps res, as seen in profile 0 below. 

I can go into gst-launch at the right resolution, if i set it. 
But when i go into camera-capture, or my python code, even with settings prior. gstcamera does the code below.

`[gstreamer] gstCamera -- found v4l2 device: vi-output, ar0821 32-0044
[gstreamer] v4l2-proplist, device.path=(string)/dev/video4, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)tegra-video, v4l2.device.card=(string)"vi-output\,\ ar0821\ 32-0044", v4l2.device.bus_info=(string)platform:tegra-capture-vi:5, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found v4l2 device: vi-output, ar0821 32-0043
[gstreamer] v4l2-proplist, device.path=(string)/dev/video3, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)tegra-video, v4l2.device.card=(string)"vi-output\,\ ar0821\ 32-0043", v4l2.device.bus_info=(string)platform:tegra-capture-vi:4, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found v4l2 device: vi-output, ar0821 31-0044
[gstreamer] v4l2-proplist, device.path=(string)/dev/video2, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)tegra-video, v4l2.device.card=(string)"vi-output\,\ ar0821\ 31-0044", v4l2.device.bus_info=(string)platform:tegra-capture-vi:3, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found v4l2 device: vi-output, ar0821 31-0043
[gstreamer] v4l2-proplist, device.path=(string)/dev/video1, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)tegra-video, v4l2.device.card=(string)"vi-output\,\ ar0821\ 31-0043", v4l2.device.bus_info=(string)platform:tegra-capture-vi:2, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found v4l2 device: vi-output, ar0821 30-0043
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)tegra-video, v4l2.device.card=(string)"vi-output\,\ ar0821\ 30-0043", v4l2.device.bus_info=(string)platform:tegra-capture-vi:0, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 6 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)UYVY, width=(int)3840, height=(int)2160, framerate=(fraction)16/1;
[gstreamer] [1] video/x-raw, format=(string)UYVY, width=(int)1920, height=(int)1080, framerate=(fraction)60/1;
[gstreamer] [2] video/x-raw, format=(string)UYVY, width=(int)1280, height=(int)720, framerate=(fraction)60/1;
[gstreamer] [3] video/x-raw, format=(string)NV16, width=(int)3840, height=(int)2160, framerate=(fraction)16/1;
[gstreamer] [4] video/x-raw, format=(string)NV16, width=(int)1920, height=(int)1080, framerate=(fraction)60/1;
[gstreamer] [5] video/x-raw, format=(string)NV16, width=(int)1280, height=(int)720, framerate=(fraction)60/1;
[gstreamer] gstCamera -- selected device profile:  codec=raw format=uyvy width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! video/x-raw, format=(string)UYVY, width=(int)1280, height=(int)720 ! appsink name=mysink sync=false
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
`

Nothing I try changes it. 
This does not:
`v4l2-ctl -d /dev/videoX --set-fmt-video=width=3840,height=2160,pixelformat=1`

Help
Jetson Orin, R35, python3.8, linux debian arm64.

dusty-nv, 2023-07-31:
Hi @manicmojo, what is the command line or Python code that you are running to set the camera resolution with camera-capture, video-viewer/video-viewer.py, ect?    Can you try it with `video-viewer.py --input-width=3840 --input-height=2160 /dev/videoX` ?

manicmojo, 2023-07-31:
The command line which gave the code i originally posted was simply `camera-capture /dev/video0`.
The code you suggests displays the screen in tat resolution 3840x2160. I have no problem displaying. But I cannot capture, or in my python code to run detectnet on get the video streams to be that resolution, or any other than it automatically comes up as.

Which, is weird as I have 6 of the same camera, some set in Gstreamer at 1920x1080 and some at 1280x720 all running in the same python program.

`v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'UYVY' (UYVY 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 3840x2160
			Interval: Discrete 0.062s (16.000 fps)
	[1]: 'NV16' (Y/CbCr 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 3840x2160
			Interval: Discrete 0.062s (16.000 fps)
`
I tried using subprocess in my python code to no luck either.
`import subprocess

#Set gstcamera profile using v4l2-ctl
def set_gstcamera_profile(device, profile_number):
    subprocess.run(f'v4l2-ctl -d {device} --set-selection=1 --selection-target=1 --set-profile={profile_number}', shell=True)
`

Related, but I couldn't find help in it:
https://developer.download.nvidia.com/embedded/L4T/r31_Release_v1.0/Docs/Accelerated_GStreamer_User_Guide.pdf?t=eyJscyI6ImdzZW8iLCJsc2QiOiJodHRwczovL3d3dy5nb29nbGUuY29tLyJ9

What really gets me is the `[gstreamer] gstCamera -- found 6 caps for v4l2 device /dev/video` and then `selected device profile: `
Please let me select GST!

manicmojo, 2023-08-02:
I've learnt the cameras dont work at the 4k resolution with ARGUS API, but do work at 1920x1080, so the issue still persists as i need to set this..

`$ gst-launch-1.0 v4l2src device=/dev/video0 ! "video/x-raw, format=(string)UYVY, width=(int)1920, height=(int)1080" ! nvvidconv ! "video/x-raw(memory:NVMM), format=(string)I420, width=(int)1920, height=(int)1080" ! nv3dsink sync=false`

Works fine. Alas. Not within the program, even if i use similar v4l2 code to set it at the start..