dusty-nv/jetson-inference #1711: cmake issue

sai-ssauto, 2023-08-01:
Hi @dusty-nv,

I am using Advantech Mic-710ai jetson nano.
I get BSP from Advantech and i flashed the board also.
Then I change the bootable mode from emmc to SD card like how they mentioned the SOP.
Now I tried to install jetson-inference on that. but i am facing some error on the Cmake step.

I will paste my error here. 

-- Configuring incomplete, errors occurred!
See also "/home/mic-710ai/jetson-inference/build/CMakeFiles/CMakeOutput.log".

And after this steps I cannot run others steps also.

Thnaks in Advance

sai-ssauto, 2023-08-02:
Hi @dusty-nv,

I have attached the complete teminal log below:

mic-710ai@ubuntu:~$ cd jetson-inference/
mic-710ai@ubuntu:~/jetson-inference$ cd build/
mic-710ai@ubuntu:~/jetson-inference/build$ cmake ../
 
Note:  this project uses git submodules in the source tree.
       if you haven't already, run the following command from
       the project's root directory:
 
           git submodule update --init

-- distro ID:       Ubuntu
-- distro version:  18.04
-- distro codename: bionic
-- using patched FindCUDA.cmake
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- CUDA version: 
-- CUDA  detected (aarch64), enabling SM_53 SM_62
-- OpenCV version:  3.2.0
-- OpenCV version >= 3.0.0, enabling OpenCV
CMake Warning at CMakeLists.txt:106 (find_package):
  By not providing "FindVPI.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "VPI", but
  CMake did not find one.

  Could not find a package configuration file provided by "VPI" (requested
  version 2.0) with any of the following names:

    VPIConfig.cmake
    vpi-config.cmake

  Add the installation prefix of "VPI" to CMAKE_PREFIX_PATH or set "VPI_DIR"
  to a directory containing one of the above files.  If "VPI" provides a
  separate development package or SDK, be sure it has been installed.

-- didn't find VPI on system, disabling VPI
-- system arch:  aarch64
-- output path:  /home/mic-710ai/jetson-inference/build/aarch64
-- BUILD_EXPERIMENTAL disabled
-- Copying /home/mic-710ai/jetson-inference/c/actionNet.h
-- Copying /home/mic-710ai/jetson-inference/c/backgroundNet.h
-- Copying /home/mic-710ai/jetson-inference/c/depthNet.h
-- Copying /home/mic-710ai/jetson-inference/c/detectNet.h
-- Copying /home/mic-710ai/jetson-inference/c/imageNet.h
-- Copying /home/mic-710ai/jetson-inference/c/modelDownloader.h
-- Copying /home/mic-710ai/jetson-inference/c/poseNet.h
-- Copying /home/mic-710ai/jetson-inference/c/segNet.h
-- Copying /home/mic-710ai/jetson-inference/c/tensorConvert.h
-- Copying /home/mic-710ai/jetson-inference/c/tensorNet.h
-- Copying /home/mic-710ai/jetson-inference/c/calibration/randInt8Calibrator.h
-- Copying /home/mic-710ai/jetson-inference/c/tracking/objectTracker.h
-- Copying /home/mic-710ai/jetson-inference/c/tracking/objectTrackerIOU.h
-- Copying /home/mic-710ai/jetson-inference/c/tracking/objectTrackerKLT.h
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- found Qt5Widgets version:  5.9.5
-- found Qt5Widgets defines:  -DQT_WIDGETS_LIB;-DQT_GUI_LIB;-DQT_CORE_LIB
-- found Qt5Widgets library:  Qt5::Widgets
-- found Qt5Widgets include:  /usr/include/aarch64-linux-gnu/qt5/;/usr/include/aarch64-linux-gnu/qt5/QtWidgets;/usr/include/aarch64-linux-gnu/qt5/QtGui;/usr/include/aarch64-linux-gnu/qt5/QtCore;/usr/lib/aarch64-linux-gnu/qt5//mkspecs/linux-g++
-- camera-capture:  building as submodule of /home/mic-710ai/jetson-inference/tools
-- jetson-utils:  building as submodule, /home/mic-710ai/jetson-inference
-- nvbuf_utils:  /usr/lib/aarch64-linux-gnu/tegra/libnvbuf_utils.so
-- NVMM zero-copy memory:  ENABLE_NVMM=ON
-- Copying /home/mic-710ai/jetson-inference/utils/alphanum.h
-- Copying /home/mic-710ai/jetson-inference/utils/commandLine.h
-- Copying /home/mic-710ai/jetson-inference/utils/csvReader.h
-- Copying /home/mic-710ai/jetson-inference/utils/csvWriter.h
[... 343 lines cut ...]
   used as include directory in directory /home/mic-710ai/jetson-inference/python
   used as include directory in directory /home/mic-710ai/jetson-inference/python
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings
   used as include directory in directory /home/mic-710ai/jetson-inference/python/bindings

-- Configuring incomplete, errors occurred!
See also "/home/mic-710ai/jetson-inference/build/CMakeFiles/CMakeOutput.log".

dusty-nv, 2023-08-02:
> CUDA_TOOLKIT_ROOT_DIR not found or specified
> -- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY)

@sai-ssauto you don't have CUDA Toolkit installed (and presumably, neither cuDNN or TensorRT either).    Did you have it flashed with SDK Manager when you setup the board?

sai-ssauto, 2023-08-03:
Hi @dusty-nv,

I didn't flash with SDK Manager. Because This is not a nvidia carrier board. I have recevied some sop and bsp files from the vendor. with using that I flashed the board.

dusty-nv, 2023-08-03:
You can either use SDK Manager to _only_ install the JetPack components (deselect flashing the L4T OS), or you can install CUDA/cuDNN/TensorRT/ect via apt

sai-ssauto, 2023-08-04:
Hi @dusty-nv 
First thing I tried, I flash my advantch jetson nano Mic-710ai with sdk manager. that time i tried to boot my system. On that time SD card is not detected because of the DIfferent BSP file. Then again I flashed my device with bsp file which is provided by the advantech. On the cuda packages are missing. So as per our discussion I installed those things from the sdk manager. After that i tried to install jetson inference while installing the requirements of the object detection i am facing error. I have mentioned below the error for your reference 

    Running command python setup.py egg_info
    warning: pandas/_libs/algos_take_helper.pxi:51:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:178:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:305:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:432:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:559:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:686:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:813:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:940:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:1067:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:1194:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:1321:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:1448:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:1575:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:1702:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:1829:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:1956:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:2083:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:2210:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos_take_helper.pxi:2337:4: The 'IF' statement is deprecated and will be removed in a future Cython version. Consider using runtime conditions or C macros instead. See https://github.com/cython/cython/issues/4310
    warning: pandas/_libs/algos.pyx:141:42: Unknown type declaration 'bint' in annotation, ignoring
    warning: pandas/_libs/groupby.pyx:1130:26: Unreachable code
[... cut ...]
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 129, in prep_for_dist
    self.req_to_install.run_egg_info()
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 439, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/usr/lib/python3/dist-packages/pip/utils/__init__.py", line 725, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pip.exceptions.InstallationError: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-6lk041k5/pandas/

dusty-nv, 2023-08-04:
Hmm, I'm not sure if you need to upgrade Cython or what - try this:

`pip3 install --upgrade 'Cython<3'`

also, you only need pandas if you are training your own object detection model with train_ssd.py

you can also try using the jetson-inference docker container which has this all pre-built in it

sai-ssauto, 2023-08-04:
Hi @dusty-nv
Thanks for your guideness, I have installed successfully. But I ran the following comment but its show error.

mic-710ai@ubuntu:~/jetson-inference/python/training/detection/ssd$ ./detectnet /home/mic-710ai/Downloads/dog.jpg
bash: ./detectnet: No such file or directory

sai-ssauto, 2023-08-04:
Hi @dusty-nv 
Yes! you are right. Now I upgrade cython so the error was cleared. All the configurations are done. I tested with default model its working good. Tomorrow I will start training for the custom dataset.

Thanks for your help and proper guideness.

sai-ssauto, 2023-08-05:
Hi @dusty-nv 

While I am trying to train a custom object detection model, on that time I am facing some errors. I mentioned below the error which I got it.

mic-710ai@ubuntu:~/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --dataset-type=voc --data=/home/mic-710ai/jetson-inference/python/training/detection/ssd/data/station5 --model-dir=models/station5 --batch-size=4 --epochs=100
2023-08-05 17:51:42 - Using CUDA...
2023-08-05 17:51:42 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/station5', dataset_type='voc', datasets=['/home/mic-710ai/jetson-inference/python/training/detection/ssd/data/station5'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, log_level='info', lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=100, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resolution=300, resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, validation_mean_ap=False, weight_decay=0.0005)
2023-08-05 17:51:56 - model resolution 300x300
2023-08-05 17:51:56 - SSDSpec(feature_map_size=19, shrinkage=16, box_sizes=SSDBoxSizes(min=60, max=105), aspect_ratios=[2, 3])
2023-08-05 17:51:56 - SSDSpec(feature_map_size=10, shrinkage=32, box_sizes=SSDBoxSizes(min=105, max=150), aspect_ratios=[2, 3])
2023-08-05 17:51:56 - SSDSpec(feature_map_size=5, shrinkage=64, box_sizes=SSDBoxSizes(min=150, max=195), aspect_ratios=[2, 3])
2023-08-05 17:51:56 - SSDSpec(feature_map_size=3, shrinkage=100, box_sizes=SSDBoxSizes(min=195, max=240), aspect_ratios=[2, 3])
2023-08-05 17:51:56 - SSDSpec(feature_map_size=2, shrinkage=150, box_sizes=SSDBoxSizes(min=240, max=285), aspect_ratios=[2, 3])
2023-08-05 17:51:56 - SSDSpec(feature_map_size=1, shrinkage=300, box_sizes=SSDBoxSizes(min=285, max=330), aspect_ratios=[2, 3])
2023-08-05 17:51:56 - Prepare training datasets.
2023-08-05 17:51:59 - VOC Labels read from file:  ('BACKGROUND', 'Red_cap', 'round_mounting_plate', 'user_manual', 'mounting_hardware_bag', 'ac_harness')
2023-08-05 17:51:59 - Stored labels into file models/station5/labels.txt.
2023-08-05 17:51:59 - Train dataset size: 2662
2023-08-05 17:51:59 - Prepare Validation datasets.
2023-08-05 17:52:00 - VOC Labels read from file:  ('BACKGROUND', 'Red_cap', 'round_mounting_plate', 'user_manual', 'mounting_hardware_bag', 'ac_harness')
2023-08-05 17:52:00 - Validation dataset size: 295
2023-08-05 17:52:00 - Build network.
2023-08-05 17:52:01 - Init from pretrained SSD models/mobilenet-v1-ssd-mp-0_675.pth
2023-08-05 17:52:01 - Took 0.54 seconds to load the model.
2023-08-05 17:52:02 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2023-08-05 17:52:02 - Uses CosineAnnealingLR scheduler.
2023-08-05 17:52:02 - Start training from epoch 0.
Traceback (most recent call last):
  File "/home/mic-710ai/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 779, in _try_get_data
    data = self._data_queue.get(timeout=timeout)
  File "/usr/lib/python3.6/multiprocessing/queues.py", line 104, in get
    if not self._poll(timeout):
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 257, in poll
    return self._poll(timeout)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 414, in _poll
    r = wait([self], timeout)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 918, in wait
    return ready
  File "/usr/lib/python3.6/selectors.py", line 202, in __exit__
    self.close()
  File "/usr/lib/python3.6/selectors.py", line 269, in close
    self._fd_to_key.clear()
  File "/home/mic-710ai/.local/lib/python3.6/site-packages/torch/utils/data/_utils/signal_handling.py", line 66, in handler
    _error_if_any_worker_fails()
RuntimeError: DataLoader worker (pid 12333) is killed by signal: Killed. 

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "train_ssd.py", line 406, in <module>
    train(train_loader, net, criterion, optimizer, device=DEVICE, debug_steps=args.debug_steps, epoch=epoch)
  File "train_ssd.py", line 139, in train
    for i, data in enumerate(loader):
  File "/home/mic-710ai/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 363, in __next__
    data = self._next_data()
  File "/home/mic-710ai/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 974, in _next_data
    idx, data = self._get_data()
  File "/home/mic-710ai/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 941, in _get_data
    success, data = self._try_get_data()
  File "/home/mic-710ai/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 792, in _try_get_data
    raise RuntimeError('DataLoader worker (pid(s) {}) exited unexpectedly'.format(pids_str))
RuntimeError: DataLoader worker (pid(s) 12333) exited unexpectedly

dusty-nv, 2023-08-05:
> RuntimeError: DataLoader worker (pid 12333) is killed by signal: Killed.

Hi @sai-ssauto anytime you get the error `Killed`, it probably means that your board ran out of memory and the process was killed by the kernel.   See the suggestions here:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md#training-the-ssd-mobilenet-model

* mount swap
* disable GUI
* `--workers=1`
* `--batch-size=1`