dusty-nv/jetson-inference #1715: WebRTC - The stream has not shown in the browser

aihamshikhomicro, 2023-08-03:
Hi,
I have an issue and one question:
I run this command:
**$ video-viewer /dev/video0 webrtc://@:8554/output**

and I got this output everything looks fine but I try to open the  **https://localhost:8554** in the browser i got an empty video element.
but the video-viewer interface  works good.
However here is my log with  **failed to find/open file /proc/device-tree/model** warning:

 -- sslKey      /jetson-inference/data/key.pem
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
**failed to find/open file /proc/device-tree/model**
[gstreamer] gstEncoder -- detected board 'Jetson AGX Orin'
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! nvvidconv name=vidconv ! video/x-raw(memory:NVMM) ! nvv4l2h264enc name=encoder bitrate=4000000 insert-sps-pps=1 insert-vui=1 idrinterval=30 maxperf-enable=1 ! video/x-h264 ! rtph264pay config-interval=1 ! application/x-rtp,media=video,encoding-name=H264,clock-rate=90000,payload=96 ! tee name=videotee ! queue ! fakesink
[webrtc] WebRTC server started @ https://microorin1-desktop:8554
[webrtc] WebRTC server thread running...
[webrtc] websocket route added /output
[video]  created gstEncoder from webrtc://@:8554/output
------------------------------------------------
gstEncoder video options:
------------------------------------------------
.
.
.
.
.
.
.
video-viewer:  captured 6150 frames (1280x720)
video-viewer:  captured 6175 frames (1280x720)
video-viewer:  captured 6200 frames (1280x720)
video-viewer:  captured 6225 frames (1280x720)
video-viewer:  captured 6250 frames (1280x720)
video-viewer:  captured 6275 frames (1280x720)
video-viewer:  captured 6300 frames (1280x720)
video-viewer:  captured 6325 frames (1280x720)
video-viewer:  captured 6350 frames (1280x720)
video-viewer:  captured 6375 frames (1280x720)
video-viewer:  captured 6400 frames (1280x720)
video-viewer:  captured 6425 frames (1280x720)

[image]

and my question is how i can publish the IP camera with RTSP using WebRTC  as same the example with the USB camera.

dusty-nv, 2023-08-03:
> **failed to find/open file /proc/device-tree/model**
> [gstreamer] gstEncoder -- detected board 'Jetson AGX Orin'

Hi @aihamshikhomicro, are you actually running this on AGX Orin, or what are you running it on?

Can you try using Chrome/Chromium as browser instead of Firefox?

> and my question is how i can publish the IP camera with RTSP using WebRTC as same the example with the USB camera.

To use RTSP as the input stream instead of USB camera, you would specify your RTSP URL as the input stream like here:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#rtsp-input

```
video-viewer rtsp://$CAMERA_IP:1234 webrtc://@:8554/output
```

However I would recommend to get it working first with your USB camera

aihamshikhomicro, 2023-08-04:
yes, I am running this command on AGX Orin.
I am try with chrom and still the same result.
[image]

dusty-nv, 2023-08-04:
@aihamshikhomicro does it work over LAN, or is that WAN IP?

Can you post the browser console debug log (Ctrl+Shift+I) and the terminal log from your Jetson?

aihamshikhomicro, 2023-08-07:
I am work over LAN.
this is terminal log :
ideo0 webrtc://@:8554/output
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: Logitech BRIO
[gstreamer] v4l2-proplist, device.path=(string)/dev/video2, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Logitech\ BRIO", v4l2.device.bus_info=(string)usb-3610000.xhci-3.4, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found v4l2 device: Logitech BRIO
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Logitech\ BRIO", v4l2.device.bus_info=(string)usb-3610000.xhci-3.4, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 43 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)960, height=(int)540, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)848, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [7] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [8] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [9] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [10] video/x-raw, format=(string)YUY2, width=(int)440, height=(int)440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [11] video/x-raw, format=(string)YUY2, width=(int)480, height=(int)270, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [12] video/x-raw, format=(string)YUY2, width=(int)340, height=(int)340, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [13] video/x-raw, format=(string)YUY2, width=(int)424, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [14] video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [15] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [16] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)180, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [17] video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [18] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [19] image/jpeg, width=(int)4096, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [20] image/jpeg, width=(int)3840, height=(int)2160, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [21] image/jpeg, width=(int)2560, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [22] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [23] image/jpeg, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [24] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 90/1, 60/1, 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[... cut ...]
[OpenGL] glDisplay -- the window has been closed
video-viewer:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped
[gstreamer] gstEncoder -- shutting down pipeline, sending EOS
[gstreamer] gstEncoder -- transitioning pipeline to GST_STATE_NULL
[gstreamer] gstEncoder -- pipeline stopped
[webrtc] WebRTC server on port 8554 is shutting down
[webrtc] WebRTC server thread stopped
video-viewer:  shutdown complete

and this is browser console debug log :
[image]

aihamshikhomicro, 2023-08-07:
this is chrome browser console debug log : 
[image]

MOXHAN, 2023-10-23:
Hi @dusty-nv, 
is there any Update regarding this problem? I have the same problem (**failed to find/open file /proc/device-tree/model**) + OpenGL is'nt able to create a display window. I use the default jetson-inference container and video-viewer actually works fine locally, but when I try to use videoSource and videoOutput from jetson.utils with a WebRTC Output, I have this problem (Posted also on NVIDIA Forum: https://forums.developer.nvidia.com/t/opengl-failed-to-create-x11-window-when-using-videooutput-in-container/270118?u=maxim.hansen2000 )

dusty-nv, 2023-10-23:
@MOXHAN see my reply to your forum thread here:

https://forums.developer.nvidia.com/t/opengl-failed-to-create-x11-window-when-using-videooutput-in-container/270118/4

Unlicensed-driver-ljx, 2024-02-27:
哈喽这个问题解决了吗，使用docker启动的话

Unlicensed-driver-ljx, 2024-02-27:
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
failed to find/open file /proc/device-tree/model
[image]
@dusty-nv

bugraaydogar, 2024-03-10:
Hi @dusty-nv,

I also have the same problem. I also have read your answer above in the nvidia developer forum but it does not really help. Do you have any guidance for those who are facing with this problem? 

Here are the logs for your reference; https://pastebin.ubuntu.com/p/tGcj2jhQry/

Thanks for the collaboration.
Bugra

piter-24, 2025-02-04:
Any updates on this?