dusty-nv/jetson-inference #1718: Hello World on Jetson Orin Nano - docker/run.sh: line 198: [: =: unary operator expected

moh-C, 2023-08-10:
Hey!
I have been having really a bad time with Jetson Orin Nano since the compatibility issues are just endless. I started S1E1 from NVIDIA Developer channel on YouTube which did not work at all (since they are for JetPack 4.6 as opposed to 5.1 which is for Orin). NVIDIA forum suggested I start watching the S3E1 from that very same channel wherein you start giving tutorials which would be compatible with Orin.
When I try to run the docker/run.sh, I get the following error:

`docker/tag.sh: line 4: docker/containers/scripts/l4t_version.sh: No such file or directory
docker/tag.sh: line 6: [: =: unary operator expected
docker/tag.sh: line 36: [: =: unary operator expected
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/jetson-inference:
DATA_VOLUME:      --volume /home/mohammad/GitHub/jetson-inference/data:/jetson-inference/data --volume /home/mohammad/GitHub/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/mohammad/GitHub/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/mohammad/GitHub/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/mohammad/GitHub/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models --volume /home/mohammad/GitHub/jetson-inference/python/www/recognizer/data:/jetson-inference/python/www/recognizer/data 
V4L2_DEVICES:     --device /dev/video0  --device /dev/video1 
DISPLAY_DEVICE:   -e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix 
docker/run.sh: line 198: [: =: unary operator expected
docker/run.sh: line 213: [: =: unary operator expected
`

I have also cloned the latest version. My camera is also plugged in. But I really do not get why should I get "[:=: unary operator expected" error. I would really appreciate it if you told me what should I do for it to be compatible with jetpack 5.1.
Or if this repo is just for Jetson Nano (so JetPack 5.x should be excluded)? How and where should I start for Jetson Orin?!

dusty-nv, 2023-08-11:
Hi @moh-C, sorry for the confusion, and yes jetson-inference is fully compatible with JetPack 5 and Jetson Orin.  When you cloned the repo, did you use `--recursive`?   If not, do `git submodule update --init`

BTW I hope that https://github.com/dusty-nv/jetson-containers is also useful to you

yasmineLalabouali, 2024-01-07:
I'm having the same errors even after running git submodule update --init, is there any way to solve it ?

topherbuckley, 2024-03-12:
Are you running it from the root of the repo? e.g. `cd docker && ./run` does not work for some reason. I'd guess it has to do with relative paths and expecting to be run from the root directory.

andanders, 2024-07-12:
@moh-C  I had the same issue. As @topherbuckley pointed out, you must run `run.sh` from the parent dir.
I.e. `docker/run.sh`.