dusty-nv/jetson-inference #1724: RTSP Output to another remote server

angshu86, 2023-09-06:
Hi dusty-nv!

First off thank you very much for this repo, it's truly helpful in learning and implementing stuff.

I just had a question regarding sending RTSP stream output from one of the binaries like video-viewer, or detectnet etc. to a remote rtsp server instead of localhost within the Jetson. 

`rtsp://@:8554/mystream` as output works just fine using the rtsp mechanism provided but I was wondering if I am able to forward this output stream to a remote streaming server that is running an RTSP server (something like rtsp-simple-server) that would then be the endpoint for media-players to play out the stream? 

So something like ` ./detectnet rtsp://192.168.0.12:8554/inputStream rtsp://192.168.0.20:8554/outputStream`. 

Sorry if I missed an already existing thread about it. I did look around but I only found one issue where you mentioned webrtc was in listening mode (similar to RTSP). That was a clue that maybe remote IP isn't possible for RTSP but I wanted to confirm anyway. Also, if this isn't possible now, do you know if it's something that I could easily implement with some tinkering, or would it be a gigantic undertaking?

Thanks in advance.

dusty-nv, 2023-09-06:
Hi @angshu86, as you alluded to, RTSP output is a server in listening mode.   Your other RTSP server is also in listening mode, and only publishes streams.   Instead, you would need to run a gstreamer pipeline with `rtspsrc` (or similar) on your other server and relay it (unsure about the details of that or which RTSP implementation you are using on this server)

Alternatively, RTP is done in push mode, so you can output things like `rtp://192.168.0.20:8554` directly to a known host

angshu86, 2023-09-07:
Hi dusty-nv,

Thanks for your prompt response and shedding light into this. For anyone trying to do this, I now managed to get relay working with a simple ffmpeg line, albeit with some `max_delay_reached` and `RTP: missed X packets` warnings. But overall the stream is not too impacted, occasional macroblocking but acceptable using:

`ffmpeg -i rtsp://localhost:8554/mystream -c:v copy -f rtsp -rtsp_transport tcp rtsp://192.168.0.20:8554/output1`

Gst pipelines I tried didn't want to work throwing some linking errors between the input to output no matter what I tried so I'll have to play around with that a bit more later. Thank you.

dusty-nv, 2023-09-07:
Oh great!, glad you got it running @angshu86.  Cool solution 👍

closed 2023-09-07 by angshu86