dusty-nv/jetson-inference #1727: no element "nvv4l2decoder" in Docker

poett1, 2023-09-12:
I'm trying to run the detectnet tracking example using docker on Xavier AGX (L4T 35.2.1). It seems like the nvv4l2decoder is not installed in the container. 
```
root@xavieragx:/jetson-inference/build/aarch64/bin# detectnet --model=peoplenet --tracking pedestrians.mp4 pedestrians_tracking.mp4
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for pedestrians.mp4
[gstreamer] gstDecoder -- discovered video resolution: 960x540  (framerate 29.970030 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)avc, alignment=(string)au, level=(string)3.1, profile=(string)high, codec_data=(buffer)0164001fffe100196764001facd940f0117ef01100000303e90000ea608f18319601000668ebe132c8b0, width=(int)960, height=(int)540, framerate=(fraction)30000/1001, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=pedestrians.mp4 ! qtdemux ! queue ! h264parse ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink
[gstreamer] gstDecoder -- failed to create pipeline
[gstreamer]    (no element "nvv4l2decoder")
[gstreamer] failed to create decoder pipeline
[gstreamer] gstDecoder -- failed to create decoder for file:///jetson-inference/build/aarch64/bin/pedestrians.mp4
detectnet:  failed to create input stream
```
Gstreamer is unable to find the plugin:
```
root@xavieragx:/jetson-inference/build/aarch64/bin# gst-inspect-1.0 nvv4l2decoder
No such element or plugin 'nvv4l2decoder'
```
Any hints?

dusty-nv, 2023-09-12:
Is you run gst-inspect outside container, can nvv4l2decoder be found?   These plug-ins get mounted into container by gstreamer

poett1, 2023-09-12:
```
user@xavieragx:~/Dev$ gst-inspect-1.0 nvv4l2decoder
Factory Details:
  Rank                     primary + 11 (267)
  Long-name                NVIDIA v4l2 video decoder
  Klass                    Codec/Decoder/Video
  Description              Decode video streams via V4L2 API
  Author                   Nicolas Dufresne <nicolas.dufresne@collabora.com>, Viranjan Pagar <vpagar@nvidia.com>

Plugin Details:
  Name                     nvvideo4linux2
  Description              Nvidia elements for Video 4 Linux
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideo4linux2.so
  Version                  1.14.0
  License                  LGPL
  Source module            nvvideo4linux2
  Binary package           nvvideo4linux2
  Origin URL               http://nvidia.com/
[...]
```
Outside the container everything looks good so far.

dusty-nv, 2023-09-13:
Hmm, does your `/etc/nvidia-container-runtime/host-files-for-container.d/l4t.csv` file look like this?

``` bash
cat /etc/nvidia-container-runtime/host-files-for-container.d/l4t.csv | grep gst
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvarguscamerasrc.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvcompositor.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvdrmvideosink.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnveglglessink.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnveglstreamsrc.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvegltransform.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvivafilter.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvjpeg.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvtee.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvv4l2camerasrc.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvidconv.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideo4linux2.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideosink.so
lib, /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideosinks.so
lib, /usr/lib/aarch64-linux-gnu/libgstnvegl-1.0.so.0
lib, /usr/lib/aarch64-linux-gnu/libgstnvexifmeta.so
lib, /usr/lib/aarch64-linux-gnu/libgstnvivameta.so
lib, /usr/lib/aarch64-linux-gnu/tegra/libgstnvcustomhelper.so.1.0.0
lib, /usr/lib/aarch64-linux-gnu/tegra/libgstnvdsseimeta.so.1.0.0
sym, /usr/lib/aarch64-linux-gnu/tegra/libgstnvcustomhelper.so
sym, /usr/lib/aarch64-linux-gnu/tegra/libgstnvdsseimeta.so
```
```

tonynajjar, 2024-06-14:
I'm having the same issue, @poett1 did you manage to make it work?

poett1, 2024-06-16:
I solved it by setting `/etc/docker/daemon.json` to:
```json 
{
    "runtimes": {
        "nvidia": {
            "path": "/usr/bin/nvidia-container-runtime",
            "runtimeArgs": []
        }
    },
	"default-runtime": "nvidia"
}
```
Otherwise the Nividia libraries are not mounted into the container.

closed 2024-06-16 by poett1