dusty-nv/jetson-inference #1728: Docker not able to find ros containers

bigcat77, 2023-09-13:
Hi Dusty, 

I just got a new jetson orin development board and am trying to replicate what I had working on the older nano.  I cloned your repo and got the standard docker working with the example programs. Then I tried to get the ros version working and ran into the following problem. 

./docker/run.sh --ros=humble
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R35.4.1
[sudo] password for eve05: 
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/ros:humble-pytorch-l4t-r35.4.1
ROS_DISTRO:       humble
DATA_VOLUME:      --volume /home/eve05/jetson-inference/data:/jetson-inference/data --volume /home/eve05/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/eve05/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/eve05/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/eve05/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models --volume /home/eve05/jetson-inference/python/www/recognizer/data:/jetson-inference/python/www/recognizer/data 
DISPLAY_DEVICE:   -e DISPLAY=:0 -v /tmp/.X11-unix/:/tmp/.X11-unix 
Unable to find image 'dustynv/ros:humble-pytorch-l4t-r35.4.1' locally
docker: Error response from daemon: manifest for dustynv/ros:humble-pytorch-l4t-r35.4.1 not found: manifest unknown: manifest unknown.
See 'docker run --help'.

After some googling, it seems that docker is not able to find the specified container in your repo. Can you provide any suggestions? 

Thanks so much!

dusty-nv, 2023-09-13:
Hi @bigcat77, try running it like this instead:

```bash
docker/run.sh -c dustynv/ros:humble-pytorch-l4t-r35.3.1
```

nan0AI, 2023-09-21:
I got the same issue, still does not work, it takes me to root instead. 

ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R35.3.1
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/ros:humble-pytorch-l4t-r35.3.1
DATA_VOLUME:      --volume /home/

Bunch of lines in the middle and this at the end

DISPLAY_DEVICE:   -e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix 
sourcing   /opt/ros/humble/install/setup.bash
sourcing   /ros_deep_learning/install/setup.bash
ROS_DISTRO humble
ROS_ROOT   /opt/ros/humble
root@ubuntu:/# ^C
root@ubuntu:/# 

> Hi @bigcat77, try running it like this instead:
> 
> ```shell
> docker/run.sh -c dustynv/ros:humble-pytorch-l4t-r35.3.1
> ```

dusty-nv, 2023-09-21:
@nan0AI that root prompt is running in the container