dusty-nv/jetson-inference #1730: Error compiling jetson-inference

barsigor, 2023-09-14:
Hello,
I'm trying to compile jetson-inference on a xavier nx, i installed latest jetpack ( 5.1.2 ) then i pull the latest L4T base container ( 35.3.1 ), when i try to compile the jetson inference repo (latest master) i got this error:

`CMake Error at utils/cuda/FindCUDA.cmake:1802 (add_library):
  add_library cannot create target "jetson-inference-python-38" because
  another target with the same name already exists.  The existing target is a
  shared library created in source directory
  "/root/jetson-inference/python/bindings".  See documentation for policy
  CMP0002 for more details.
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:57 (cuda_add_library)`

before cmake i installed:

`apt install build-essential git cmake nvidia-cuda-dev nvidia-cuda
`
i need to compile from source because i don't have enough storage to download the jetson-inference container (i have only a xavier nx with a 16GB internal storage...)

dusty-nv, 2023-09-14:
@barsigor are you attempting to build jetson-inference inside l4t-base container?

That's not going to save you space really, because by the time you add in cuDNN and TensorRT, you'll be back to l4t-jetpack container size.

Are you able to build jetson-inference outside container?

Xavier NX devkit has NVME slot, I recommend using it.

barsigor, 2023-09-15:
The carrier board I used doesn't have the nvme slot, I cannot install complete jetpack SDK so i cannot build from source, and I don't have space for docker container.
I solved mounting /var/lib/docker on an external USB pendrive with 64GB, it's pretty slow, but I have enough space to load the latest jetson-inference container.

Thanks,
Igor

closed 2023-09-15 by barsigor