dusty-nv/jetson-inference #1731: installation of jetson inference

GirishPawar1999, 2023-09-16:
model: jetson nano 4gb 
OS: jetpack 4 .6. 1
Opencv: 4.1.1 with cuda:No
cuda:10.2.300
cNN: 8.2.1.32
tensorRT:8.2.1.8
VPI: 1.2.3
vulkan: 1.2.70

I have been following this steps:
sudo apt-get update
sudo apt-get install git cmake libpython3-dev python3-numpy
git clone --recursive --depth=1 https://github.com/dusty-nv/jetson-inference
cd jetson-inference
mkdir build
cd build
cmake ../
make -j$(nproc)
sudo make install
sudo ldconfig

I haven't got window to install the models , ima directly prompted with pytorch installation.

after all the steps i, when i am read to run model from the jetson-inference/python/examples and try to run the model i am greeted with the below error:

"" requested fasted precision for device GPU without providing valid calibrator, disabling INT8 ""

i am new to this , pls help me out

dusty-nv, 2023-09-16:
@GirishPawar1999 the Model Downloader window was replaced with an automatic on-demand model downloader, so you don't need to manually select the models for download anymore.  They will automatically be downloaded when you try to run them.

That text about the precision you can ignore, it's not an error.

Just let it run for a few minutes while TensorRT builds the engine, and it should complete.  Then when you launch the model on subsequent runs, it will load much faster.

GirishPawar1999, 2023-09-18:
thanks , it worked. Small suggestion, ls do update the changes  in some kind of update log so that one is able to know the changes and modifications done... noobs like me refer to the video series and couldn't understand the new developed changes and things that there is some kind of issues in the procedure of implementations. Great work btw :)

dusty-nv, 2023-09-18:
Sorry @GirishPawar1999 , the change log is here:  https://github.com/dusty-nv/jetson-inference/blob/master/CHANGELOG.md

Mentions of the Model Downloader TUI were removed from the docs, but yea that's an issue with the videos getting outdated, and I don't have the bandwidth to re-record them.  Alas it's still better than having no videos I feel.

anasderkaoui, 2023-11-08:
Hello, I have one issue with the installation. When I try to "cmake ../" I don't get the same generated files as in your video.
For example, I don't have : the detectnet-console.py file, I only have detectnet-snap.py and detectnet.py
One other thing is that when I try to "make" the build directory, I get this error : nvcc fatal : unsupported gpu architecture 'compute_87' along with some other errors like "eroor genereting ".cu.o" files...
Do you know how to fix this please ?

dusty-nv, 2023-11-09:
Hi @anasderkaoui, the imagenet-console and detectnet-console variants were deprecated and removed in lieu of imagenet.py/detectnet.py that handle both camera input and file input (you can use them the same way that you would have used the -console versions)

Regarding the `compute_87` error, I'm unsure why you would get that because it's only enabled for CUDA 10, which supports sm_87.  However, you can disable it by commenting out this line of CMakeLists.txt (then re-run `cmake ../`)

https://github.com/dusty-nv/jetson-inference/blob/667bb680883f81bc5ab08463c1664cf89589938e/CMakeLists.txt#L83

anasderkaoui, 2023-11-09:
Thank you that got rid of that error! But now I have another one, in "Scanning dependencies of target jetson-utils":

In file included from /home/anas/jetson-inference/build/aarch64/include/jetson-utils/gstBufferManager.h:26:0,
                 from /home/anas/jetson-inference/utils/camera/gstCamera.h:27,
                 from /home/anas/jetson-inference/utils/camera/gstCamera.cpp:23:
/home/anas/jetson-inference/build/aarch64/include/jetson-utils/gstUtility.h:30:10: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
          ^~~~~~~~~~~
compilation terminated.
utils/CMakeFiles/jetson-utils.dir/build.make:301: recipe for target 'utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o] Error 1
CMakeFiles/Makefile2:735: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Note that from the line 9 to the end, those errors were showing before when I had that first error and still are appearing.

dusty-nv, 2023-11-09:
@anasderkaoui which Jetson are you using and version of JetPack?   TensorRT should have already come pre-installed with the SD card image or after you flashed the device with SDK Manager.

anasderkaoui, 2023-11-09:
I have a Jetson Nano and I have JetPack 4.6.b197
By typing this command apt-cache search tensorrt , I found that there was already tensorrt. But Running the other command above prompted me to install 3Gb more, so I presumed that there might be some missing packages !

dusty-nv, 2023-11-09:
@anasderkaoui okay interesting - use of VPI should be automatically disabled in jetson-inference on JetPack 4.x, but try doing it manually by commenting out these lines of CMakeLists.txt:

https://github.com/dusty-nv/jetson-inference/blob/dc7478c3c088179f4b8182d1a6390de396f502e8/CMakeLists.txt#L112

```bash
	#set(HAS_VPI 1)
	#add_definitions(-DHAS_VPI)
```

anasderkaoui, 2023-11-09:
Thanks ! That resolved the "make" issue.
Now onto another problem :) that I encountered while running my first example : './detectnet.py images/peds_0.jpg output_0.jpg'
I get this error + The other ones on the photo
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[image]

dusty-nv, 2023-11-09:
> `[TRT] Could not register plugin creator - ::FlattenConcat_TRT version 1`

That's not an error, you can ignore it.  But the other errors show that your CUDA drivers/installation aren't working.   Try the deviceQuery sample under `/usr/local/cuda/samples`.   Due to the other problems you have mentioned, I'm thinking that your board wasn't setup probably with SDK Manager or from the SD card image.

dusty-nv, 2023-11-09:
> And you are right, I did not use SDK Manager to flash the board because it's another carrier called Auvidea and they suggest their own method of flashing.

OK, gotcha - you should still be able to install SDK Manager to install all the CUDA/cuDNN/TensorRT stuff after the Auvidea L4T/BSP flashing.   Just de-select flashing of L4T/BSP in SDK Manager, and only have it install the CUDA stuff.   I'm unsure of the current environment of your board so it may need setup again.   

In that CUDA samples directory, go into `1_Utilities/deviceQuery`, then do `sudo make` and run `./deviceQuery` and see if it works.

anasderkaoui, 2023-11-09:
I get an error :

./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

cudaGetDeviceCount returned 999
-> unknown error
Result = FAIL

Okay then if I don't have CUDA correctly installed, eventhough I have installed it manually using this link : 
https://jfrog.com/connect/post/installing-cuda-on-nvidia-jetson-nano/

What should I do, knowing that I have never installed or used SDK Manager

dusty-nv, 2023-11-09:
That link points to SDK Manager first as the recommended way, it looks like that article is also manually installing CUDA 11 for ARM SBSA which explains why it doesn't work.  Restart and install CUDA 10.2 instead, or use SDK Manager from a Linux PC to do it.

anasderkaoui, 2023-11-09:
I think I might have the right version of CUDA which is 10.2 now. I saw the [image] of the carrier board and figured that I had to run a command that installs additional NVIDIA SDK components. Within those components I saw some with cuda 10.2 version. Now, I think I might have two verions of CUDA, how can I specify which one to use in TRT so that I don't get that error.

dusty-nv, 2023-11-09:
@anasderkaoui it should be the version that is symbolically linked to `/usr/local/cuda`

If you keep having much problems with this, probably re-flashing the device and keeping CUDA 10.2 from the beginning is easiest.

anasderkaoui, 2023-11-09:
I really don't want to go this way because I have a lot of other files that I don't want to lose. I am trying my best to figure it out. In the worst case I can completely uninstall CUDA with all the versions and reinstall version 10.2. I just want to know before if there is a CmakeFIle or system file of tensorrt where I can modify the version of CUDA it uses so I can default it to 10.2 !

dusty-nv, 2023-11-09:
I believe TensorRT statically links to the cuda runtime, because it's not shown under ldd:

```bash
nvidia@jax-461:/usr/lib/aarch64-linux-gnu$ ldd libnvinfer.so
        linux-vdso.so.1 (0x0000007f8fbd2000)
        librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000007f85be8000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f85bd3000)
        libnvdla_compiler.so => /usr/lib/aarch64-linux-gnu/tegra/libnvdla_compiler.so (0x0000007f85741000)
        libEGL.so.1 => /usr/lib/aarch64-linux-gnu/./libEGL.so.1 (0x0000007f85720000)
        libnvmedia.so => /usr/lib/aarch64-linux-gnu/tegra/libnvmedia.so (0x0000007f856b3000)
        libstdc++.so.6 => /usr/lib/aarch64-linux-gnu/./libstdc++.so.6 (0x0000007f8551f000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f85466000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f85442000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f852e9000)
        /lib/ld-linux-aarch64.so.1 (0x0000007f8fba6000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f852bd000)
        libnvos.so => /usr/lib/aarch64-linux-gnu/tegra/libnvos.so (0x0000007f8529f000)
        libGLdispatch.so.0 => /usr/lib/aarch64-linux-gnu/libGLdispatch.so.0 (0x0000007f85173000)
        libnvrm.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm.so (0x0000007f8513f000)
        libnvrm_graphics.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_graphics.so (0x0000007f8511f000)
        libnvdc.so => /usr/lib/aarch64-linux-gnu/tegra/libnvdc.so (0x0000007f85102000)
        libnvtvmr.so => /usr/lib/aarch64-linux-gnu/tegra/libnvtvmr.so (0x0000007f85071000)
        libnvparser.so => /usr/lib/aarch64-linux-gnu/tegra/libnvparser.so (0x0000007f85033000)
        libnvdla_runtime.so => /usr/lib/aarch64-linux-gnu/tegra/libnvdla_runtime.so (0x0000007f84f8d000)
        libnvimp.so => /usr/lib/aarch64-linux-gnu/tegra/libnvimp.so (0x0000007f84f78000)
```

If you can run deviceQuery/ect again, and have `/usr/local/cuda` pointing to 10.02, try that first.

anasderkaoui, 2023-11-09:
I have the same output as previously : 

anas@anas-desktop:/usr/local/cuda-10.2/samples/1_Utilities/deviceQuery$ ./deviceQuery

./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

cudaGetDeviceCount returned 999
-> unknown error
Result = FAIL

And I get the same result when running this command whether it be in the "cuda 10.2" directory or in the other 3 remaining directories

Some suggested to add some CUDA paths to the environment variables by running : 
export PATH=/usr/local/cuda/bin:$PATH
export LD_LIBRARY_PATH=/usr/local/cuda/lib64:$LD_LIBRARY_PATH
or adding them manually to the .bashrc file, but it did not work