dusty-nv/jetson-inference #1734: GoogleNet-12 failed to run on Jetson Nano - Message:  failed to retrieve tensor for Output "prob"

srevandro, 2023-09-18:
Hello, 

I ran all the Image Classification networks successfully, except GoogleNet-12. I tried to download the model again using de downloader script, but I still got the same error on running GoogleNet-12. I am JupyterLab on Jetson Nano to run all the models.

May you help on that? Thanks.

`
**GoogleNet-12**
!./imagenet.py --network=googlenet-12 images/jellyfish.jpg images/test/output_jellyfish.jpg
jetson.utils -- compiled without NumPy array conversion support (warning)
jetson.utils -- if you wish to have support for converting NumPy arrays,
jetson.utils -- first run 'sudo apt-get install python-numpy python3-numpy'

imageNet -- loading classification network model from:
         -- prototxt     networks/GoogleNet-ILSVRC12-subset/deploy.prototxt
         -- model        networks/GoogleNet-ILSVRC12-subset/snapshot_iter_184080.caffemodel
         -- class_labels networks/GoogleNet-ILSVRC12-subset/labels.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   1

[TRT]    TensorRT version 8.0.1
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::ScatterND version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_ONNX_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - caffe  (extension '.caffemodel')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    [MemUsageChange] Init CUDA: CPU +202, GPU +0, now: CPU 223, GPU 3802 (MiB)
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    could not find engine cache networks/GoogleNet-ILSVRC12-subset/snapshot_iter_184080.caffemodel.1.1.8001.GPU.FP16.engine
[TRT]    cache file invalid, profiling network model on device GPU
[TRT]    [MemUsageChange] Init CUDA: CPU +0, GPU +0, now: CPU 223, GPU 3802 (MiB)
[TRT]    device GPU, loading networks/GoogleNet-ILSVRC12-subset/deploy.prototxt networks/GoogleNet-ILSVRC12-subset/snapshot_iter_184080.caffemodel
[TRT]    failed to retrieve tensor for Output "prob"`

gaochuanchao, 2024-04-16:
Hi, I am not sure if someone modified the `GoogleNet-ILSVRC12-subset/deploy.prototxt`. I opened this file, and found the last layer is different from `Googlenet/googlenet.prototxt`.

the last layer of `Googlenet/googlenet.prototxt` is 
```
layer {
  name: "prob"
  type: "Softmax"
  bottom: "loss3/classifier"
  top: "prob"
}
```
and the last layer of `GoogleNet-ILSVRC12-subset/deploy.prototxt` is
```
layer {
  name: "softmax"
  type: "Softmax"
  bottom: "loss3/classifier"
  top: "softmax"
}
```
I assume this is the reason why the error message is  `failed to retrieve tensor for Output "prob"`.
After I change the last layer of `deploy.prototxt` into the same as `googlenet.prototxt` (change "softmax" into "prob"), I can run googlenet-12 successfully. **I am not sure whether this is allowed**, but it is the only way I found to run googlenet-12 successfully.

gaochuanchao, 2024-04-17:
> Thanks @gaochuanchao , interesting that this error started popping up. You should also be able to just change the output layer name that imagenet.py looks for with `--output_blob=softmax` (and not needing to edit deploy.prototxt)

Thanks for the suggestion, however, I encounter a problem when using my own python script.
I initialize the network using `net = imageNet(network_name, output_blob="softmax")`
When I run the code, the output layer is still the default option ("prob") from the output of the terminal.
[image]
It seems like passing the variable `output_blob="softmax"` does not change the initialization behavior. I am not sure if there exists any mechanism that stops users from changing the default parameters.

closed 2024-04-17 by srevandro