dusty-nv/jetson-inference #1736: imagenet: error while loading shared libraries: /usr/lib/aarch64-linux-gnu/libnvinfer.so.8: file too short

letpastgo, 2023-09-21:
Hi Mr/Mrs,
                 I  download this tutorial and run on my jetson nano develop kit. After I enter into docker mode, I type command, imagenet data/images/jellyfish.jpg data/images/test/jellyfish.jpg, error message: imagenet: error while loading shared libraries: /usr/lib/aarch64-linux-gnu/libnvinfer.so.8: file too short showed.
                I checked my jetson nano info:Software part of jetson-stats 4.2.3 - (c) 2023, Raffaello Bonghi
                                                               Model: NVIDIA Jetson Nano Developer Kit - Jetpack 4.6.4 [L4T 32.7.4]
                                                               NV Power Mode[1]: 5W
                                                               Serial Number: [XXX Show with: jetson_release -s XXX]
                                                               Hardware:
                                                                   - P-Number: p3448-0000
                                                                   - Module: NVIDIA Jetson Nano (4 GB ram)
                                                               Platform:
                                                                 - Distribution: Ubuntu 18.04 Bionic Beaver
                                                                 - Release: 4.9.337-tegra
                                                              jtop:
                                                                - Version: 4.2.3
                                                                - Service: Active
                                                             Libraries:
                                                                - CUDA: 10.2.300
                                                                - cuDNN: 8.2.1.32
                                                               - TensorRT: 8.2.1.8
                                                               - VPI: 1.2.3
                                                               - Vulkan: 1.2.70
                                                               - OpenCV: 4.1.1 - with CUDA: NO.
I still not have any idea about this. Could you give me some advise, thank you.

letpastgo, 2023-09-22:
Hi dusty-nv:
                  Thanks for your reply. I try command below, and get error message: ModuleNotFoundError: No module named 'tensorrt'. According to the system info, I should have installed tensorrt. Can you give me some advice, thank you.

dusty-nv, 2023-09-22:
OK gotcha.  If you run `python3 -c 'import tensorrt'` outside of container, does it work?

If it works outside of container but not inside, something got messed up with your nvidia-container-runtime and how it auto-mounts those files in.  Unfortunately the most reliable way to fix that is to flash a fresh SD card.

letpastgo, 2023-09-22:
Hi dusty-nv:
Thanks for your reply. The situation is just like what you say. I tried to reinstall container related packages with apt but it does not work. So I flash the same SD card and the problem is solved. Thanks for you help and time.

closed 2023-09-22 by letpastgo