dusty-nv/jetson-inference #1738: Docker "jetson-inference" on Jetson Nano and face_recognition package

peltiern, 2023-09-24:
Hi @dusty-nv 

For a personal robotics project, I use a Jetson Nano.
For the "Artificial vision" part and to avoid having to build lots of things, I am testing the "jetson-inference" Docker container. I want to create a Python server for object recognition and face recognition.
I successfully installed the container and was able to test the detectnet.py example.

Now I try to install the face-recognition package (https://pypi.org/project/face-recognition/) in the container but I get the following error : 
```
root@Wall-E:/jetson-inference# pip3 install face-recognition
Collecting face-recognition
  Using cached https://files.pythonhosted.org/packages/1e/95/f6c9330f54ab07bfa032bf3715c12455a381083125d8880c43cbe76bb3d0/face_recognition-1.3.0-py2.py3-none-any.whl
Collecting face-recognition-models>=0.3.0 (from face-recognition)
  Downloading https://files.pythonhosted.org/packages/cf/3b/4fd8c534f6c0d1b80ce0973d01331525538045084c73c153ee6df20224cf/face_recognition_models-0.3.0.tar.gz (100.1MB)
    100% |################################| 100.2MB 5.1kB/s 
Requirement already satisfied: Pillow in /usr/local/lib/python3.6/dist-packages/Pillow-9.0.1-py3.6-linux-aarch64.egg (from face-recognition)
Pillow requires Python '>=3.7' but the running Python is 3.6.9
```
However, a version of Pillow seems to have been found !!!
Can you help me, please.

Best regards,
Nicolas

Container configuration : 
```
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.7.4
[sudo] password for jetson: 
localuser:root being added to access control list
CONTAINER_IMAGE:  dustynv/jetson-inference:r32.7.1
```

jtop informations : 
[image]

dusty-nv, 2023-09-25:
Hi @peltiern, I'm not sure about face-recognition package (haven't tried that before) or what is going on with Pillow, but you could try `pip3 install 'Pillow<10'` and see if that helps

peltiern, 2023-10-01:
Hi @dusty-nv ,
It seems to work better with `pip3 install 'Pillow<9'`. 
I successfully installed face_recognition. But it overwrote the version of Dlib that I had built.
Now I have a problem further. I'll look in more detail.
Thanks

peltiern, 2023-10-08:
Hi @dusty-nv ,
When I restart my Jetson Nano and the docker container, the Pillow 9 package came back, even though I had Pillow 8 installed. How to keep packages installed after a reboot ? I am a newbie about Docker.

Best regards,
Nicolas

dusty-nv, 2023-10-08:
Hi @peltiern, you can use `docker commit` to save your manual changes to a new container tag:

https://docs.docker.com/engine/reference/commandline/commit/

or you can make/build your own Dockerfile using jetson-inference container as base, or modify jetson-inference dockerfile and re-build it.

peltiern, 2023-10-16:
Hi @dusty-nv,

When i use the default installation of the jetson-inference container (`sudo docker/run.sh`), I have no problem to install DLib and face_recognition packages in the container. CUDA is found during building DLib libraries (cf installation_with_run.txt file) and all the examples of jetson detections and face recognition work fine. And I manage to commit.

But when I create my own Dockerfile with dlib and face recognition installation instructions and i build it, CUDA is NOT found (cf build_from_own_dockerfile.txt).

Do you have any idea what is wrong?

Best regards,
Nicolas

My Dockerfile
```
FROM dustynv/jetson-inference:r32.7.1

ENV DEBIAN_FRONTEND=noninteractive
ENV SHELL /bin/bash

WORKDIR /jetson-inference

# Installation de DLIB
RUN apt-get install cmake && \
    apt-get install sed && \
    wget https://github.com/davisking/dlib/archive/refs/tags/v19.24.2.tar.gz && \
    tar xvf v19.24.2.tar.gz && \
    cd dlib-19.24.2/ && \
    sed -i 's,forward_algo = forward_best_algo;,//forward_algo = forward_best_algo;,g' dlib/cuda/cudnn_dlibapi.cpp && \
    mkdir build && \
    cd build && \
    cmake .. -DDLIB_USE_CUDA=1 -DUSE_AVX_INSTRUCTIONS=1 && \
    cmake --build . && \
    cd .. && \
    python3 setup.py install

# Installation de face_recognition
RUN pip3 install 'Pillow<9' && \
    pip3 install face_recognition
```

[installation_with_run.txt](https://github.com/dusty-nv/jetson-inference/files/12913258/installation_with_run.txt)
[build_from_own_dockerfile.txt](https://github.com/dusty-nv/jetson-inference/files/12913280/build_from_own_dockerfile.txt)

peltiern, 2023-10-21:
Hi @dusty-nv ,

It works well !!!
I managed to create a Docker image with DLib and face-recognition packages working with CUDA !

Thanks a lot.

Nicolas

peltiern, 2023-11-23:
Hi @dusty-nv ,

Here is the docker image including jetson-inference and face-recognition, if it helps anyone : 
https://hub.docker.com/r/peltiern/jetson-nano-computer-vision/tags

closed 2023-11-23 by peltiern