dusty-nv/jetson-inference #1739: how to increase the FPS on Peoplenet model?

saisatheesh5, 2023-09-26:
Hi All,

I have **Advantech jetson nano (MIC-710AI)**. I am using the PEOPLENET Model and  I have connected 2 cameras to that for live detection. Now I have received only 3FPS. For my requirement, I need at least 25 to 30 FPS. How can I increase my FPS. Please provide you suggestions to improve that. And Previous forum observation I have tried the below comments.

1) sudo nvpmodel -m 0 
2) sudo jetson_clocks

Above two commands I ran but after that also I don't have any improvement on that.

Thanks in Advance!

dusty-nv, 2023-09-26:
@saisatheesh5 what program are you using?  Which camera?   Are you able to stream one camera alone with video-viewer at > 3FPS?

saisatheesh5, 2023-09-27:
Hi dusty,

Thanks for the response and valuable time.

I am using detectnet program with peoplenet model. I am using Upcamera 2MP. it is made by intel.

Yeah, I tried with one camera alone with the same program still i received the same.

when I tried with SSD-mobilenet-V2 model, on that time i have received 14 to 16 FPS.

[up camera datasheet.pdf](https://github.com/dusty-nv/jetson-inference/files/12734619/up.camera.datasheet.pdf)

Thanks!

dusty-nv, 2023-09-27:
@saisatheesh5 my guess is that since Jetson Nano is unable to do INT8, and PeopleNet is a higher-resolution model that becomes fast using INT8 quantization, it is slower on FP16-only Nano.  You can try the other support TAO detectNet_v2 models and see if the performance scales.

saisatheesh5, 2023-09-27:
Hi dusty-nv,

Thanks for your response.

If I have installed the deepstream is any possible is there to increase the performance?

Please suggest any idea to detect only the human in a FOV using jetson nano.

My main purpose was a need to inspect a specific area if any humans are there or not. So please suggest which model I can use for that with Jetson Nano. I have tried SSD-mobile net-V2 with person class but most of the time it detects the robot as a human from various angles.

Thanks

saisatheesh5, 2023-09-29:
Hi @dusty-nv,

Thanks for your suggestion. I have found the more accurate Yolo model. I tested with my trail videos it's working perfectly so please help me to implement that on the jetson nano from Advantech(Mic-710ai).

I have mentioned the link below which I followed to use the Yolo human detection model.

[](https://github.com/deepakcrk/yolov5-crowdhuman)

Thanks!