dusty-nv/jetson-inference #1740: No Issue, just a question ;-)

Pepper-Inmoov, 2023-09-27:
Hello everyone. How could i send the data (names) of the detected objects to a python client? In my case the detectnet.py example.
I have already added a socket server to detectnet.py and a matching client. But i don't understand what i have to send from detectnet.py to the client. Unfortunately, i don't know anything about the whole server thing. Maybe someone can help me with my problem and make me a little smarter:-) The goal is to integrate object recognition into my Inmoov project.

dusty-nv, 2023-09-27:
@Pepper-Inmoov I'd probably just send it via sockets and JSON:   https://stackoverflow.com/questions/39817641/how-to-send-a-json-object-using-tcp-socket-in-python

JSON is like a serialized dict to text

Pepper-Inmoov, 2023-09-27:
@dusty-nv Thank you for your quick response. I now have createt a json socket server.py and integrated a json socket client into detectnet.py. If i now send some text as data, the server receives the text out of detectnet.py. But what and how do i now have to send from detectnet.py to the server so i receive the names of the detected objects there?

dusty-nv, 2023-09-27:
@Pepper-Inmoov the members of the `jetson.inference.detectNet.Detection` objects which get returned from net.Detect() are listed on this page:

https://rawgit.com/dusty-nv/jetson-inference/master/docs/html/python/jetson.inference.html#detectNet

So basically something like this:

```bash
detections = net.Detect(img)

for detection in detection:
   class_name = net.GetClassLabel(detection.ClassID)
```

Pepper-Inmoov, 2023-09-28:
@dusty-nv figured it out. :-) Thank you again for the help. Now I just have to figure out how to ensure that detectnet.py doesn't shut down after transferring the data to the server.