dusty-nv/jetson-inference #1743: Jetson Orion TAO emotionnet

theDVSman, 2023-10-05:
Morning Dusty,
Great tutorials. Totally appreciate your wisdom on all the Jetson topics. Looking for some advice. In regards to your TAO section / models tutorials. How hard would it be to adapt the Emotionnet data set ?  I am looking for a way to capture emotion and I wouldn't consider myself a senior dev at this point.  I have gone through alot of your tutorials and they definitely help.  Any advice you can share would be appreciated.

Cheers
Shane

dusty-nv, 2023-10-05:
Hi @theDVSman, from the model card (https://catalog.ngc.nvidia.com/orgs/nvidia/teams/tao/models/emotionnet) it appears that it takes in facial landmarks (not pixels), so you would need the landmarks model too.   I would probably just run this one in DeepStream which already supports all the TAO models in leui of adding all that stuff to jetson-inference.

theDVSman, 2023-10-05:
HI @dusty-nv ok thanks for the advice.  Just curious, if I wanted to go the pixel route, could I just train my own custom model on a data set of pictures expressing emotion ?

dusty-nv, 2023-10-05:
@theDVSman yes I have seen people just train a simple ResNet18/ResNet50 classifier on images of people's faces with different emotions, and it worked fine for limited use-cases

theDVSman, 2023-10-06:
Ok thanks Dusty.  Appreciate your feedback 👍