dusty-nv/jetson-inference #1744: Unable to install jetson utils or jetson inference

Nicholas-B1, 2023-10-05:
I have followed the steps to installing both libraries but when I run them on the nano in visual studio it doesn't seem to work is there updated steps for installing these python libraries?

dusty-nv, 2023-10-05:
Hi @Nicholas-B1, what is the issue you are having?   Is it related to Visual Studio, or if you SSH into your board (or use the desktop if you have display connected), are you able to run the jetson-inference applications?

Nicholas-B1, 2023-10-05:
It seems my jetson_utils sometimes works but the jetson_inference always gives import errors. I followed online tutorials to install using the cmake files but it seems like it did not work.

Nicholas-B1, 2023-10-05:
When I check pip3 list only utils is there not inference.

dusty-nv, 2023-10-05:
Hard to debug without seeing the errors or the build logs.  Try running the docker container if you haven't already and see if those work - `cd /path/to/your/jetson-inference && docker/run.sh`

Nicholas-B1, 2023-10-05:
I will run that again also in visual studio code it seems like it wont accept some jetson utils commands like gstCamera

Nicholas-B1, 2023-10-05:
Is there something I should send to make it easier to see any issues?

dusty-nv, 2023-10-06:
@Nicholas-B1 I haven't used VS Code - is it using a virtual environment or something?  Can you try running these steps and send me the terminal log?

```bash
cd /path/to/your/jetson-inference
cd build
cmake ../
make
sudo make install
sudo ldconfig
python3 -c 'import jetson_utils'
python3 -c 'import jetson_inference'
```

Nicholas-B1, 2023-10-12:
[image]