dusty-nv/jetson-inference #1748: invalid configuration argument (error 9) when running custom detection model

ryuchanhoe, 2023-10-24:
Hi I am trying to run custom detection model I have ran this model on different platform so I assume the model is just fine. (It is working properly on Orin AGX but I am trying to run on Xavier)

<launch>

	

	
	
	<arg name="model_name" default=""/>
	
	<arg name="model_path" default="/home/usrg/catkin_ws/src/ros_deep_learning/model/object/pruned.engine"/>
	<arg name="prototxt_path" default=""/>
	<arg name="class_labels_path" default="/home/usrg/catkin_ws/src/ros_deep_learning/model/object/labels.txt"/>
	<arg name="input_blob" default="input_1"/>
	<arg name="output_cvg" default="output_cov/Sigmoid"/>
	<arg name="output_bbox" default="output_bbox/BiasAdd"/>
	<arg name="overlay_flags" default="box,labels,conf"/>
	<arg name="mean_pixel_value" default="0.0"/>
	<arg name="threshold" default="0.4"/>

	<node pkg="ros_deep_learning" type="detectnet" name="detectnet" output="screen">
		<remap from="/detectnet/image_in" to="/front"/>
		<param name="model_name" value="$(arg model_name)"/>
		<param name="model_path" value="$(arg model_path)"/>
		<param name="prototxt_path" value="$(arg prototxt_path)"/>
		<param name="class_labels_path" value="$(arg class_labels_path)"/>
		<param name="input_blob" value="$(arg input_blob)"/>
		<param name="output_cvg" value="$(arg output_cvg)"/>
		<param name="output_bbox" value="$(arg output_bbox)"/>
		<param name="overlay_flags" value="$(arg overlay_flags)"/>
		<param name="mean_pixel_value" value="$(arg mean_pixel_value)"/>
		<param name="threshold" value="$(arg threshold)"/>
	</node>

	
	
	
	

	
		
		
		
		
	

</launch>

Above is my launch file, and below is the inference result.

[ INFO] [1687951300.813664192]: object 1336 bounding box (986.550903,
75.449356)  (996.284973, 68.318001)  w=9.734070  h=-7.131355
[ INFO] [1687951300.813730048]: object 1337 class #3 (road_sign)  confidence=0.604986
[ INFO] [1687951300.813781120]: object 1337 bounding box (1017.987976, 74.440765)  (1028.644775, 67.842247)  w=10.656799  h=-6.598518
[cuda]      invalid configuration argument (error 9) (hex 0x09)
[cuda]      /home/usrg/jetson-inference/c/detectNet.cpp:1074
[cuda]      invalid configuration argument (error 9) (hex 0x09)
[cuda]      /home/usrg/jetson-inference/c/detectNet.cpp:1074
 
As you can see, it does detect but it won't be able to generate overlayed image with outputting that errors?
This error won't shutdown the detectnet but still generating that error and seems to be slowing down the hz?

Thanks for the help in advance!

dusty-nv, 2023-10-24:
> [ INFO] [1687951300.813781120]: object 1337 bounding box (1017.987976, 74.440765) (1028.644775, 67.842247) w=10.656799 h=-6.598518

Hi @ryuchanhoe, it's odd that the bounding box is malformed (y2>y1), this shouldn't happen.  Did you delete/re-generate the TRT engine after changing platforms?   Does it work through detectnet/detectnet.py as a test?

ryuchanhoe, 2023-10-25:
Hi @dusty-nv, thanks for the reply!

Yeah, I had to regenerate the TRT engine using tao-converter because the engine converted on Orin would not run on Xavier, it exits with errors. 

Command (I did pretty much the same thing when converting in Orin too):
./tao-converter -d 3,544,960 -k tlt_encode -o output_cov/Sigmoid,output_bbox/BiasAdd resnet18_dashcamnet_pruned.etlt

DashcamNet Acquired:
https://catalog.ngc.nvidia.com/orgs/nvidia/teams/tao/models/dashcamnet/files?version=pruned_v1.0.3

I had one suspicious caution while converting the engine, but I should not bother because it would output an engine anyway?
caution:
[INFO] Some tactics do not have sufficient workspace memory to run. Increasing workspace size may increase performance, please check verbose output.

I believe if there is something wrong with the engine file, then it would have the be converting process?
Somethings should be modified when converting and inferring the model in Xavier than Orin?

I am trying different approaches, but I am not able to solve the problem at the moment.
Thanks for the help in advance.