dusty-nv/jetson-inference #1750: gstEncoder failed to create pipeline

Yarmish-P, 2023-10-29:
Hello, I've ran example from article Locating Objects with DetectNet.

./detectnet pedestrians.mp4 images/test/pedestrians_ssd.mp4

after that I had an error:

[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! x264enc name=encoder bitrate=4000 speed-preset=ultrafast tune=zerolatency ! video/x-h264 ! h264parse ! qtmux ! filesink location=images/test/pedestrians_ssd.mp4
[gstreamer] gstEncoder -- failed to create pipeline
[gstreamer]    (no element "x264enc")
[gstreamer] failed to create encoder pipeline
[gstreamer] gstEncoder -- failed to create encoder engine
detectnet:  failed to create output stream.

My environment Windows 11, WSL2, Ubuntu 22.04

What possible reason of this error?
Thank you!

dusty-nv, 2023-10-29:
Hi @Yarmish-P, it's been awhile since I ran this under WSL2 and the primary platform this project is supported on is Jetson (arm64) - but I think you need to install `gstreamer1.0-libav gstreamer1.0-plugins-ugly`

https://stackoverflow.com/a/74290514

dusty-nv, 2023-10-29:
BTW before sinking too much time into this, I would just test detectnet on a single image first, because there could be a number of other things wrong for x86 platform, like TensorRT version, CUDA memory synchronization, ect.

Yarmish-P, 2023-10-29:
detectnet on a single image works well. Your piss of advise was useful for me, gstEncoder error was fixed.  After that I faced with next error:

5 objects detected

detected obj 0  class #1 (person)  confidence=0.702637
bounding box 0  (25.22, 280.28)  (92.81, 451.14)  w=67.59  h=170.86

detected obj 1  class #64 (potted plant)  confidence=0.674316
bounding box 1  (181.76, 208.83)  (280.78, 288.46)  w=99.02  h=79.63

detected obj 2  class #1 (person)  confidence=0.648926
bounding box 2  (288.52, 164.27)  (342.89, 293.20)  w=54.38  h=128.94

detected obj 3  class #1 (person)  confidence=0.830078
bounding box 3  (577.97, 82.73)  (616.41, 198.02)  w=38.44  h=115.29

detected obj 4  class #1 (person)  confidence=0.687988
bounding box 4  (656.72, 58.27)  (696.09, 137.90)  w=39.38  h=79.63

[cuda]   cudaGraphicsGLRegisterBuffer(&interop, allocDMA(type), cudaGraphicsRegisterFlagsFromGL(flags))
[cuda]      OS call failed or operation not supported on this OS (error 304) (hex 0x130)
[cuda]      /mnt/d/Projects/Machine_Learning/Jetson/jetson-inference/utils/display/glTexture.cpp:360
[cuda]   cudaGetLastError()
[cuda]      OS call failed or operation not supported on this OS (error 304) (hex 0x130)
[cuda]      /mnt/d/Projects/Machine_Learning/Jetson/jetson-inference/utils/cuda/cudaYUV-YV12.cu:257
[cuda]   cudaRGBToI420((uchar3*)input, output, width, height)
[cuda]      OS call failed or operation not supported on this OS (error 304) (hex 0x130)
[cuda]      /mnt/d/Projects/Machine_Learning/Jetson/jetson-inference/utils/cuda/cudaColorspace.cpp:128
[cuda]   cudaConvertColor(image, format, nextYUV, IMAGE_I420, width, height)
[cuda]      OS call failed or operation not supported on this OS (error 304) (hex 0x130)
[cuda]      /mnt/d/Projects/Machine_Learning/Jetson/jetson-inference/utils/codec/gstEncoder.cpp:703
[gstreamer] gstEncoder::Render() -- unsupported image format (rgb8)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f

I searched about this error:

[cuda]   cudaGraphicsGLRegisterBuffer(&interop, allocDMA(type), cudaGraphicsRegisterFlagsFromGL(flags))
[cuda]      OS call failed or operation not supported on this OS (error 304) (hex 0x130)

The reason can be connected with integration issue between CUDA and OpenGL. I found that this error can be fixed if video out puts throw DisplayPort. Not I use laptop and have no ability connect monitor with DisplayPort. I suggest that reason of problem consists in laptop is equipped with two graphic cards: integrated and discrete. I set up my RTX 4060 as default GPU:

export MESA_D3D12_DEFAULT_ADAPTER_NAME=NVIDIA

But error was not fixed.
Do you have some ideas?

dusty-nv, 2023-10-30:
I'm not sure I ever got OpenGL working on WSL, I was trying on Windows 10.  I think the OpenGL display is unlikely to work under it.