dusty-nv/jetson-inference #1751: OpenCV Compilation Error When Building Docker Conatiner on x86

mohamad1998630, 2023-10-31:
Hi there, 
I am trying to build the docker container on an x86 architecture but I am facing an error with OpenCV. Here is the error message:
`38.17 -- CUDA version: 11.7
38.18 CMake Error at /usr/lib/cmake/opencv4/OpenCVConfig.cmake:111 (message):
38.18   OpenCV static library was compiled with CUDA 11.6 support.  Please, use the
38.18   same version or rebuild OpenCV with CUDA 11.7
38.18 Call Stack (most recent call first):
38.18   CMakeLists.txt:90 (find_package)
`

I tried to configure the cuda version that opencv compiles with as suggested by the error message but could not figure how to do it. Note that both these versions are different from my host machine cuda version which is 12.0. 

Could anyone support on this matter?

dusty-nv, 2023-10-31:
Are you installing your own build of OpenCV instead of the one that is installed for JetPack+aarch64 here?

https://github.com/dusty-nv/jetson-inference/blob/44c7661dfbb97018fbc4651fa4608d5491d3f39e/Dockerfile#L89

You can just skip doing that on x86 and install normal version of OpenCV

mohamad1998630, 2023-11-01:
Thanks for your quick replay @dusty-nv 

I did not change this line so I guess I am trying to install a version that is built for JetPack and aarch64 as you said. 
I will try to install a normal opencv version as you suggested and see how it goes. 

I really appreciate your help

mohamad1998630, 2023-11-06:
I tried compiling a normal version of opencv and it worked fine. Thanks for the support @dusty-nv. I will close the issue now