dusty-nv/jetson-inference #1752: Linking errors with libcupva_host.so.2.3

kanishkanarch, 2023-10-31:
I am trying to build [Jetson-Inference](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md) on my Jetson Orin Nano 8 GB board. Upon `sudo make`, I face the following errors:
```
[ 67%] Linking CXX executable ../../aarch64/bin/imagenet
/usr/bin/ld: /opt/nvidia/cupva-2.3/lib/aarch64-linux-gnu/libcupva_host.so.2.3: undefined reference to `PvaProgramSetDMADescriptorsV2'
/usr/bin/ld: /opt/nvidia/cupva-2.3/lib/aarch64-linux-gnu/libcupva_host.so.2.3: undefined reference to `PvaProgramSetPointerValueEx'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/imagenet/CMakeFiles/imagenet.dir/build.make:97: aarch64/bin/imagenet] Error 1
make[1]: *** [CMakeFiles/Makefile2:469: examples/imagenet/CMakeFiles/imagenet.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
My [system](https://github.com/jetsonhacks/jetsonUtilities) is following:
```
NVIDIA NVIDIA Orin Nano - AVerMedia D131
 L4T 35.3.1 [ JetPack UNKNOWN ]
   Ubuntu 20.04.6 LTS
   Kernel Version: 5.10.104-tegra
 CUDA 11.4.315
   CUDA Architecture: 8.7
 OpenCV version: 4.5.4
   OpenCV Cuda: NO
 CUDNN: 8.6.0.166
 TensorRT: 8.5.2.2
 Vision Works: NOT_INSTALLED
 VPI: 2.3.9
 Vulcan: 1.3.204
```
Please help me build this awesome project on my awesome board
Thanks

dusty-nv, 2023-10-31:
Hmm, I'm not sure why you are getting that error with PVA/VPI - try commenting out lines 106-114 of CMakeLists.txt:

https://github.com/dusty-nv/jetson-inference/blob/44c7661dfbb97018fbc4651fa4608d5491d3f39e/CMakeLists.txt#L106

or try the jetson-inference docker container:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md

kanishkanarch, 2023-11-01:
Thanks, that worked! I can see demos of different models now.

closed 2023-11-01 by kanishkanarch