dusty-nv/jetson-inference #1761: ImportError: jetson_utils_python.so: cannot open shared object file: No such file or directory

sh3rlock14, 2023-11-29:
Hi 👋🏻 
I'm having trouble in building from source jetson-inference project.
I've followed the steps as in: [Build the Project from Source](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md#building-the-project-from-source). 

Then when I try to launch `./imagenet.py images/orange_0.jpg images/test/output_0.jpg`, I get the error:
```python
Traceback (most recent call last):
  File "./imagenet.py", line 27, in <module>
    from jetson_inference import imageNet
  File "/usr/lib/python3.8/dist-packages/jetson_inference/__init__.py", line 8, in <module>
    from jetson_inference_python import *
ImportError: jetson_utils_python.so: cannot open shared object file: No such file or directory
```
I read all the other issues and answers related to this problem or similar, but with no success.

Here what's inside `usr/lib/python3.8/dist-packages` after building:
[image].

I tried `sudo ldconfig` both in `jetson-inference/build/aarch64/bin` and `jetson-inference/build` but nothing worked for me.

Do you have any idea how to solve this problem?

Thanks in advance for the help!

dusty-nv, 2023-11-29:
Hi @sh3rlock14, if you run this from your home directory (or any other directory), does it work, or make the same error?

```python
python3 -c 'import jetson_utils'
python3 -c 'import jetson_inference'
```

What does `ldd /usr/lib/python3.8/dist-packages/jetson_inference_python.so` show for you?

```
$ ldd /usr/lib/python3.8/dist-packages/jetson_inference_python.so
        linux-vdso.so.1 (0x0000ffff92811000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffff92692000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000ffff9267e000)
        librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000ffff92666000)
        libjetson-inference.so => /usr/local/lib/libjetson-inference.so (0x0000ffff92328000)
        libpython3.8.so.1.0 => /lib/aarch64-linux-gnu/libpython3.8.so.1.0 (0x0000ffff91dd3000)
        jetson_utils_python.so => not found
        libjetson-utils.so => /usr/local/lib/libjetson-utils.so (0x0000ffff91836000)
        libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000ffff91651000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000ffff9162d000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff914ba000)
        /lib/ld-linux-aarch64.so.1 (0x0000ffff927e1000)
        libnvinfer.so.8 => /lib/aarch64-linux-gnu/libnvinfer.so.8 (0x0000ffff84361000)
        libnvinfer_plugin.so.8 => /lib/aarch64-linux-gnu/libnvinfer_plugin.so.8 (0x0000ffff82f3e000)
        libnvparsers.so.8 => /lib/aarch64-linux-gnu/libnvparsers.so.8 (0x0000ffff82bcd000)
        libnvonnxparser.so.8 => /lib/aarch64-linux-gnu/libnvonnxparser.so.8 (0x0000ffff82900000)
        libnvvpi.so.2 => /opt/nvidia/vpi2/lib/aarch64-linux-gnu/libnvvpi.so.2 (0x0000ffff795bc000)
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000ffff79511000)
        libexpat.so.1 => /lib/aarch64-linux-gnu/libexpat.so.1 (0x0000ffff794da000)
        libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000ffff794b0000)
        libutil.so.1 => /lib/aarch64-linux-gnu/libutil.so.1 (0x0000ffff7949c000)
        libGL.so.1 => /lib/aarch64-linux-gnu/libGL.so.1 (0x0000ffff793a5000)
        libGLU.so.1 => /lib/aarch64-linux-gnu/libGLU.so.1 (0x0000ffff79332000)
        libGLEW.so.2.1 => /lib/aarch64-linux-gnu/libGLEW.so.2.1 (0x0000ffff79272000)
        libgstreamer-1.0.so.0 => /lib/aarch64-linux-gnu/libgstreamer-1.0.so.0 (0x0000ffff79116000)
        libgstapp-1.0.so.0 => /lib/aarch64-linux-gnu/libgstapp-1.0.so.0 (0x0000ffff790f6000)
        libgstpbutils-1.0.so.0 => /lib/aarch64-linux-gnu/libgstpbutils-1.0.so.0 (0x0000ffff790aa000)
        libgstwebrtc-1.0.so.0 => /lib/aarch64-linux-gnu/libgstwebrtc-1.0.so.0 (0x0000ffff79090000)
        libgstsdp-1.0.so.0 => /lib/aarch64-linux-gnu/libgstsdp-1.0.so.0 (0x0000ffff79067000)
        libgstrtspserver-1.0.so.0 => /lib/aarch64-linux-gnu/libgstrtspserver-1.0.so.0 (0x0000ffff78ff4000)
        libjson-glib-1.0.so.0 => /lib/aarch64-linux-gnu/libjson-glib-1.0.so.0 (0x0000ffff78fbb000)
        libsoup-2.4.so.1 => /lib/aarch64-linux-gnu/libsoup-2.4.so.1 (0x0000ffff78f11000)
        libnppicc.so.11 => /usr/local/cuda-11.4/lib64/libnppicc.so.11 (0x0000ffff788a6000)
        libnvdla_compiler.so => /usr/lib/aarch64-linux-gnu/tegra/libnvdla_compiler.so (0x0000ffff7816b000)
        libcudla.so.1 => /usr/local/cuda-11.4/lib64/libcudla.so.1 (0x0000ffff7812e000)
        libcublas.so.11 => /usr/local/cuda-11.4/lib64/libcublas.so.11 (0x0000ffff6e053000)
        libcublasLt.so.11 => /usr/local/cuda-11.4/lib64/libcublasLt.so.11 (0x0000ffff56ab2000)
        libcudnn.so.8 => /lib/aarch64-linux-gnu/libcudnn.so.8 (0x0000ffff56a7b000)
        libnvsocsys.so => /usr/lib/aarch64-linux-gnu/tegra/libnvsocsys.so (0x0000ffff56a67000)
        libnvvic.so => /usr/lib/aarch64-linux-gnu/tegra/libnvvic.so (0x0000ffff56a3e000)
        libnvos.so => /usr/lib/aarch64-linux-gnu/tegra/libnvos.so (0x0000ffff56a1c000)
        libEGL.so.1 => /lib/aarch64-linux-gnu/libEGL.so.1 (0x0000ffff569f8000)
        libEGL_nvidia.so.0 => /usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0 (0x0000ffff568b6000)
        libnvbufsurface.so.1.0.0 => /usr/lib/aarch64-linux-gnu/tegra/libnvbufsurface.so.1.0.0 (0x0000ffff567fb000)
        libnvmedia.so => /usr/lib/aarch64-linux-gnu/tegra/libnvmedia.so (0x0000ffff56783000)
        libnvrm_mem.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_mem.so (0x0000ffff5676b000)
        libnvrm_host1x.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_host1x.so (0x0000ffff56748000)
        libnvrm_surface.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_surface.so (0x0000ffff56712000)
        libgomp.so.1 => /lib/aarch64-linux-gnu/libgomp.so.1 (0x0000ffff566c4000)
        libcupva_host.so.2.3 => /opt/nvidia/cupva-2.3/lib/aarch64-linux-gnu/libcupva_host.so.2.3 (0x0000ffff56661000)
        libcupva_host_utils.so.2.3 => /opt/nvidia/cupva-2.3/lib/aarch64-linux-gnu/libcupva_host_utils.so.2.3 (0x0000ffff56633000)
        libGLdispatch.so.0 => /lib/aarch64-linux-gnu/libGLdispatch.so.0 (0x0000ffff564a6000)
        libGLX.so.0 => /lib/aarch64-linux-gnu/libGLX.so.0 (0x0000ffff56464000)
        libgobject-2.0.so.0 => /lib/aarch64-linux-gnu/libgobject-2.0.so.0 (0x0000ffff563f2000)
        libgmodule-2.0.so.0 => /lib/aarch64-linux-gnu/libgmodule-2.0.so.0 (0x0000ffff563de000)
        libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000ffff562a3000)
        libgstbase-1.0.so.0 => /lib/aarch64-linux-gnu/libgstbase-1.0.so.0 (0x0000ffff5621c000)
[... cut ...]
        libkeyutils.so.1 => /lib/aarch64-linux-gnu/libkeyutils.so.1 (0x0000ffff52d28000)
        libicudata.so.66 => /lib/aarch64-linux-gnu/libicudata.so.66 (0x0000ffff51259000)
        libXau.so.6 => /lib/aarch64-linux-gnu/libXau.so.6 (0x0000ffff51245000)
        libXdmcp.so.6 => /lib/aarch64-linux-gnu/libXdmcp.so.6 (0x0000ffff5122f000)
        libbsd.so.0 => /lib/aarch64-linux-gnu/libbsd.so.0 (0x0000ffff51206000)
```

Also, for the time being you can try the jetson-inference docker container to get up & running awhile.

dusty-nv, 2023-11-29:
Also, maybe try adding `/usr/lib/python3.8/dist-packages` to LD_LIBRARY_PATH

sh3rlock14, 2023-11-30:
```python
python3 -c 'import jetson_utils' # this runs fine
python3 -c 'import jetson_inference' # this throw the error
```
The first runs fine, the second one throws the error I mentioned in my first post.

This is the output from `ldd` command:

```bash
	linux-vdso.so.1 (0x0000ffff9e8a5000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffff9e724000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000ffff9e710000)
	librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000ffff9e6f8000)
	libjetson-inference.so => /usr/local/lib/libjetson-inference.so (0x0000ffff9e3b9000)
	libpython3.8.so.1.0 => /lib/aarch64-linux-gnu/libpython3.8.so.1.0 (0x0000ffff9de64000)
	jetson_utils_python.so => not found
	libjetson-utils.so => /usr/local/lib/libjetson-utils.so (0x0000ffff9d8c6000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000ffff9d6e1000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000ffff9d6bd000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff9d54a000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff9e875000)
	libnvinfer.so.8 => /lib/aarch64-linux-gnu/libnvinfer.so.8 (0x0000ffff903f1000)
	libnvinfer_plugin.so.8 => /lib/aarch64-linux-gnu/libnvinfer_plugin.so.8 (0x0000ffff8efce000)
	libnvparsers.so.8 => /lib/aarch64-linux-gnu/libnvparsers.so.8 (0x0000ffff8ec5d000)
	libnvonnxparser.so.8 => /lib/aarch64-linux-gnu/libnvonnxparser.so.8 (0x0000ffff8e990000)
	libnvvpi.so.2 => /opt/nvidia/vpi2/lib/aarch64-linux-gnu/libnvvpi.so.2 (0x0000ffff855cd000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000ffff85522000)
	libexpat.so.1 => /lib/aarch64-linux-gnu/libexpat.so.1 (0x0000ffff854eb000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000ffff854c1000)
	libutil.so.1 => /lib/aarch64-linux-gnu/libutil.so.1 (0x0000ffff854ad000)
	libGL.so.1 => /lib/aarch64-linux-gnu/libGL.so.1 (0x0000ffff853b6000)
	libGLU.so.1 => /lib/aarch64-linux-gnu/libGLU.so.1 (0x0000ffff85343000)
	libGLEW.so.2.1 => /lib/aarch64-linux-gnu/libGLEW.so.2.1 (0x0000ffff85283000)
	libgstreamer-1.0.so.0 => /lib/aarch64-linux-gnu/libgstreamer-1.0.so.0 (0x0000ffff85127000)
	libgstapp-1.0.so.0 => /lib/aarch64-linux-gnu/libgstapp-1.0.so.0 (0x0000ffff85107000)
	libgstpbutils-1.0.so.0 => /lib/aarch64-linux-gnu/libgstpbutils-1.0.so.0 (0x0000ffff850bb000)
	libgstwebrtc-1.0.so.0 => /lib/aarch64-linux-gnu/libgstwebrtc-1.0.so.0 (0x0000ffff850a1000)
	libgstsdp-1.0.so.0 => /lib/aarch64-linux-gnu/libgstsdp-1.0.so.0 (0x0000ffff85078000)
	libgstrtspserver-1.0.so.0 => /lib/aarch64-linux-gnu/libgstrtspserver-1.0.so.0 (0x0000ffff85005000)
	libjson-glib-1.0.so.0 => /lib/aarch64-linux-gnu/libjson-glib-1.0.so.0 (0x0000ffff84fcc000)
	libsoup-2.4.so.1 => /lib/aarch64-linux-gnu/libsoup-2.4.so.1 (0x0000ffff84f22000)
	libnppicc.so.11 => /usr/local/cuda-11.4/lib64/libnppicc.so.11 (0x0000ffff848b7000)
	libnvdla_compiler.so => /usr/lib/aarch64-linux-gnu/tegra/libnvdla_compiler.so (0x0000ffff8417c000)
	libcudla.so.1 => /usr/local/cuda-11.4/lib64/libcudla.so.1 (0x0000ffff8413f000)
	libcublas.so.11 => /usr/local/cuda-11.4/lib64/libcublas.so.11 (0x0000ffff7a064000)
	libcublasLt.so.11 => /usr/local/cuda-11.4/lib64/libcublasLt.so.11 (0x0000ffff62ac3000)
	libcudnn.so.8 => /lib/aarch64-linux-gnu/libcudnn.so.8 (0x0000ffff62a8c000)
	libnvsocsys.so => /usr/lib/aarch64-linux-gnu/tegra/libnvsocsys.so (0x0000ffff62a78000)
	libnvvic.so => /usr/lib/aarch64-linux-gnu/tegra/libnvvic.so (0x0000ffff62a4f000)
	libnvos.so => /usr/lib/aarch64-linux-gnu/tegra/libnvos.so (0x0000ffff62a2d000)
	libEGL.so.1 => /lib/aarch64-linux-gnu/libEGL.so.1 (0x0000ffff62a09000)
	libEGL_nvidia.so.0 => /usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0 (0x0000ffff628c7000)
	libnvbufsurface.so.1.0.0 => /usr/lib/aarch64-linux-gnu/tegra/libnvbufsurface.so.1.0.0 (0x0000ffff6280c000)
	libnvmedia.so => /usr/lib/aarch64-linux-gnu/tegra/libnvmedia.so (0x0000ffff62794000)
	libnvrm_mem.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_mem.so (0x0000ffff6277c000)
	libnvrm_host1x.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_host1x.so (0x0000ffff62759000)
	libnvrm_surface.so => /usr/lib/aarch64-linux-gnu/tegra/libnvrm_surface.so (0x0000ffff62723000)
	libgomp.so.1 => /lib/aarch64-linux-gnu/libgomp.so.1 (0x0000ffff626d5000)
	libcupva_host.so.2.3 => /opt/nvidia/cupva-2.3/lib/aarch64-linux-gnu/libcupva_host.so.2.3 (0x0000ffff62672000)
	libcupva_host_utils.so.2.3 => /opt/nvidia/cupva-2.3/lib/aarch64-linux-gnu/libcupva_host_utils.so.2.3 (0x0000ffff62644000)
	libGLdispatch.so.0 => /lib/aarch64-linux-gnu/libGLdispatch.so.0 (0x0000ffff624b7000)
	libGLX.so.0 => /lib/aarch64-linux-gnu/libGLX.so.0 (0x0000ffff62475000)
	libgobject-2.0.so.0 => /lib/aarch64-linux-gnu/libgobject-2.0.so.0 (0x0000ffff62403000)
	libgmodule-2.0.so.0 => /lib/aarch64-linux-gnu/libgmodule-2.0.so.0 (0x0000ffff623ef000)
	libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000ffff622b4000)
	libgstbase-1.0.so.0 => /lib/aarch64-linux-gnu/libgstbase-1.0.so.0 (0x0000ffff6222d000)
	libgstvideo-1.0.so.0 => /lib/aarch64-linux-gnu/libgstvideo-1.0.so.0 (0x0000ffff62178000)
	libgstaudio-1.0.so.0 => /lib/aarch64-linux-gnu/libgstaudio-1.0.so.0 (0x0000ffff620f4000)
	libgsttag-1.0.so.0 => /lib/aarch64-linux-gnu/libgsttag-1.0.so.0 (0x0000ffff620a8000)
	libgstrtp-1.0.so.0 => /lib/aarch64-linux-gnu/libgstrtp-1.0.so.0 (0x0000ffff62072000)
[... 48 lines cut ...]
	libpcre2-8.so.0 => /lib/aarch64-linux-gnu/libpcre2-8.so.0 (0x0000ffff5ed51000)
	libkeyutils.so.1 => /lib/aarch64-linux-gnu/libkeyutils.so.1 (0x0000ffff5ed3a000)
[... cut ...]
```python
python3 -c 'import jetson_utils'
python3 -c 'import jetson_inference'
```

and both ran fine.

Yet, launching `./imagenet.py images/orange_0.jpg images/test/output_0.jpg` raises the error: 

```python
Traceback (most recent call last):
  File "./imagenet.py", line 28, in <module>
    from jetson_utils import videoSource, videoOutput, cudaFont, Log
ImportError: cannot import name 'videoSource' from 'jetson_utils' (/home/phygital/.local/lib/python3.8/site-packages/jetson_utils/__init__.py)
```

dusty-nv, 2023-11-30:
> ```python
> ImportError: cannot import name 'videoSource' from 'jetson_utils' (/home/phygital/.local/lib/python3.8/site-packages/jetson_utils/__init__.py)
> ```

Interesting, okay...getting closer.   Can you try this next:

```
python3 -c 'import jetson_utils; print(dir(jetson_utils))'
```

sh3rlock14, 2023-11-30:
Some updates:
once I got the `videoSource` error I looked for similar problems and I found this issue https://github.com/dusty-nv/jetson-inference/issues/1761. So I updated the `LD_LIBRARY_PATH` and I got to launch `imagenet.py` w/o `Errors` but... it looped endlessly.
I wonder why, since I launched it from the `/bin` folder, with the arguments as defined in the guide (checked the existance of both the input image and the output folder).
Anyhow, I changed the arguments a little bit by simply prepending `./` to their paths and it worked like a charm.

Eventually, the output images was correctly generated. 

Below a more detailed guide on what I did to run the script and solve for the errors I encountered. As you can see I have printed the locations of the modules (utils and inference) loaded by python3 and they are differen: is it a problem?
According to your experience is it fine to edit the bash profile to make permanent changes to `LD_LIBRARY_PATH`?

## Troubleshooting Guide

Launching the script may fail for several reasons. The first error I got is:

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/dist-packages/jetson_inference/__init__.py", line 8, in <module>
    from jetson_inference_python import *
ImportError: jetson_utils_python.so: cannot open shared object file: No such file or directory
```

As suggested by @dusty_nv, entering a Python Dev Environemnt and running:

```python
>>> import jetson_inference
```

raised the `ImportError` mentioned above...

and:

```python
>>> import jetson_utils
>>> print(jetson_utils)
```

prints:

```python
<module 'jetson_utils' from '/home/phygital/.local/lib/python3.8/site-packages/jetson_utils/__init__.py'>
```

while I expected it to load the one located in `/usr/lib/python3.8/dist-packages/`.

Then I updated the `LD_LIBRARY_PATH` variable by exporting the path to: `usr/lib/python3.8/dist-packages/`:

```bash
echo $LD_BINARY_PATH # /usr/local/cuda-11.4/lib64:/usr/local/cuda-11.4/lib64
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/python3.8/dist-packages/
echo $LD_BINARY_PATH # /usr/local/cuda-11.4/lib64:/usr/local/cuda-11.4/lib64:/usr/lib/python3.8/dist-packages/
```

and then, executing again:

```python
>>> import jetson_utils
>>> import jetson_inference
>>> print(jetson_utils)
>>> print(jetson_inference)
```

I had no problem with the import, and the module are located at:

```python
<module 'jetson_utils' from '/home/user/.local/lib/python3.8/site-packages/jetson_utils/__init__.py'>
<module 'jetson_inference' from '/usr/lib/python3.8/dist-packages/jetson_inference/__init__.py'>
```

> Note: I did not update `bash.profile` or similar, but just worked locally in the open terminal, so to be able to revert back all the edits. If these changes are fine they can be made permanent.

Make sure nothing has been overwritten when appending the new path, otherwise some other dynamic libraries cannot be loaded.

After that, trying to launch the detection script with the following command from `jetson-inference/build/aarch64/bin`: 

```bash
./imagenet.py images/orange_0.jpg images/test/output_0.jpg  # (default network is googlenet)
```

I incurred in a sitution where the script looped indefinitely w/o producing the desidered output.

A solution to this problem was to either:
1. add refer to both `input` and `output` with absolute paths, *i.e.:* `home/phygital/HelloAIWorld/jetson-inference/build/aarch64/bin/images/orange_0.jpg` and `home/phygital/HelloAIWorld/jetson-inference/build/aarch64/bin/images/test/output_0.jpg`
2. add `./` to both `input` and `output` paths, *i.e.:* `./images/orange_0.jpg` and `./images/test/output_0.jpg`

sh3rlock14, 2023-11-30:
Sorry to bother you again, but after closing and reopening a terminal, even after following the steps I wrote in the guide, I got:
```python
Traceback (most recent call last):
  File "./imagenet.py", line 28, in <module>
    from jetson_utils import videoSource, videoOutput, cudaFont, Log
ImportError: cannot import name 'videoSource' from 'jetson_utils' (/home/phygital/.local/lib/python3.8/site-packages/jetson_utils/__init__.py)
```

I temporarily fixed the problem by setting `PYTHONPATH=/usr/lib/python3.8/dist-packages/` variable in the shell, but I would like a permanent solution: how can I do it w/o messing with the default configuration?