dusty-nv/jetson-inference #1764: Jetson Inference - Could not open DRM module (NULL) - gstreamer - 5.10.104-tegra

w4sten, 2023-12-18:
Hello everyone,

I am using the jetson inference repo with the Neousys Technology NRU-110V. To map ROS functionalities, I have flashed the system to Jetpack 5.1 according to the manufacturer’s instructions. As an example pipeline, I use the following gstreamer instructions outside the Docker container:

`GST_DEBUG=3 gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,width=1280,height=720,format=UYVY ! autovideosink`

Outside the container, the pipeline works without errors. As soon as I want to run the pipeline inside the container
(dustynv/jetson-inference:r35.2.1 | L4T R35.2.1 | JetPack 5.1 [dustynv/jetson-inference:r35.2.1] L4T R35.2.1	JetPack 5.1), I get the following error message:

```
*root@tegra-ubuntu:/# GST_DEBUG=3 gst-launch-1.0  v4l2src device=/dev/video0 ! video/x-raw,width=1280,height=720,format=UYVY ! autovideosink*
*0:00:01.180704546    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0x120000: 'AVR (Audio Visual Research)' is not mapped*
*0:00:01.180838087    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0x180000: 'CAF (Apple Core Audio File)' is not mapped*
*0:00:01.180885832    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0x100000: 'HTK (HMM Tool Kit)' is not mapped*
*0:00:01.180942410    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0xc0000: 'MAT4 (GNU Octave 2.0 / Matlab 4.2)' is not mapped*
*0:00:01.180999820    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0xd0000: 'MAT5 (GNU Octave 2.1 / Matlab 5.0)' is not mapped*
*0:00:01.181068462    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0x210000: 'MPC (Akai MPC 2k)' is not mapped*
*0:00:01.181134673    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0xe0000: 'PVF (Portable Voice Format)' is not mapped*
*0:00:01.181239444    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0x160000: 'SD2 (Sound Designer II)' is not mapped*
*0:00:01.181327479    71 0xaaaafea7c780 WARN                 default gstsf.c:97:gst_sf_create_audio_template_caps: format 0x190000: 'WVE (Psion Series 3)' is not mapped*
*0:00:01.772811883    71 0xaaaafea7c780 WARN                  ladspa gstladspa.c:507:plugin_init:<plugin174> no LADSPA plugins found, check LADSPA_PATH*
*Setting pipeline to PAUSED ...*
*0:00:07.687856436    70 0xaaaae8e802a0 WARN                 kmssink gstkmssink.c:835:gst_kms_sink_start:<autovideosink0-actual-sink-kms> error: Could not open DRM module (NULL)*
*0:00:07.687991193    70 0xaaaae8e802a0 WARN                 kmssink gstkmssink.c:835:gst_kms_sink_start:<autovideosink0-actual-sink-kms> error: reason: No such file or directory (2)*
*0:00:07.688391654    70 0xaaaae8e802a0 WARN                basesink gstbasesink.c:5367:gst_base_sink_change_state:<autovideosink0-actual-sink-kms> error: Failed to start*
*Pipeline is live and does not need PREROLL ...*
*Setting pipeline to PLAYING ...*
*0:00:07.879654162    70 0xaaaae8f15000 WARN                 basesrc gstbasesrc.c:3072:gst_base_src_loop:<v4l2src0> error: Internal data stream error.*
*0:00:07.879871961    70 0xaaaae8f15000 WARN                 basesrc gstbasesrc.c:3072:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason not-negotiated (-4)*
*New clock: GstSystemClock*
*ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.*
*Additional debug info:*
*gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:*
*streaming stopped, reason not-negotiated (-4)*
*Execution ended after 0:00:00.000731737*
*Setting pipeline to NULL ...*
*Freeing pipeline ...*
```
I have already been able to narrow down the cause field to possibly missing authorizations or unrecognized DRM drivers via other posts. My goal is ultimately to use the functionalities of the repository, but these give me similar error messages. Therefore, the basic gstreamer pipeline should run first.

Here is my info from jtop. I did not install the missing libraries due to limited memory space. The nvidia docker container toolkit is installed. If further information is required, I will be happy to provide it:

[image]

I'am grateful for any help.
Thanks a lot.

dusty-nv, 2023-12-18:
Hi @w4sten, are you able to reproduce this issue with your gstreamer pipeline in the l4t-jetpack base container?

If so, you may want to post this to the Jetson forums so we can take a look at what additional mounts or drivers may needed in the container for DRM.

w4sten, 2023-12-20:
Hello @dusty-nv, 

thanks for your hint. 

According to the current status, the error also occurs with the base container. I have already written a post in the forum https://forums.developer.nvidia.com/t/jetson-inference-could-not-open-drm-module-null-gstreamer-5-10-104-tegra/276491 . 

As an addition to the previous post: I start both containers with -v /lib/modules/5.10.104-tegra:/lib/modules/5.10.104-tegra to avoid getting modprobe errors. I got the info according to your comment from https://github.com/dusty-nv/jetson-inference/issues/1556. 

Thanks again for the help. 
I hope to solve the problem via the nvidia forum.

Jeevi10, 2024-02-29:
GST_DEBUG=3 gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,width=1280,height=720,format=UYVY ! videoconvert ! autovideosink -v
the above code actually works!
however I couldn't get the stream through video-viewer, or VideoSource API? Can anyone help me with this issue?