dusty-nv/jetson-inference #1770: Trying to combine road_following program with detectnet program jetson nano jetracer

CamRoni1339, 2024-01-04:
@dusty-nv 

Hello, 

I am trying to combine your detectnet program with a road_rollowing program for the jetson nano jetracer. The road_following program uses csiCamera and detectnet uses gStreamer so it has been a little bit of a struggle trying to combine them into one. Below is the code for each program, the combined program, and the error message I receive when trying to run it. 

[image]
[image]
[image]
[image]

I am trying to combine the programs so I can make the car stop whenever it recognizes a stop sign. Any help with fixing this would be much appreciated.

Thank you so much.

dusty-nv, 2024-01-04:
Hi @CamRoni1339, try calling `cudaFromNumpy()` followed by `cudaConvertColor()` on the image returned by cv2 `camera.read()` before passing it to `net.Detect()`.   No idea what passing camera.cap to net.Detect() would do.

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-from-numpy-arrays
* https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/cuda-from-cv.py

CamRoni1339, 2024-01-05:
@dusty-nv 

Hello, 

I'm sorry to be asking what has to be simple questions but I am new to this and am not very good at it. I have tried calling cudafromnumpy() and cudaconvertcolor(), but I end up getting the error shown below. I'm sure that I am just simply not calling it correctly, so if you could show me how to do that, that would be great.

[image]

Thank you so much for your time.

CamRoni1339, 2024-01-05:
UPDATE

I figured out the format for cudaFromNumpy, but if you could show me what to do for cudaConvertColor that would be great.

Thank you so much for you time again.

dusty-nv, 2024-01-05:
@CamRoni1339 see this example that I had linked to above that shows the use of cudaFromNumpy/cudaConvertColor to convert an OpenCV image:

https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/cuda-from-cv.py

CamRoni1339, 2024-01-10:
Thank you so much for the help! I have gotten to work.

closed 2024-01-10 by CamRoni1339