dusty-nv/jetson-inference #1771: How to use jetson-inference to conda env?

gradypark86, 2024-01-04:
Hi Dusty. I have an issue about using jetson-inference in conda env. 

I have installed jetson-inference in system Python env (/usr/lib/python*/dist-packages/), and I can `import jetson_inference` and `import jetson_utils` normally.

Now I want to install jetson-inference to a conda env, as you mentioned in this issue #1523, I modified  [jetson-inference/python/bindings/CMakeLists.txt](https://github.com/dusty-nv/jetson-inference/blob/9ee9a950a80fbc7597d7e78a7ba0a282e85fae78/python/bindings/CMakeLists.txt#L76) and [jetson-utils/python/bindings/CMakeLists.txt](https://github.com/dusty-nv/jetson-utils/blob/e82dad8a9e1f2599435671052a39c616e6ab7c56/python/bindings/CMakeLists.txt#76), change `PYTHON_BINDING_INSTALL_DIR` from `/usr/lib/python${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR}/dist-packages` to `/home/username/miniconda3/envs/envname/lib/python3.8/site-packages` and then 
```bash
cmake ../
make -j$(nproc)
sudo make install
sudo ldconfig
```

But when I `import jetson_inference` and `import jetson_utils` in this conda env, it reported an error:
> Traceback (most recent call last):
  File "/home/username/Lab/jetson-inference/test.py", line 1, in <module>
    import jetson_inference
  File "/home/username/miniconda3/envs/envname/lib/python3.8/site-packages/jetson_inference/__init__.py", line 5, in <module>
    import jetson_utils
  File "/home/username/miniconda3/envs/envname/lib/python3.8/site-packages/jetson_utils/__init__.py", line 4, in <module>
    from jetson_utils_python import *
ImportError: /lib/aarch64-linux-gnu/libgstrtspserver-1.0.so.0: undefined symbol: gst_rtsp_watch_send_messages

I can only `import jetson` with no error.
Then I installed jetson-inference to another conda env and `import jetson_inference`, the error is:
> Traceback (most recent call last):
  File "/home/username/Lab/jetson-inference/test.py", line 1, in <module>
    import jetson_inference
  File "/home/username/miniconda3/envs/envname1/lib/python3.8/site-packages/jetson_inference/__init__.py", line 5, in <module>
    import jetson_utils
  File "/home/username/miniconda3/envs/envname1/lib/python3.8/site-packages/jetson_utils/__init__.py", line 4, in <module>
    from jetson_utils_python import *
ImportError: /lib/aarch64-linux-gnu/libgobject-2.0.so.0: undefined symbol: ffi_type_uint32, version LIBFFI_BASE_7.0

Could you give me a hint about that? Is it because I've already installed it in the system Python environment and it won't install properly in conda virtual environments?

Thanks.

dusty-nv, 2024-01-04:
Hi @gradypark86 sorry I'm not sure, I don't use conda - does it have an option to include `--system-site-packages` like virtualenv does?

gradypark86, 2024-01-05:
Hi Dusty, thank you for your reply and sorry for taking up your time.

Unfortunately, conda does not have `--system-site-packages` option. But I figured it out and `import jetson_inference` successfully in conda env.

1. Modify 2 `Cmakelists.txt` files as I mentioned above and install. Or just cp `jetson` `Jetson` `jetson_inference` `jetson_utils` folders and `jetson_utils_python.so` `jetson_inference_python.so` from `/usr/lib/python3.8/dist-packages` to `~/miniconda3/envs/envname/lib/python3.8/site-packages`.
2. `export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libffi.so.7`, otherwise it will report an error:
    > ImportError: /lib/aarch64-linux-gnu/libgobject-2.0.so.0: undefined symbol: ffi_type_uint32, version LIBFFI_BASE_7.0
3. Normally `import jetson_inference` and `import jetson_utils` in conda env.

The previous issue may have been due to a problematic conda environment.

closed 2024-01-05 by gradypark86