dusty-nv/jetson-inference #1774: gstDecoder -- failed to set pipeline state to PLAYING - Orin AGX

BenHamm, 2024-01-04:
I am trying to get Tao person detection running my AGX, but I'm hitting an apparent issue with GST at the capture stage. Please assist. (Also, thank you @dusty-nv for all of the great containers and work! Would be totally stuck without you)

Hardware: Jetson Orin AGX
Container: dustynv/jetson-inference:r36.2.0 (launched using the run.sh command from the jetson_containers repo)

Code:
```
from jetson_inference import detectNet
from jetson_utils import videoSource, videoOutput, Log

# RTSP URL format: rtsp://username:password@ip_address:port/stream_path
rtsp_url = "rtsp://orintapo:nvidia@192.168.0.93:554/stream1"

# create video sources and outputs
input = videoSource(rtsp_url)
# input = videoSource('/jetson-inference/pedestrians.mp4')

img = input.Capture()
```
I get the folllowing output if I try to run the code above. If I use the pedestrians video instead, the output is a bit different but the error is the same. I have confirmed that I can pull a frame from the camera using OpenCV, and I have confirmed that the pedstrians.mp4 is valid.

```
[gstreamer] initialized gstreamer, version 1.20.3.0
[gstreamer] gstDecoder -- creating decoder for orintapo:nvidia@192.168.0.93
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
sh: 1: lsmod: not found
sh: 1: modprobe: not found
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
[gstreamer] gstDecoder -- discovered video resolution: 2560x1440  (framerate 15.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)5, profile=(string)main, width=(int)2560, height=(int)1440, framerate=(fraction)15/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://orintapo:nvidia@192.168.0.93:554/stream1 latency=10 ! queue ! rtph264depay ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink sync=false
[video]  created gstDecoder from rtsp://orintapo:nvidia@192.168.0.93:554/stream1
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: rtsp://orintapo:nvidia@192.168.0.93:554/stream1
     - protocol:  rtsp
     - location:  orintapo:nvidia@192.168.0.93
     - port:      554
  -- deviceType: ip
  -- ioType:     input
  -- codec:      H264
  -- codecType:  v4l2
  -- width:      2560
  -- height:     1440
  -- frameRate:  15
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- latency     10
------------------------------------------------
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
[gstreamer] gstDecoder -- failed to set pipeline state to PLAYING (error 0)
Traceback (most recent call last):
  File "/jetson-inference/test.py", line 12, in <module>
    img = input.Capture()
Exception: jetson.utils -- videoSource failed to capture image
```

BenHamm, 2024-01-05:
Thanks @dusty-nv! Seems like the same issue:

```
root@orin-desktop:/jetson-inference# video-viewer pedestrians.mp4 pedestrians_2.mp4
[gstreamer] initialized gstreamer, version 1.20.3.0
[gstreamer] gstDecoder -- creating decoder for pedestrians.mp4
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
sh: 1: lsmod: not found
sh: 1: modprobe: not found
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
[gstreamer] gstDecoder -- discovered video resolution: 960x540  (framerate 29.970030 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)high, width=(int)960, height=(int)540, framerate=(fraction)30000/1001, pixel-aspect-ratio=(fraction)1/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=pedestrians.mp4 ! qtdemux ! queue ! h264parse ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from file:///opt/jetson-inference/pedestrians.mp4
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///opt/jetson-inference/pedestrians.mp4
     - protocol:  file
     - location:  pedestrians.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      H264
  -- codecType:  v4l2
  -- width:      960
  -- height:     540
  -- frameRate:  29.97
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
failed to find/open file /proc/device-tree/model
[gstreamer] gstEncoder -- detected board 'Jetson AGX Orin Developer Kit'
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! nvvidconv name=vidconv ! video/x-raw(memory:NVMM) ! nvv4l2h264enc name=encoder bitrate=4000000 maxperf-enable=1 ! video/x-h264 ! h264parse ! qtmux ! filesink location=pedestrians_2.mp4 
[video]  created gstEncoder from file:///opt/jetson-inference/pedestrians_2.mp4
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: file:///opt/jetson-inference/pedestrians_2.mp4
     - protocol:  file
     - location:  pedestrians_2.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     output
  -- codec:      H264
  -- codecType:  v4l2
  -- frameRate:  30
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  3440x1440
[OpenGL] glDisplay -- X window resolution:    3440x1440
[OpenGL] glDisplay -- display device initialized (3440x1440)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- width:      3440
  -- height:     1440
  -- frameRate:  0
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
[gstreamer] gstDecoder -- failed to set pipeline state to PLAYING (error 0)
video-viewer:  shutting down...
video-viewer:  shutdown complete
```

dusty-nv, 2024-01-05:
Hmm, can you run it like this, and see if GStreamer prints additional relevant information right around the `gstDecoder -- failed to set pipeline state to PLAYING` error?

```bash
GST_DEBUG=5 video-viewer --debug /mount/videos/pedestrians.mp4 /mount/videos/pedestrians_4.mp4
```

I can't really find much about that `Inappropriate ioctl for device` message (which I don't receive), so I think it may be worth testing GStreamer outside the container (with gst-launch-1.0 and a standalone pipeline) to confirm that your drivers and OS are functioning correctly...

https://docs.nvidia.com/jetson/archives/r34.1/DeveloperGuide/text/SD/Multimedia/AcceleratedGstreamer.html

BenHamm, 2024-01-05:
Here is the result of what you suggested, run within the container

```
root@orin-desktop:/opt/jetson-inference# GST_DEBUG=5 vide
[TRUNCATED THINGS THAT LOOK FINE]
0:00:03.115263457 1916809 0xaaaad9230a00 DEBUG     GST_PLUGIN_LOADING gstplugin.c:1696:gst_plugin_ext_dep_scan_dir_and_match_names:<plugin157> g_dir_open(/usr/lib/aarch64-linux-gnu/lv2) failed: Error opening directory “/usr/lib/aarch64-linux-gnu/lv2”: No such file or directory
0:00:03.116351852 1916809 0xaaaad9230a00 DEBUG           GST_REGISTRY gstregistry.c:1414:gst_registry_scan_path_internal:<registry0> registry changed in path /usr/lib/aarch64-linux-gnu/gstreamer-1.0: 0
0:00:03.116365580 1916809 0xaaaad9230a00 DEBUG           GST_REGISTRY gstregistry.c:1640:gst_registry_remove_cache_plugins:<registry0> removing cached plugins
0:00:03.116473613 1916809 0xaaaad9230a00 INFO            GST_REGISTRY gstregistry.c:1785:scan_and_update_registry: Registry cache has not changed
0:00:03.116485550 1916809 0xaaaad9230a00 INFO            GST_REGISTRY gstregistry.c:1862:ensure_current_registry: registry reading and updating done, result = 1
0:00:03.116502190 1916809 0xaaaad9230a00 INFO                GST_INIT gst.c:833:init_post: GLib runtime version: 2.72.4
0:00:03.196776325 1916809 0xaaaad9230a00 INFO                GST_INIT gst.c:835:init_post: GLib headers version: 2.72.1
0:00:03.196923430 1916809 0xaaaad9230a00 INFO                GST_INIT gst.c:837:init_post: initialized GStreamer successfully
0:00:03.196954375 1916809 0xaaaad9230a00 DEBUG                default gsttracerutils.c:77:_priv_gst_tracing_init: Initializing GstTracer
[gstreamer] initialized gstreamer, version 1.20.3.0
[gstreamer] gstDecoder -- couldn't find file '/mount/videos/pedestrians.mp4'
[gstreamer] gstDecoder -- failed to create decoder for file:///mount/videos/pedestrians.mp4
video-viewer:  failed to create input stream
```

Now, doing as you suggest outside the container

BenHamm, 2024-01-05:
Ok, I installed GSTstreamer outside the container per the doc you shared, and tried my best to follow test instructions:

```
orin@orin-desktop:~/jetson-containers$ gst-launch-1.0 filesrc location=<pedestrians.mp4> ! \
>      qtdemux ! queue ! h264parse ! nvv4l2decoder ! nv3dsink -e
No protocol specified
No protocol specified
No protocol specified
No protocol specified
No protocol specified
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstNv3dSink:nv3dsink0: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure.
Additional debug info:
gstbasesink.c(5367): gst_base_sink_change_state (): /GstPipeline:pipeline0/GstNv3dSink:nv3dsink0:
Failed to start
```

BenHamm, 2024-01-05:
I should note that i am ssh'd into this machine. Perhaps there's some issue with X11? I can try doing all of this directly on the machine...

dusty-nv, 2024-01-06:
> [gstreamer] gstDecoder -- couldn't find file '/mount/videos/pedestrians.mp4'
> [gstreamer] gstDecoder -- failed to create decoder for file:///mount/videos/pedestrians.mp4

You need to change the filenames in that command to reflect where you have pedestrians.mp4 at

> I should note that i am ssh'd into this machine. Perhaps there's some issue with X11? I can try doing all of this directly on the machine...

Yes, using `nv3dsink` without display attached would be problematic for that particular pipeline you tried running.  You can either run it directly on machine with display attached, or use another pipeline (replacing `nv3dsink` with `fakesink` may help)

BenHamm, 2024-01-21:
Hey sorry for my delay in getting back on this and thanks for your help so far!  I tried running your test command directly on the machine with the display attached: 

`root@orin-desktop:/jetson-inference# video-viewer pedestrians.mp4                  
[gstreamer] initialized gstreamer, version 1.20.3.0
[gstreamer] gstDecoder -- creating decoder for pedestrians.mp4
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
sh: 1: lsmod: not found
sh: 1: modprobe: not found
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
[gstreamer] gstDecoder -- discovered video resolution: 960x540  (framerate 29.970030 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)high, width=(int)960, height=(int)540, framerate=(fraction)30000/1001, pixel-aspect-ratio=(fraction)1/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=pedestrians.mp4 ! qtdemux ! queue ! h264parse ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from file:///opt/jetson-inference/pedestrians.mp4
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///opt/jetson-inference/pedestrians.mp4
     - protocol:  file
     - location:  pedestrians.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      H264
  -- codecType:  v4l2
  -- width:      960
  -- height:     540
  -- frameRate:  29.97
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  3440x1440
[OpenGL] glDisplay -- X window resolution:    3440x1440
[OpenGL] glDisplay -- display device initialized (3440x1440)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- width:      3440
  -- height:     1440
  -- frameRate:  0
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
[gstreamer] gstDecoder -- failed to set pipeline state to PLAYING (error 0)
video-viewer:  shutting down...
video-viewer:  shutdown complete`

AhmedL99, 2024-01-31:
Hello everyone,
 
I am facing the same issue! trying to get TAO person detection running on my Jetson Orin Nano. I have followed instructions in the [Jetson AI Fundamentals - S3E4 - Object Detection Inference](https://www.youtube.com/watch?v=obt60r8ZeB0&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=4&t=219s&ab_channel=NVIDIADeveloper) There is no display attached to it. The module works fine without exporting to an output file `detectnet --model=peoplenet pedestrians.mp4`. I got this log output whenever I export to an output file or when I run `video-viewer` also:  

```
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for pedestrians.mp4
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- discovered video resolution: 960x540  (framerate 29.970030 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)high, width=(int)960, height=(int)540, framerate=(fraction)30000/1001, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=pedestrians.mp4 ! qtdemux ! queue ! h264parse ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from file:///working_dir/pedestrians.mp4
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///working_dir/pedestrians.mp4
     - protocol:  file
     - location:  pedestrians.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      H264
  -- codecType:  v4l2
  -- width:      960
  -- height:     540
  -- frameRate:  29.97
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
failed to find/open file /proc/device-tree/model
[gstreamer] gstEncoder -- detected board 'PCB Arts - EdgeKit'
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! nvvidconv name=vidconv ! video/x-raw(memory:NVMM) ! nvv4l2h264enc name=encoder bitrate=4000000 maxperf-enable=1 ! video/x-h264 ! h264parse ! qtmux ! filesink location=pedestrians_peoplenet.mp4 
[video]  created gstEncoder from file:///working_dir/pedestrians_peoplenet.mp4
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: file:///working_dir/pedestrians_peoplenet.mp4
     - protocol:  file
     - location:  pedestrians_peoplenet.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     output
  -- codec:      H264
  -- codecType:  v4l2
  -- frameRate:  30
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
------------------------------------------------
[OpenGL] failed to open X11 server connection.
[OpenGL] failed to create X11 Window.

detectNet -- loading detection network model from:
          -- prototxt     
          -- model        networks/peoplenet_deployable_quantized_v2.6.1/resnet34_peoplenet_int8.etlt.engine
          -- input_blob   'input_1'
          -- output_cvg   'output_cov/Sigmoid'
          -- output_bbox  'output_bbox/BiasAdd'
          -- mean_pixel   0.000000
          -- class_labels networks/peoplenet_deployable_quantized_v2.6.1/labels.txt
          -- class_colors networks/peoplenet_deployable_quantized_v2.6.1/colors.txt
          -- threshold    0.500000
[... 217 lines cut ...]
detected obj 17  class #2 (face)  confidence=0.984040
bounding box 17  (694.51, 73.12)  (698.40, 81.07)  w=3.88  h=7.94

detected obj 18  class #2 (face)  confidence=0.583265
bounding box 18  (657.94, 54.64)  (661.21, 61.72)  w=3.27  h=7.07
[cuda]   allocated 2 ring buffers (777600 bytes each, 1555200 bytes total)
[gstreamer] gstEncoder -- starting pipeline, transitioning to GST_STATE_PLAYING
[gstreamer] gstEncoder -- failed to set pipeline state to PLAYING (error 0)
detectnet:  shutting down...
[gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstDecoder -- onPreroll()
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> capsfilter2
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> vidconv
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> decoder
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> h264parse1
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> queue0
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> qtdemux1
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> filesrc0
[gstreamer] gstreamer changed state from PLAYING to PAUSED ==> pipeline0
[gstreamer] gstDecoder -- pipeline stopped
detectnet:  shutdown complete.
```
Looking forward to hearing from you @dusty-nv! 
Thank you in advance!