dusty-nv/jetson-inference #1778: ROS2 Dashing + Jetson Inference

Rolix57, 2024-01-09:
Hi guys!

I am trying to run detectnet and imagenet using ROS Dashing in a Jetson Nano 2GB with the JetPack 4.6.
Dashing seems to be working properly - I've just tested some dummy publisher and subscriber nodes.
The real issue comes when I want to integrate Jetson Inference.
The build works without an issue. However, when I try to do the launch for video_source, imagenet or detectnet.

`ros2 launch ros_deep_learning detectnet.ros2.launch`

I have the following error:

`1704759117.6031718 [INFO] [launch]: All log files can be found below/home/puzzlebot/.ros/log/2024-01-08-18-11-56-045550-puzzlebot-desktop-32545
1704759117.6037271 [INFO] [launch]: Default logging verbosity is set to INFO
1704759117.7718298 [ERROR] [launch]: Caught exception in launch (see debug for traceback): invalid syntax (video_source.ros2.launch, line 1)`

The log file contains the information I posted above.
Previously I have used the ROS2 docker images without an issue, any clue what might be the problem?

Best

dusty-nv, 2024-01-09:
> invalid syntax (video_source.ros2.launch, line 1)

Hi @Rolix57, it was probably some syntax change that occurred between Dashing and Foxy, and Dashing being EOL for some time I haven't ever used it

Rolix57, 2024-01-10:
So I figured it out!

Thank you for your response. So... you were totally right, they changed some stuff related to ROS2.
Starting Foxy you can create launches in XML, Yaml and Python. From Dashing to all the previous ones, you can only use Python for launch files.

I wrote the corresponding launch files for the ones available in the repository (video_source, video_output, video_viewer, detectnet, and imagenet), and they work well enough, they just lag a little bit. I assume it is because of the available computational power of the Jetson Nano.

Do you want me to share them with you?

Best

dusty-nv, 2024-01-10:
Thank you @Rolix57, yes sure! - for other ROS2 projects I use the Python-based launch files, but in ros_deep_learning I had previously kept it XML for pseudo-compatibility between ROS1/ROS2 (years ago at this point).  But with the waning user base of ROS1, I think it would be good to add the Python launch files 👍

Rolix57, 2024-01-10:
Sure!
They work fine for ROS2 Dashing. However, they might need some slight changes for Foxy+ versions.
[launches.zip](https://github.com/dusty-nv/jetson-inference/files/13893975/launches.zip)