dusty-nv/jetson-inference #1779: The nvjpeg plugin is not used

Baominchao, 2024-01-11:
My system environment is as follows:
jetson nano : jetpack 4.6
camera: astra pro plus

1. I used the command:`gst-inspect-1.0 nvjpeg` and the output: 

Plugin Details:
  Name                     nvjpeg
  Description              Nvidia accelerated JPeg plugin library
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvjpeg.so
  Version                  1.2.3
  License                  LGPL
  Source module            gst-plugins-good
  Source release date      2014-02-08
  Binary package           GStreamer Good Plug-ins source release
  Origin URL               Unknown package origin

  nvjpegdec: JPEG image decoder
  nvjpegenc: JPEG image encoder

  2 features:
  +-- 2 elements

2.  I used the command `v4l2-ctl --list-formats-ext -d /dev/video0` to  check my camera stream ：
ioctl: VIDIOC_ENUM_FMT

	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 2048x1536
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x960
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1920x1080
			Interval: Discrete 0.333s (3.000 fps)
		Size: Discrete 2048x1536
			Interval: Discrete 0.333s (3.000 fps)
		Size: Discrete 1280x960
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)

3. My python code as follow:
#!/usr/bin/env python3

from jetson_inference import detectNet
from jetson_utils import videoSource, videoOutput

input = videoSource("/dev/video0")
output = videoOutput()	
net = detectNet("ssd-mobilenet-v2", threshold=0.5)

while True:
    img = input.Capture()
    if img is None: 
        continue  
    detections = net.Detect(img)
    output.Render(img)
    output.SetStatus("{:s} | Network {:.0f} FPS".format(args.network, net.GetNetworkFPS()))
    if not input.IsStreaming() or not output.IsStreaming():
        break

4. The linux terminal output:
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! image/jpeg, width=(int)1280, height=(int)720, framerate=30/1 ! jpegdec name=decoder ! video/x-raw ! appsink name=mysink sync=false
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      MJPEG
  -- codecType:  cpu
  -- width:      1280
  -- height:     720
  -- frameRate:  30
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
 
The decoder in the  gstreamer pipeline is jpegdec not  nvjpeg , so  it use cpu (  -- codecType:  cpu,  soft decoding). How can i use the nvjpeg( hard decoding)?

dusty-nv, 2024-01-11:
> The decoder in the gstreamer pipeline is jpegdec not nvjpeg , so it use cpu ( -- codecType: cpu, soft decoding). How can i use the nvjpeg( hard decoding)?

I had previously encountered some reliability issues with nvjpeg decoder on some (but not all) JPEG sources (primarily older MJPEG cameras) on older JetPack versions, hence why I have it as CPU-based jpegdec.  However you can try changing this behavior by commenting out these two lines under `jetson-inference/utils/codec/gstUtility.cpp` and recompiling/reinstalling jetson-inference:

https://github.com/dusty-nv/jetson-utils/blob/e82dad8a9e1f2599435671052a39c616e6ab7c56/codec/gstUtility.cpp#L452

Baominchao, 2024-01-12:
Thank you very much for your reply.  
I commented out these two lines under jetson-inference/utils/codec/gstUtility.cpp and recompiling/reinstalling jetson-inference.  The terminal output:
...
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! image/jpeg, width=(int)1280, height=(int)720, framerate=30/1 ! nvjpegdec name=decoder ! video/x-raw(memory:NVMM) ! appsink name=mysink sync=false
...
[gstreamer] gstBufferManager recieve caps:  video/x-raw(memory:NVMM), format=(string)I420, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
[gstreamer] gstBufferManager -- recieved first frame, codec=MJPEG format=i420 width=1280 height=720 size=1382407
[gstreamer] gstBufferManager -- recieved NVMM memory
nvbuf_utils: nvbuffer Payload Type not supported
[gstreamer] gstBufferManager -- failed to get NVMM buffer params
[gstreamer] gstCamera -- failed to handle incoming buffer
[gstreamer] gstreamer message qos ==> v4l2src0
[gstreamer] gstBufferManager -- map buffer size was less than max size (1008 vs 1382407)
I ask chatgpt for GStreamer pipeline and test it : 
# Define the GStreamer pipeline for hardware-accelerated decoding
# This is a generic example and may need to be adjusted for your specific hardware and camera
gst_pipeline = (
    'v4l2src device=/dev/video0 ! '
    'image/jpeg, width=(int)640, height=(int)480, framerate=(fraction)30/1 ! '
    'jpegparse ! '
    'nvjpegdec ! '
    'videoconvert ! '
    'video/x-raw, format=(string)BGR ! '
    'appsink'
)

# Create a VideoCapture object with the GStreamer pipeline
cap = cv2.VideoCapture(gst_pipeline, cv2.CAP_GSTREAMER)

It's ok. What is the problem with the two different GStreamer pipelines? How can I run success in Jetson inference?
Best

dusty-nv, 2024-01-12:
OpenCV does not use the NVMM zero-copy extension, and it appears NVJPEG uses it differently than jetson-inference expects.   When you compile jetson-inference, try running `cmake -DENABLE_NVMM=OFF ../`

Baominchao, 2024-01-15:
Thank you very much for your reply. I have tried your steps to enable the nvjpeg plugin. The reason why I want to enable nvjpg is to reduce the usage of the Jetson Nano CPU. However, I have found that the CPU usage after enabling nvjpeg is similar to not enabling it, and even higher (I only roughly checked with Jtop, which may not be accurate). I suspect that the CPU may have gone to do memory copying and other work (because - ENABLE NVMM=OFF). Thank you very much again for your answer, which allowed me to understand the content of soft decoding and hard decoding in  jetson-inference. Thank you.

closed 2024-01-18 by Baominchao