dusty-nv/jetson-inference #1781: [gstreamer] gstCamera::Capture() -- an error occurred retrieving the next image buffer`

AnilSarode, 2024-01-13:
I am getting the following error in "Coding your own Object Detection Program in Python for live camera stream" as I am following this video link:https://www.youtube.com/watch?v=obt60r8ZeB0&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=13
 
`
[cuda]   cudaGraphicsGLRegisterBuffer(&interop, allocDMA(type), cudaGraphicsRegisterFlagsFromGL(flags))
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /jetson-inference/utils/display/glTexture.cpp:360
[cuda]   cudaGetLastError()
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /jetson-inference/utils/cuda/cudaYUV-YV12.cu:119
[cuda]   cudaI420ToRGB(input, (uchar3*)output, width, height)
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /jetson-inference/utils/cuda/cudaColorspace.cpp:53
[cuda]   cudaConvertColor(latestYUV, mFormatYUV, nextRGB, format, mOptions->width, mOptions->height)
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /jetson-inference/utils/codec/gstBufferManager.cpp:445
[gstreamer] gstBufferManager -- unsupported image format (rgb8)
[gstreamer]                     supported formats are:
[gstreamer]                        * rgb8
[gstreamer]                        * rgba8
[gstreamer]                        * rgb32f
[gstreamer]                        * rgba32f
[gstreamer] gstCamera::Capture() -- an error occurred retrieving the next image buffer`

dusty-nv, 2024-01-13:
@AnilSarode do you have a physical display attached to your Jetson?   If not, the GUI won't work over SSH with X11 forwarding because of the OpenGL/CUDA interop.  Instead, see some of the streaming options from here like RTP/RTSP if you are running headlessly and want to view the video remotely:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md

AnilSarode, 2024-01-14:
@dusty-nv Thanks for your quick response. I am also accessing the Jetson Nano through Physical Displays (i.e. physical display attached) and also by using SSH. I am running a Python script on the terminal of the attached monitor display and not through the SSH terminal.

dusty-nv, 2024-01-14:
OK gotcha @AnilSarode - can you run `sudo apt-get install mesa-utils` and then `glxinfo` ?   I wonder if somehow you got another OpenGL driver installed at some point.  You can also test `glxgears` that way.

AnilSarode, 2024-01-15:
@dusty-nv This the `glxinfo` : [glxinfo.txt](https://github.com/dusty-nv/jetson-inference/files/13935133/glxinfo.txt)

and `glxgears` works correctly.

dusty-nv, 2024-01-15:
@AnilSarode are you able to run any of the other examples from jetson-inference like imagenet, detectnet, video-viewer, ect?

AnilSarode, 2024-01-22:
@dusty-nv Sorry for the late response. any of the other examples from jetson-inference like imagenet, detectnet, video-viewer, etc also give me the same error as above.

GITsologun, 2024-02-02:
@AnilSarode Have you made any progress on this issue? It's quite the same problem i meet.

dusty-nv, 2024-02-02:
@GITsologun this typically happens when the NVIDIA OpenGL driver gets overwritten by another graphics driver, or you are attempting to use SSH with X11 forwarding.   If you have another SD card, I would quickly try a fresh install of JetPack to see if the issue re-occurs just to eliminate that possibility.

GITsologun, 2024-02-02:
> @GITsologun this typically happens when the NVIDIA OpenGL driver gets overwritten by another graphics driver, or you are attempting to use SSH with X11 forwarding. If you have another SD card, I [...]

Appreciate for your quick reply. I didn't use SSH or X11. I bought this Jetson Orin NX few days ago, and the only thing I did is following the basic tutorial with the default system.  Maybe you are right that I can try a fresh install of JetPack, however is there any other quick solution?

dusty-nv, 2024-02-02:
Hmm okay, since it's already a fresh install, hopefully you needn't do that again.  Do the graphics work if you use the jetson-inference [docker container](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md) instead?  Or are you already using container?

GITsologun, 2024-02-03:
> Hmm okay, since it's already a fresh install, hopefully you needn't do that again. Do the graphics work if you use the jetson-inference [docker container](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md) instead? Or are you already using container?

Not yet. I am trying to run the docker container by "docker/run.sh", suffering the very slow downloading of "pulling fs layer"...

GITsologun, 2024-02-03:
> Hmm okay, since it's already a fresh install, hopefully you needn't do that again. Do the graphics work if you use the jetson-inference [docker container](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md) instead? Or are you already using container?

I just tested the docker container, and it didn't work. See the attached picture.
[image]

GITsologun, 2024-02-03:
> Hmm okay, since it's already a fresh install, hopefully you needn't do that again. Do the graphics work if you use the jetson-inference [docker container](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md) instead? Or are you already using container?

The problem has been solved by fresh install of both Jetson Linux and Jetpack. The experience is that one should be very careful on the screen info, in case of missing any errors.  Thank you for your help, and good luck to the other guys.