dusty-nv/jetson-inference #1782: Detectnet postprocessing and device synchronization

andrejlevkovitch, 2024-01-15:
I want to clarify: do we need device synchronization (cudaDeviceSynchronize) before detectnet postprocessing, or not?

https://github.com/dusty-nv/jetson-inference/blob/fe8b42c8da75c1c353dc59fa1fd079820024b89d/c/detectNet.cpp#L518-L537

I see that some time ago the call was commented, __but__ the call is __after__ detectnet postprocessing, what doesn't make a lot of sense, because, if I understand logic correctly, at postprocessing time we __already__ should have access to results on CPU. If yes, then should we synchronize before postprocessing? If not, then why? I don't see any other synchronization calls, not in Detectnet class, not in detectnet example, not in basic tensor class. Is it a bug?

dusty-nv, 2024-01-16:
@andrejlevkovitch by default, the call to [`tensorNet::ProcessNetwork()`](https://github.com/dusty-nv/jetson-inference/blob/fe8b42c8da75c1c353dc59fa1fd079820024b89d/c/tensorNet.cpp#L1745) automatically synchronizes the CUDA stream.  Hence `detectNet::postProcess()` is able to cluster/filter the results on the CPU.   

IIRC the commented out call to `cudaDeviceSynchronize()` pertained to `detectNet::Overlay()`, because that uses CUDA kernels on the overlay image.  So if you need to access the overlay image, then a sync would be needed.

andrejlevkovitch, 2024-01-16:
Oh, I see now, thanks!

closed 2024-01-16 by andrejlevkovitch