dusty-nv/jetson-inference #1789: How to receive image using ROS msg from backgroundnet

Hogushake, 2024-02-01:
hello guys

I'm trying to use backgroundnet in a ROS1 environment.
so i modified the code as follows to receive image_msg from ROS image_publisher.

`img_input = input.Capture(format='rgba8')`
 to 
`img_input = bridge.imgmsg_to_cv2(msg, "rgba8")`

then an error occurred in this code
`net.Process(img_input, filter=args.filter_mode)`

The error is as follows
`TypeError: jetson.inference --failed to get CUDA image from imput argument`

what should i do??

dusty-nv, 2024-02-01:
Hi @Hogushake try converting the numpy array from cv2 to cudaImage using jetson_utils.cudaFromNumpy()

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-from-numpy-arrays
* https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/cuda-from-cv.py

You'll also need to do the BGR->RGB conversion since by default cv2 images are in BGR colorspace.

Hogushake, 2024-02-02:
@dusty-nv Thanks for your reply.
I modified the code based on the answers.

```python
class backgroundnet():
    def __init__(self):
        self.image_subcriber = rospy.Subscriber("image_raw", Image, self.detect, queue_size=1)

    def detect(self, msg):
        try:
           #Edited part
            img_input = bridge.imgmsg_to_cv2(msg, "rgba8")
            img_input = cudaFromNumpy(img_input, isBGR=True)

            net.Process(img_input, filter=args.filter_mode)

            img_output = img_input
                
            output.Render(img_output)

            output.SetStatus("backgroundNet {:s} | Network {:.0f} FPS".format(net.GetNetworkName(), net.GetNetworkFPS()))
            net.PrintProfilerTimes()
           
        except CvBridgeError as error:
            print(error)

if __name__ == '__main__':
    bridge = CvBridge()
    rospy.init_node('backgroundnet', anonymous=True)
    backgroundnet()

    try:
        rospy.spin()
    except KeyboardInterrupt:
        print("Shutting down!")
``` 

But the following error occurs
```python
[cuda]   cudaTensorNormMeanRGB(input, format, width, height, mInputs[0].CUDA, GetInputWidth(), GetInputHeight(), make_float2(0.0f, 1.0f), make_float3(0.485f, 0.456f, 0.406f), make_float3(0.229f, 0.224f, 0.225f), GetStream())
[cuda]      invalid argument (error 1) (hex 0x01)
[cuda]      /home/jaewoo/github_save/jetson-inference/c/backgroundNet.cpp:192
[ERROR] [1706843315.155789]: bad callback: <bound method backgroundnet.detect of <__main__.backgroundnet object at 0x7f68feda58>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "testtt.py", line 76, in detect
    net.Process(self.img_input, filter=args.filter_mode)
Exception: jetson.inference -- backgroundNet.Process() encountered an error processing the image
``` 
I don't know where the error occurred;;;

dusty-nv, 2024-02-02:
@Hogushake can you add a `print(img_input)` after `cudaFromNumpy()` so we can inspect the image dimensions and attributes?

Also maybe try changing it from `rgba8` to `rgb8`

Hogushake, 2024-02-05:
@dusty-nv thank you for your help.
I tried putting print(img_input) as you said.

and the error is
```python
nvbuf_utils: Could not get EGL display connection

backgroundNet -- loading background network from:
           -- model       networks/Background-U2Net/u2net.onnx
           -- input_blob  'input_0'
           -- output_blob 'output_0'
           -- batch_size  1

[TRT]    TensorRT version 8.0.1
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::ScatterND version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_ONNX_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    completed loading NVIDIA plugins.
[TRT]    detected model format - ONNX  (extension '.onnx')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    [MemUsageChange] Init CUDA: CPU +353, GPU +0, now: CPU 389, GPU 6076 (MiB)
[TRT]    native precisions detected for GPU:  FP32, FP16, INT8
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    found engine cache file /usr/local/bin/networks/Background-U2Net/u2net.onnx.1.1.8001.GPU.FP16.engine
[TRT]    found model checksum /usr/local/bin/networks/Background-U2Net/u2net.onnx.sha256sum
[TRT]    echo "$(cat /usr/local/bin/networks/Background-U2Net/u2net.onnx.sha256sum) /usr/local/bin/networks/Background-U2Net/u2net.onnx" | sha256sum --check --status
[TRT]    model matched checksum /usr/local/bin/networks/Background-U2Net/u2net.onnx.sha256sum
[TRT]    loading network plan from engine cache... /usr/local/bin/networks/Background-U2Net/u2net.onnx.1.1.8001.GPU.FP16.engine
[TRT]    device GPU, loaded /usr/local/bin/networks/Background-U2Net/u2net.onnx
[TRT]    [MemUsageChange] Init CUDA: CPU +0, GPU +0, now: CPU 476, GPU 6165 (MiB)
[TRT]    Loaded engine size: 86 MB
[TRT]    [MemUsageSnapshot] deserializeCudaEngine begin: CPU 476 MiB, GPU 6165 MiB
[TRT]    Using cublas a tactic source
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +226, GPU +228, now: CPU 704, GPU 6394 (MiB)
[TRT]    Using cuDNN as a tactic source
[TRT]    [MemUsageChange] Init cuDNN: CPU +307, GPU +312, now: CPU 1011, GPU 6706 (MiB)
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +0, GPU +0, now: CPU 1011, GPU 6706 (MiB)
[TRT]    Deserialization required 2927770 microseconds.
[TRT]    [MemUsageSnapshot] deserializeCudaEngine end: CPU 1011 MiB, GPU 6706 MiB
[TRT]    [MemUsageSnapshot] ExecutionContext creation begin: CPU 1011 MiB, GPU 6706 MiB
[TRT]    Using cublas a tactic source
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +0, GPU +0, now: CPU 1011, GPU 6706 (MiB)
[TRT]    Using cuDNN as a tactic source
[TRT]    [MemUsageChange] Init cuDNN: CPU +0, GPU +0, now: CPU 1011, GPU 6706 (MiB)
[TRT]    Total per-runner device memory is 89660928
[TRT]    Total per-runner host memory is 421120
[... 153 lines cut ...]

```python

<cudaImage object>
   -- ptr:      0x2233dc000
   -- size:     2764800
   -- width:    1280
   -- height:   720
   -- channels: 3
   -- format:   bgr8
   -- mapped:   true
   -- freeOnDelete: true
   -- timestamp:    0.000000

[cuda]   cudaTensorNormMeanRGB(input, format, width, height, mInputs[0].CUDA, GetInputWidth(), GetInputHeight(), make_float2(0.0f, 1.0f), make_float3(0.485f, 0.456f, 0.406f), make_float3(0.229f, 0.224f, 0.225f), GetStream())
[cuda]      invalid argument (error 1) (hex 0x01)
[cuda]      /home/jaewoo/github_save/jetson-inference/c/backgroundNet.cpp:192
[ERROR] [1707116431.632183]: bad callback: <bound method backgroundnet.detect of <__main__.backgroundnet object at 0x7f839f9940>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/jaewoo/catkin_ws/src/u2net/src/testtt.py", line 78, in detect
    net.Process(img_input, filter=args.filter_mode)
Exception: jetson.inference -- backgroundNet.Process() encountered an error processing the image
```

dusty-nv, 2024-02-05:
OK gotcha, thanks @Hogushake - I think it throws the error because it is still in BGR colorspace.   Try `cudaFromNumpy(img_input, isBGR=False)` just to see if it will run.  If that runs, then change back to `isBGR=True` and add the use of `cudaConvertColor()` like in this example:

https://github.com/dusty-nv/jetson-utils/blob/e82dad8a9e1f2599435671052a39c616e6ab7c56/python/examples/cuda-from-cv.py#L60

Hogushake, 2024-04-04:
@dusty-nv thank you.
 I modified code as you said, and No error occurs.

But I can't check if it's working properly.

my modified code is
```
net = backgroundNet(args.network, sys.argv)
input = videoSource(args.input_URI, argv=sys.argv)
output = videoOutput(args.output_URI, argv=sys.argv)

class backgroundnet():
    def __init__(self):
        self.image_sub = rospy.Subscriber("/camera/color/image_raw", Image, self.callback, queue_size=1)
        self.bridge = CvBridge()

    def callback(self, data):
        try:
            cv_image = self.bridge.imgmsg_to_cv2(data, "rgba8")
            self.detect(cv_image)
        except CvBridgeError as e:
            print(e)
           

    def detect(self, image):
        try:
            bgr_img = cudaFromNumpy(image, isBGR=True)
            rgb_img = cudaAllocMapped(width=bgr_img.width,
                                      height=bgr_img.height,
                                      format='rgba8')

            cudaConvertColor(bgr_img, rgb_img)
            net.Process(rgb_img, filter=args.filter_mode)

            output.Render(rgb_img)
            output.SetStatus("backgroundNet {:s} | Network {:.0f} FPS".format(net.GetNetworkName(), net.GetNetworkFPS()))
            net.PrintProfilerTimes()

        except CvBridgeError as error:
            print(error)

if __name__ == '__main__':
    rospy.init_node('backgroundnet', anonymous=True)
    backgroundnet()
    try:
        rospy.spin()
    except KeyboardInterrupt:
        print("Shutting down!")
```

But when I run that code I get the following error
```
backgroundNet -- loading background network from:
           -- model       networks/Background-U2Net/u2net.onnx
           -- input_blob  'input_0'
           -- output_blob 'output_0'
           -- batch_size  1

[TRT]    TensorRT version 8.5.2
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Registered plugin creator - ::BatchTilePlugin_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::CoordConvAC version 1
[TRT]    Registered plugin creator - ::CropAndResizeDynamic version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DecodeBbox3DPlugin version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_Explicit_TF_TRT version 1
[... 190 lines cut ...]
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.05776ms  CUDA   0.05968ms
[TRT]    Network       CPU  59.59396ms  CUDA  59.20493ms
[cuda]   cudaEventElapsedTime(&cuda_time, mEventsGPU[evt], mEventsGPU[evt+1])
[cuda]      device not ready (error 600) (hex 0x258)
[cuda]      /home/frlab/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:769
[TRT]    Post-Process  CPU   0.04883ms  CUDA   0.00000ms
[TRT]    Total         CPU  59.70056ms  CUDA  59.26461ms
[TRT]    ------------------------------------------------

[TRT]    ------------------------------------------------
[TRT]    Timing Report /usr/local/bin/networks/Background-U2Net/u2net.onnx
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.02909ms  CUDA   0.06000ms
[TRT]    Network       CPU  54.75547ms  CUDA  54.34557ms
**[cuda]   cudaEventElapsedTime(&cuda_time, mEventsGPU[evt], mEventsGPU[evt+1])
[cuda]      device not ready (error 600) (hex 0x258)
[cuda]      /home/frlab/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:769**
[TRT]    Post-Process  CPU   0.04512ms  CUDA   0.00000ms
[TRT]    Total         CPU  54.82968ms  CUDA  54.40557ms
[TRT]    ------------------------------------------------

```
I think it's an **output.** error in the code, but I'm not sure.
I would appreciate your help.