dusty-nv/jetson-inference #1790: After disconnection from Jetson RTSP translation, new connection cannot be established again.

At2024Start, 2024-02-01:
1. Run command "./video-viewer csi://0 rtsp://@:8554/stream_1" on Jetson TX2 NX.
2. Run gstreamer on Windows PC connected to Jetson trought USB cable : "gst-launch-1.0 rtspsrc location=rtsp://192.168.55.1:8554/stream_1 latency=100 ! queue ! rtph264depay ! h264parse ! decodebin ! videoconvert ! videoscale ! video/x-raw,width=640,height=480 ! autovideosink"
(Or open "rtsp://192.168.55.1:8554/stream_1" in VLC on Windows ... no difference)
3. Successfully receive video translation on Windows PC.
4. Close receiving translation.
5. All next tryings to open receive translation ends with error.

Log from terminal :

developer@developer-desktop:~/ijGround$ DISPLAY=:0 GST_DEBUG=3 ./video-viewer csi://0 rtsp://@:8554/stream_1
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw(memory:NVMM) ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0
[video]  created gstCamera from csi://0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- width:      1280
  -- height:     720
  -- frameRate:  30
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: rotate-180
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc name=encoder bitrate=4000000 insert-sps-pps=1 insert-vui=1 ! video/x-h264 ! rtph264pay config-interval=1 name=pay0
[gstreamer] GST_LEVEL_WARNING   omx
              /dvs/git/dirty/git-master_linux/3rdparty/gst/gst-omx/omx/gstomx.c:2826  plugin_init
              Failed to load configuration file: Valid key file could not be found in search dirs (searched in: /home/developer/.config:/etc/xdg as per GST_OMX_CONFIG_DIR environment variable, the xdg user config directory (or XDG_CONFIG_HOME) and the system config directory (or XDG_CONFIG_DIRS)
[rtsp]   waiting for RTSP server to start...
[rtsp]   RTSP server started @ rtsp://developer-desktop:8554
[rtsp]   RTSP route added /stream_1 @ rtsp://developer-desktop:8554
[video]  created gstEncoder from rtsp://@:8554/stream_1
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: rtsp://@:8554/stream_1
     - protocol:  rtsp
     - location:  0.0.0.0
     - port:      8554
  -- deviceType: ip
  -- ioType:     output
  -- codec:      H264
  -- codecType:  omx
  -- frameRate:  30
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- latency     10
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
[... 376 lines cut ...]
              gstbasesrc.c:3055  gst_base_src_loop
              error: Internal data stream error.
[gstreamer] GST_LEVEL_WARNING GstAppSrc basesrc
              gstbasesrc.c:3055  gst_base_src_loop
              error: streaming stopped, reason not-linked (-1)
[gstreamer] gstEncoder -- pipeline full, skipping frame 675 (1280x720, 1382400 bytes)
video-viewer:  captured 675 frames (1280x720)
[gstreamer] gstEncoder -- pipeline full, skipping frame 700 (1280x720, 1382400 bytes)
video-viewer:  captured 700 frames (1280x720)
[gstreamer] gstEncoder -- pipeline full, skipping frame 725 (1280x720, 1382400 bytes)
video-viewer:  captured 725 frames (1280x720)
^Creceived SIGINT
video-viewer:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
[gstreamer] gstCamera -- pipeline stopped
[gstreamer] gstEncoder -- shutting down pipeline, sending EOS
[gstreamer] gstEncoder -- transitioning pipeline to GST_STATE_NULL
[gstreamer] gstEncoder -- pipeline stopped
[rtsp]   RTSP server on port 8554 is shutting down
[rtsp]   RTSP server thread stopped
video-viewer:  shutdown complete
developer@developer-desktop:~/ijGround$

Andreanbudi, 2024-02-03:
I have same issue

rahulkgupta, 2024-04-03:
I'm running into a similar issue. Were y'all able to solve it?

dusty-nv, 2024-04-03:
@rahulkgupta ended up using a higher-level connection manager like here:

https://github.com/dusty-nv/jetson-containers/blob/1a7fb07dd6183be744784a1c418799c1e0796ca3/packages/llm/local_llm/plugins/video.py#L83

rahulkgupta, 2024-04-17:
I'm not sure I follow. Are you saying it's solved in a recent version or to use the higher level connection manager?

dusty-nv, 2024-04-17:
Use a higher level connection manager

rahulkgupta, 2024-04-30:
it seems like something is happening lower level w/ gstreamer / videooutput. Specifically, the following:
```
[gstreamer] gstEncoder -- new caps: video/x-raw, width=1280, height=720, format=(string)I420, framerate=30/1
video-viewer:  captured 1 frames (1280x720)
[gstreamer] gstreamer message latency ==> encoder
video-viewer:  captured 2 frames (1280x720)
[gstreamer] gstreamer mysource ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline1/GstAppSrc:mysource:
streaming stopped, reason not-linked (-1)
```

The higher level connection manager you reference seems to help with the input side, but maybe i'm missing something (still new to all of this).