dusty-nv/jetson-inference #1793: Tried to Plot Centroid for Custom Object Detection but didn't work

ImTheMan153, 2024-02-05:
Hello,
I am trying a custom object detection and it is successfully working. However beyond drawing the overlay bbox, I tried to find the centre of the box and plot it as a circle so that I get a centroid right in the middle of the object. I tried to make changes to the source code (detectNet.cpp) from Jetson Inference C directory. Please let me know the problem to proceed further with my project.

https://forums.developer.nvidia.com/t/plotting-centroid-with-cuda/281924?u=nishanthrajkumar1

ImTheMan153, 2024-02-05:
[image]

ImTheMan153, 2024-02-05:
[image]

dusty-nv, 2024-02-06:
Hi @ImTheMan153, the `detectNet::Detection` struct has a Center() function which will return the centroid, and there are these CUDA functions for drawing basic shapes:

* https://github.com/dusty-nv/jetson-inference/blob/fe8b42c8da75c1c353dc59fa1fd079820024b89d/c/detectNet.h#L159
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#drawing-shapes

There are Python analogs of these too.

ImTheMan153, 2024-02-06:
Hi dusty,
Thank you for the reply. I changed the code to whatever you told me to go with, however, I get the same output as the gif shown above. If I remove the cuda circle loop, I am properly getting the overlay of the detection. If I use drawcircle function from jetson utils as per the repo you shared and make my build, it shows an error saying that it expects my input image to be some specific formats. I am attaching the error below.
`/home/ubuntu/jetson-inference/build/aarch64/include/jetson-utils/cudaDraw.h:67:68:   required from ‘cudaError_t cudaDrawCircle(T*, size_t, size_t, int, int, float, const float4&) [with T = void; cudaError_t = cudaError; size_t = long unsigned int; float4 = float4]’
/home/ubuntu/jetson-inference/c/detectNet.cpp:1024:13:   required from here
/home/ubuntu/jetson-inference/build/aarch64/include/jetson-utils/imageFormat.inl:234:48: error: static assertion failed: invalid image format type - supported types are uchar3, uchar4, float3, float4
  234 |  static_assert(__image_format_assert_false<T>::value, "invalid image format type - supported types are uchar3, uchar4, float3, float4");
      |                                                ^~~~~
make[2]: *** [CMakeFiles/jetson-inference.dir/build.make:1473: CMakeFiles/jetson-inference.dir/c/detectNet.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:447: CMakeFiles/jetson-inference.dir/all] Error 2`

Source code (just the overlay function) for now:
`if( flags & OVERLAY_BOX )
{   
    // Draw bounding box overlays once for all detections
    if( CUDA_FAILED(cudaDetectionOverlay(input, output, width, height, format, detections, numDetections, mClassColors)) )
        return false;

    // Then iterate over each detection to draw the centroid circles
    for( int n=0; n < numDetections; n++ )
    {   
        const Detection* d = detections + n;
        
        // Use the Center function to calculate the center coordinates
        float cx, cy;
        d->Center(&cx, &cy);
        
        // Adjust the radius as needed
        float radius = 12.5f;

        // Draw the centroid circle for the current detection
        // Ensure that the format of the image and the call to cudaDrawCircle are compatible
        if( CUDA_FAILED(cudaDrawCircle(input, output, width, height, format, static_cast<int>(cx), static_cast<int>(cy), radius, make_float4(255,0,0,200))))
            return false;
    }
}`
I tried a different cuda draw circle function according to the source code shown above, but it works the same way as the gif. 

If you know the fix, please send me the modified function. I will try it out and let you know!

ImTheMan153, 2024-02-06:
```
if( flags & OVERLAY_BOX )
{   
    // Draw bounding box overlays once for all detections
    if( CUDA_FAILED(cudaDetectionOverlay(input, output, width, height, format, detections, numDetections, mClassColors)) )
        return false;
    
    // Then iterate over each detection to draw the centroid circles
    for( int n=0; n < numDetections; n++ )
    {   
        const Detection* d = detections + n;
        
        // Use the Center function to calculate the center coordinates
        float cx, cy;
        d->Center(&cx, &cy);
        
        // Adjust the radius as needed
        float radius = 12.5f; // Use this variable or decide if you want to use a fixed value

        // Draw the centroid circle for the current detection
        if( CUDA_FAILED(cudaDrawCircle((uchar3*)output, width, height, static_cast<int>(cx), static_cast<int>(cy), radius, make_float4(255,0,0,200)))) // Corrected line
            return false;
    }
}
```

Just simply casting the output to one of the supported formats worked! Thank you.

closed 2024-02-06 by ImTheMan153