dusty-nv/jetson-inference #1794: detectnet unable to use other network(peoplenet/ssd-mobilenet-v1) with latest git version

lida2003, 2024-02-10:
Hi

I found that detectnet failed using peoplenet/ssd-mobilenet-v1 , but it's OK with ssd-mobilenet-v2(default).

Any idea?

```
$ git log -n 1
commit fe8b42c8da75c1c353dc59fa1fd079820024b89d (grafted, HEAD -> master, origin/master, origin/HEAD)
Author: Dustin Franklin <dustinf@nvidia.com>
Date:   Wed Jan 3 10:33:22 2024 -0500

    Merge pull request #1769 from ABADY1000/patch-1
    
    Update detectnet-camera-2.md to add track parameter to --overlay options
```

- peoplenet failed

```
daniel@daniel-nvidia:~/Jetson/jetson-inference/build/aarch64/bin$ ./detectnet ../../../../mobile_car_ext2.mp4 --network=peoplenet
[gstreamer] initialized gstreamer, version 1.20.3.0
[gstreamer] gstDecoder -- creating decoder for ../../../../mobile_car_ext2.mp4
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NvMMLiteBlockCreate : Block : BlockType = 261 

(detectnet:9409): GStreamer-CRITICAL **: 17:32:48.769: gst_debug_log_valist: assertion 'category != NULL' failed

(detectnet:9409): GStreamer-CRITICAL **: 17:32:48.769: gst_debug_log_valist: assertion 'category != NULL' failed

(detectnet:9409): GStreamer-CRITICAL **: 17:32:48.769: gst_debug_log_valist: assertion 'category != NULL' failed

(detectnet:9409): GStreamer-CRITICAL **: 17:32:48.769: gst_debug_log_valist: assertion 'category != NULL' failed
[gstreamer] gstDecoder -- discovered video resolution: 1280x720  (framerate 25.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)high, width=(int)1280, height=(int)720, framerate=(fraction)25/1, pixel-aspect-ratio=(fraction)1/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=../../../../mobile_car_ext2.mp4 ! qtdemux ! queue ! h264parse ! nvv4l2decoder name=decoder enable-max-performance=1 ! video/x-raw(memory:NVMM) ! nvvidconv name=vidconv ! video/x-raw ! appsink name=mysink
[video]  created gstDecoder from file:///home/daniel/Jetson/jetson-inference/build/aarch64/bin/../../../../mobile_car_ext2.mp4
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///home/daniel/Jetson/jetson-inference/build/aarch64/bin/../../../../mobile_car_ext2.mp4
     - protocol:  file
     - location:  ../../../../mobile_car_ext2.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      H264
  -- codecType:  v4l2
  -- width:      1280
  -- height:     720
  -- frameRate:  25
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
[... 198 lines cut ...]
                -- dim #0  1
                -- dim #1  100
                -- dim #2  7
[TRT]       binding 2
                -- index   2
                -- name    'Postprocessor_1'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  3
                -- dim #0  1
                -- dim #1  1
                -- dim #2  1
[TRT]    
[TRT]    binding to input 0 Input  binding index:  0
[TRT]    binding to input 0 Input  dims (b=1 c=3 h=300 w=300) size=1080000
[TRT]    binding to output 0 Postprocessor  binding index:  1
[TRT]    binding to output 0 Postprocessor  dims (b=1 c=1 h=100 w=7) size=2800
[TRT]    3: Cannot find binding of given name: PostProcessor_1
[TRT]    failed to find requested output layer PostProcessor_1 in network
[TRT]    device GPU, failed to create resources for CUDA engine
[TRT]    failed to create TensorRT engine for networks/SSD-Mobilenet-v1/ssd_mobilenet_v1_coco.uff, device GPU
[TRT]    detectNet -- failed to initialize.
detectnet:  failed to load detectNet model

```

wangke5437, 2024-08-21:
hello, have you fixed the problem, I met the same issue

lida2003, 2024-08-28:
@wangke5437 Network issue, use a proxy or download model mannually.

closed 2024-08-28 by lida2003