dusty-nv/jetson-inference #1798: posedge.py cannot be found

monophonics, 2024-02-25:
I am an AI newbie! I am navigating "Hello Embedded AI" course. I am using a Jetson Nano for my development which has SD card version  4.6.1.
I am stuck here:
-   https://github.com/dusty-nv/jetson-inference/blob/master/docs/posenet.md

**_I am not finding posenet in my container folders. So I cannot find this command recommended in the text:_**

# C++
$ ./posenet "images/humans_*.jpg" images/test/pose_humans_%i.jpg

# Python
$ ./posenet.py "images/humans_*.jpg" images/test/pose_humans_%i.jpg

When I am outside of container, I can see "posenet.py" under "jetson-inference/python/examples" folder but as soon I enter container, this file will not be available. 
All the folders are up to date per "git pull" results. Even though I did not clone a new container folder because it overwrites the container and I don't want to lose my images that I worked with.

I also tried to run "downlaod-models.sh" under tools folder but I am not seeing posenet or Pose-DenseNet121-Body or something similar. 

I also looked at the issue number 1183 here: https://github.com/dusty-nv/jetson-inference/issues/1183
 even though I am one step behind because I am not even seeing the script! 
I don't see Pose-DenseNet121-Body folder and this is probably the first thing that I need to resolve? 

How and where can I download these Pose-XYZ models? I have downloded all the Resnet pre-trained models

dusty-nv, 2024-02-25:
@monophonics in container, jetson-inference code can be found under `/jetson-inference` and the posenet binaries are already installed under `/usr/local/bin`  so you should just be able to run them like `posenet --help` and `posenet.py --help` regardless of your current working directory. 

The `download-models.sh` tool is no longer needed as the models are downloaded on-demand for you by the library the first time you try to load them.

monophonics, 2024-02-25:
Thanks @dusty-nv. 
I am not seeing posenet.py file under /usr/local/bin when I am inside the container. This is what I am seeing:

[image]

I followed this link to setup and run container from scratch:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md
and also this video:
https://www.youtube.com/watch?v=QXIwdsyK7Rw&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=10
 I still cannot see posenet.py under /usr/local/bin or other folders when I am inside the container.

dusty-nv, 2024-02-25:
Ah okay, I wonder if the container you are running is older than jetson-inference master - you are missing actionnet too.   Try building/installing it from source outside container, and then you should have it natively on your device in `/usr/local/bin` (and also your `jetson-inference/build/aarch64/bin` directory):

https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md

monophonics, 2024-02-26:
Thanks. I can make it working with following the [link](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md) and build from source instead of using the container.

closed 2024-02-26 by monophonics