dusty-nv/jetson-inference #1800: Don't render GUI when running output.render()

taher968, 2024-02-28:
I am trying to run a custom script on my Jetson Nano which takes an rtp stream as input and outputs a webrtc stream and it is working correctly. The only issue is that I always need to connect the Jetson to a screen as it throws OpenGL/CUDA errors when I don't. I don't necessarily need the GUI display as I am outputting a webrtc screen which I can view a browser. Is there any way of the Jetson to not render a GUI when running `output.Render()` and just output to the stream. I want to be able to do so even when connected through SSH.

tianyatianya, 2024-03-06:
+1, I have the same problem.

dusty-nv, 2024-03-06:
@taher968 @tianyatianya yes, disable X11-forwarding on your SSH connection (or `unset DISPLAY`), or set `--headless` when you run the script or video option

closed 2024-03-18 by taher968