dusty-nv/jetson-inference #1801: fatal error: gst/rtsp-server/rtsp-server.h: No such file or directory

omrode1, 2024-02-29:
This is the error i am encountering after running 
`make - j$(nproc)`

[ 14%] Building CXX object utils/CMakeFiles/jetson-utils.dir/logging.cpp.o
[ 14%] Building CXX object utils/CMakeFiles/jetson-utils.dir/filesystem.cpp.o
[ 14%] Building CXX object utils/CMakeFiles/jetson-utils.dir/timespec.cpp.o
[ 15%] Building CXX object utils/CMakeFiles/jetson-utils.dir/xml.cpp.o
[ 15%] Building CXX object utils/CMakeFiles/jetson-utils.dir/commandLine.cpp.o
[ 16%] Building CXX object utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o
[ 16%] Building CXX object utils/CMakeFiles/jetson-utils.dir/camera/v4l2Camera.cpp.o
[ 17%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstBufferManager.cpp.o
[ 18%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o
[ 18%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstUtility.cpp.o
[ 20%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o
[ 20%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaBayer.cpp.o
[ 21%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaColorspace.cpp.o
[ 22%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaPointCloud.cpp.o
[ 22%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glBuffer.cpp.o
[ 23%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glCamera.cpp.o
[ 24%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o
[ 25%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glEvents.cpp.o
[ 25%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glTexture.cpp.o
[ 26%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glWidget.cpp.o
[ 27%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageIO.cpp.o
[ 27%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageLoader.cpp.o
[ 28%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageWriter.cpp.o
[ 29%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devInput.cpp.o
[ 29%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devJoystick.cpp.o
[ 30%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devKeyboard.cpp.o
[ 31%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/IPv4.cpp.o
[ 31%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/IPv6.cpp.o
[ 32%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/Networking.cpp.o
[ 33%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/RTSPServer.cpp.o
[ 33%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/Socket.cpp.o
[ 34%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/URI.cpp.o
/home/jetson/jetson-inference/utils/network/RTSPServer.cpp:30:10: fatal error: gst/rtsp-server/rtsp-server.h: No such file or directory
 #include <gst/rtsp-server/rtsp-server.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
utils/CMakeFiles/jetson-utils.dir/build.make:4859: recipe for target 'utils/CMakeFiles/jetson-utils.dir/network/RTSPServer.cpp.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/network/RTSPServer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:706: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

dusty-nv, 2024-02-29:
Hi @omrode1, what version of JetPack-L4T are you on?

This should have automatically been installed when you ran cmake, in the `libgstrtspserver-1.0-dev` package:

https://github.com/dusty-nv/jetson-inference/blob/3f0603a3957b8ccde59caa848792d427ad0a6742/CMakePreBuild.sh#L34

Can you try running `sudo apt-get install libgstrtspserver-1.0-dev` ?

omrode1, 2024-03-01:
cat /etc/nv_tegra_release
# R32 (release), REVISION: 5.0, GCID: 25531747, BOARD: t186ref, EABI: aarch64, DATE: Fri Jan 15 23:21:05 UTC 2021

i found this, does that anything to do with the l4t version or else there is an another command ?

omrode1, 2024-03-01:
so, the RTSP error seems to have been resolved using `sudo apt-get install libgstrtspserver-1.0-dev`

but the new error is being emerged and the following texts has been thrown out. 

[  1%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
utils/CMakeFiles/jetson-utils.dir/build.make:5196: recipe for target 'aarch64/lib/libjetson-utils.so' failed
make[2]: *** [aarch64/lib/libjetson-utils.so] Error 1
CMakeFiles/Makefile2:706: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

dusty-nv, 2024-03-01:
Hi @omrode1, do you have libGL.so under `/usr/lib/aarch64-linux-gnu` ?

```bash
$ find /usr -name 'libGL*'
/usr/local/cuda-10.2/samples/common/lib/linux/aarch64/libGLEW.a
/usr/local/cuda-10.2/samples/common/lib/linux/x86_64/libGLEW.a
/usr/local/cuda-10.2/samples/common/lib/linux/armv7l/libGLEW.a
/usr/lib/chromium-browser/swiftshader/libGLESv2.so
/usr/lib/chromium-browser/libGLESv2.so
/usr/lib/aarch64-linux-gnu/libGL.so.1.0.0
/usr/lib/aarch64-linux-gnu/libGLU.a
/usr/lib/aarch64-linux-gnu/libGLdispatch.so.0
/usr/lib/aarch64-linux-gnu/libGLEWmx.so.1.13
/usr/lib/aarch64-linux-gnu/libGLESv1_CM.so
/usr/lib/aarch64-linux-gnu/tegra/libGLX_nvidia.so.0
/usr/lib/aarch64-linux-gnu/libGLX.so.0.0.0
/usr/lib/aarch64-linux-gnu/libGL.so
/usr/lib/aarch64-linux-gnu/libGL.so.1
/usr/lib/aarch64-linux-gnu/libGLX_mesa.so.0
/usr/lib/aarch64-linux-gnu/libGLESv2.so.2.0.0
/usr/lib/aarch64-linux-gnu/libGLX.so.0
/usr/lib/aarch64-linux-gnu/libGLdispatch.so.0.0.0
/usr/lib/aarch64-linux-gnu/tegra-egl/libGLESv2_nvidia.so.2
/usr/lib/aarch64-linux-gnu/tegra-egl/libGLESv1_CM_nvidia.so.1
/usr/lib/aarch64-linux-gnu/libGLESv1_CM.so.1
/usr/lib/aarch64-linux-gnu/libGLX.so
/usr/lib/aarch64-linux-gnu/libGLESv1_CM.so.1.0.0
/usr/lib/aarch64-linux-gnu/libGLESv2.so.2
/usr/lib/aarch64-linux-gnu/libGLdispatch.so
/usr/lib/aarch64-linux-gnu/libGLU.so.1
/usr/lib/aarch64-linux-gnu/libGLX_mesa.so.0.0.0
/usr/lib/aarch64-linux-gnu/libGLEWmx.so.1.13.0
/usr/lib/aarch64-linux-gnu/libGLU.so.1.3.1
/usr/lib/aarch64-linux-gnu/libGLESv2.so
/usr/lib/aarch64-linux-gnu/libGLX_indirect.so.0
/usr/lib/aarch64-linux-gnu/libGLU.so
```