dusty-nv/jetson-inference #1802: WebRTC Server issues

pq53ui, 2024-03-02:
- Jetson Nano
- JetPack 4.6
- jetson-inference (main branch, cloned about a month ago)
- Built code from source, not using any dockers 
- Both Jetson Nano and Laptop I am accessing the webrtc stream are on local network.

I haven't been able to find a similar issue to what I am experiencing.

Running either detectnet, segnet, or my custom script I am unable to reliably connect to the webrtc output in browser.
I followed the tutorial and tried both the secure version with SSL keys and no keys. 

When I run the script, while the models and streams are initializing I open the browser and access the output stream. Sometimes it is successful sometimes all I get is a gray screen with video trying to load.

[image]

So, it seems like there is some sweet spot while I can access the stream, but I have no idea when exactly that is. Sometime between initialization. While it is in the image capture loop, it is impossible to see any output. 

** This is the output when I run the script and connect to the stream. I removed the [TRT] logs and others that have nothing to do with webrtc or gstreamer in my opinion. I don't know much about web related things, so any help would be much appreciated. 

```

[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      raw
  -- codecType:  cpu
  -- width:      640
  -- height:     360
  -- frameRate:  30
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc name=encoder bitrate=4000000 insert-sps-pps=1 insert-vui=1 ! video/x-h264 ! rtph264pay config-interval=1 ! application/x-rtp,media=video,encoding-name=H264,clock-rate=90000,payload=96 ! tee name=videotee ! queue ! fakesink
[webrtc] WebRTC server started @ http://jetson-desktop:8554
[webrtc] websocket route added /output
[video]  created gstEncoder from webrtc://@:8554/output
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: webrtc://@:8554/output
     - protocol:  webrtc
     - location:  0.0.0.0
     - port:      8554
  -- deviceType: ip
  -- ioType:     output
  -- codec:      H264
  -- codecType:  omx
[webrtc] WebRTC server thread running...
  -- width:      640
  -- height:     360
  -- frameRate:  30
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- latency     10
------------------------------------------------

[webrtc] HTTP 192.168.1.5 GET '/'
[webrtc] /
[webrtc] HTTP 192.168.1.5 GET '/favicon.ico'
[... 102 lines cut ...]
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:7 2 UDP 2013266428 2a00:ee2:900:6500:5d99:bf18:b020:7b34 50122 typ host"}}
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:8 2 TCP 1015022334 2a00:ee2:900:6500:5d99:bf18:b020:7b34 9 typ host tcptype active"}}
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:9 2 TCP 1010828030 2a00:ee2:900:6500:5d99:bf18:b020:7b34 38667 typ host tcptype passive"}}
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:10 2 UDP 2013266427 172.17.0.1 56839 typ host"}}
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:11 2 TCP 1015022590 172.17.0.1 9 typ host tcptype active"}}
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:12 2 TCP 1010828286 172.17.0.1 48383 typ host tcptype passive"}}
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:13 2 UDP 2013266426 192.168.1.4 53093 typ host"}}
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:14 2 TCP 1015022846 192.168.1.4 9 typ host tcptype active"}}
[webrtc] sending ICE candidate for /output to 192.168.1.5 (peer_id=0): 
{"type":"ice","data":{"sdpMLineIndex":0,"candidate":"candidate:15 2 TCP 1010828542 192.168.1.4 56675 typ host tcptype passive"}}
[webrtc] resolved 6593b3ef-688b-4aea-a0e7-ec8281fd24bd.local for 192.168.1.5 in incoming ICE message
[webrtc] candidate:1263540956 1 udp 2113937151 192.168.1.5 44018 typ host generation 0 ufrag tp+0 network-cost 999
[webrtc] websocket /output -- recieved message from 192.168.1.5 (peer_id=0) (236 bytes)
[webrtc] received ICE message on /output from 192.168.1.5 (peer_id=0) with mline index 0; candidate: 
candidate:3086921863 1 udp 2113939711 40925334-02af-46e9-9d8a-9b1b36f8da7d.local 49420 typ host generation 0 ufrag tp+0 network-cost 999
[gstreamer] gstreamer stream status CREATE ==> src
```

pq53ui, 2024-03-02:
Thanks for the quick response.

I ran the video-viewer.py
Same issue, I get the same gray screen as in the screenshot. 
I have tried enabling the chrome flag as well, no difference. 

I am getting the same output as I have posted in the above

Maybe a few additional outputs

```
[webrtc] resolved 5ff934f4-6ebf-4a7c-984c-b4a83b1134df.local for 192.168.1.5 in incoming ICE message
[webrtc] candidate:2191427754 1 udp 2113937151 192.168.1.5 57232 typ host generation 0 ufrag l8h+ network-cost 999
[webrtc] websocket /output -- recieved message from 192.168.1.5 (peer_id=0) (236 bytes)
[webrtc] received ICE message on /output from 192.168.1.5 (peer_id=0) with mline index 0; candidate: 
candidate:2128326268 1 udp 2113939711 9aed251a-ffef-4d2c-a1bd-845889b8c5e0.local 54438 typ host generation 0 ufrag l8h+ network-cost 999
[gstreamer] gstEncoder -- pipeline full, skipping frame 25 (640x360, 345600 bytes)
video-viewer:  captured 25 frames (640 x 360)
```

I have also seen 
```
[network] getHostByName() trying to resolve host '9aed251a-ffef-4d2c-a1bd-845889b8c5e0.local' (retry 1 of 10)
```
this line a few times before. Trying to understand if that plays any role as well. 

Its bugging me that it sometimes works under certain (unknown) conditions. But in most cases I can't get it to work.

dusty-nv, 2024-03-02:
> I have also seen
> 
> ```
> [network] getHostByName() trying to resolve host '9aed251a-ffef-4d2c-a1bd-845889b8c5e0.local' (retry 1 of 10)
> ```
> 
> this line a few times before. Trying to understand if that plays any role as well.

Ok yea, that is the mDNS thing - if you disable that chrome flag and restart the browser, you shouldn't get that anymore

> [gstreamer] gstEncoder -- pipeline full, skipping frame 25 (640x360, 345600 bytes)

Try commenting out this if block of code, and recompiling/reinstalling:

https://github.com/dusty-nv/jetson-utils/blob/5a5bb422eada7fdbd7d0f53806cf8e0889f2352f/codec/gstEncoder.cpp#L515

Sorry that WebRTC can be finicky sometimes with the connection/negotiation, and the gstreamer plugins for it aren't super robust

pq53ui, 2024-03-02:
Disabling that chrome flag helped. I am able to see the stream and reconnect to it while the capture loop is going too!

Thank you! 
I am mostly working in headless mode, so `pipeline full` doesn't worry me too much right now. But it's good to know how to make it disappear. 

And again, thanks for quick response.

closed 2024-03-03 by pq53ui

dgdgksj, 2024-04-05:
> Ok yea, that is the mDNS thing - if you disable that chrome flag and restart the browser, you shouldn't get that anymore
>
>
> Try commenting out this if block of code, and recompiling/reinstalling: [...]

Thanks @dusty-nv  Just by disabling the "Anonymize local IPs exposed by WebRTC" option in chrome://flags/, I got it resolved.