dusty-nv/jetson-inference #1807: How to convert data into BCHW format

lofyol, 2024-03-09:
Thank you for the great work of the author.  After resizing with jetson_utils, how do I convert the HWC format to BCHW format data?

dusty-nv, 2024-03-09:
Hi @lofyol , this is done with the kernels in `c/tensorConvert.cu`, although these aren't usually exposed API's:

https://github.com/dusty-nv/jetson-inference/blob/master/c/tensorConvert.cu

lofyol, 2024-03-09:
Thank you.  Is there a conversion interface for python?

lofyol, 2024-03-09:
@dusty-nv Thank you. Is there a conversion interface for python?

dusty-nv, 2024-03-09:
No, I don't expose those to python:  you could try `torch.as_tensor()` and just do it in PyTorch:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#cuda-array-interface

lofyol, 2024-03-09:
OK，thank you~

closed 2024-03-09 by lofyol