dusty-nv/jetson-inference #1810: docker space full

beneslami, 2024-03-12:
Hi. 
Even though I have enough space to host large files, the process of downloading, extracting and pulling docker files ends up with "not enough disk space" error. Note that,  run the docker with below command:

```sudo dockerd --storage-driver=vfs --iptables=False &```

if I don't set the storage driver to vas, it returns error "overlay2". 
Is it something related to vfs ?

dusty-nv, 2024-03-12:
@beneslami no idea about other storage drivers or if it would work with the already built container images - you will either need to mount additional storage and relocate your docker data-root to it (https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#relocating-docker-data-root), or to just build jetson-inference from source and not use container.

beneslami, 2024-03-14:
Hi @dusty-nv 
Thanks for the reply. I am specifically working with [r32.7.1](https://hub.docker.com/layers/dustynv/jetson-inference/r32.7.1/images/sha256-7ee3ab60052aa2372e4d17ba428eeac4f34fb16ffec5dec6b5cb984495d42d58?context=explore). The thing is the total size of the container is around 1GB but when it comes to downloading/extracting/pulling, it consumes total storage and stop the process. Is it a typical thing in docker? There is no other way than building the source ?

dusty-nv, 2024-03-14:
Yea it downloads a compressed tar of the container image first, so it needs that temp space.  The other way is to mount more storage.