dusty-nv/jetson-inference #1814: NVIDIA GeForce RTX 4060 Laptop GPU  is not yet supported in this version of the container

DaXiaWei, 2024-03-20:
This container image and its contents are governed by the NVIDIA Deep Learning Container License.
By pulling and using the container, you accept the terms and conditions of this license:
https://developer.nvidia.com/ngc/nvidia-deep-learning-container-license
WARNING: Detected NVIDIA NVIDIA GeForce RTX 4060 Laptop GPU GPU, which is not yet supported in this version of the container
ERROR: No supported GPU(s) detected to run this container

dusty-nv, 2024-03-23:
@DaXiaWei it has been a long time since I build/tested the x86 version of jetson-inference container, and you would need to rebuild it against a newer version of NGC pytorch base container image:

https://github.com/dusty-nv/jetson-inference/blob/644f5158a55a7f313f4d43b57ad3ed109fcbf9ab/docker/tag.sh#L37

DaXiaWei, 2024-03-26:
Sure, thank you for your response.

closed 2024-03-26 by DaXiaWei