dusty-nv/jetson-inference #1819: SyntaxError: future feature annotations is not defined

bskkim1999, 2024-03-29:
my environment is
jetson nano
jetpack 4.4 
python3.6.9
======================

I issued these below commands.

$ cd jetson-inference/build
$ ./install-pytorch.sh

but there was error that I don't know how to solve...

[Shell terminal]
Installed /usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0+78ed10c-py3.6-linux-aarch64.egg
Processing dependencies for torchvision==0.7.0a0+78ed10c
Searching for pillow>=4.1.1
Reading https://pypi.org/simple/pillow/
/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py:119: PkgResourcesDeprecationWarning:  is an invalid version and will not be supported in a future release
  PkgResourcesDeprecationWarning,
/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py:119: PkgResourcesDeprecationWarning: rc1 is an invalid version and will not be supported in a future release
  PkgResourcesDeprecationWarning,
Downloading https://files.pythonhosted.org/packages/f8/3e/32cbd0129a28686621434cbf17bb64bf1458bfb838f1f668262fefce145c/pillow-10.2.0.tar.gz#sha256=e87f0b2c78157e12d7686b27d63c070fd65d994e8ddae6f328e0dcf4a0cd007e
Best match: pillow 10.2.0
Processing pillow-10.2.0.tar.gz
Writing /tmp/easy_install-wbi20k_2/pillow-10.2.0/setup.cfg
Running pillow-10.2.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-wbi20k_2/pillow-10.2.0/egg-dist-tmp-frg3dziw
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 156, in save_modules
    yield saved
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 198, in setup_context
    yield
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 259, in run_setup
    _execfile(setup_script, ns)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 45, in _execfile
    code = compile(script, filename, 'exec')
  File "/tmp/easy_install-wbi20k_2/pillow-10.2.0/setup.py", line 10
    from __future__ import annotations
    ^
SyntaxError: future feature annotations is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "setup.py", line 255, in <module>
    'clean': clean,
  File "/usr/local/lib/python3.6/dist-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py", line 74, in run
    self.do_egg_install()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py", line 124, in do_egg_install
    cmd.run(show_deprecation=False)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 414, in run
    self.easy_install(spec, not self.no_deps)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 656, in easy_install
    return self.install_item(None, spec, tmpdir, deps, True)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 703, in install_item
    self.process_distribution(spec, dist, deps)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 751, in process_distribution
    [requirement], self.local_index, self.easy_install
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 773, in resolve
    replace_conflicting=replace_conflicting
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 1056, in best_match
    return self.obtain(req, installer)
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 1068, in obtain
    return installer(requirement)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 675, in easy_install
    return self.install_item(spec, dist.location, tmpdir, deps)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 701, in install_item
    dists = self.install_eggs(spec, download, tmpdir)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 896, in install_eggs
    return self.build_and_install(setup_script, setup_base)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 1168, in build_and_install
    self.run_setup(setup_script, setup_base, args)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/easy_install.py", line 1152, in run_setup
    run_setup(setup_script, args)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 262, in run_setup
    raise
  File "/usr/lib/python3.6/contextlib.py", line 99, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 198, in setup_context
    yield
  File "/usr/lib/python3.6/contextlib.py", line 99, in __exit__
    self.gen.throw(type, value, traceback)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 169, in save_modules
    saved_exc.resume()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 143, in resume
    raise exc.with_traceback(self._tb)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 156, in save_modules
    yield saved
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 198, in setup_context
    yield
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 259, in run_setup
    _execfile(setup_script, ns)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/sandbox.py", line 45, in _execfile
    code = compile(script, filename, 'exec')
SyntaxError: future feature annotations is not defined
 
[jetson-inference]  installation complete, exiting with status code 0
[jetson-inference]  to run this tool again, use the following commands:
 
    $ cd <jetson-inference>/build
    $ ./install-pytorch.sh

=======================================================
Can you recommened how to solve this issue..? Thanks :)

dusty-nv, 2024-03-29:
@bskkim1999 can you try running `pip3 install 'pillow<7'` ?

bskkim1999, 2024-03-30:
@dusty-nv 
Before instructing ./install-pytorch.sh, I executed the command pip3 install 'pillow<7', so there are no errors anymore! Thank you for your kindness!