dusty-nv/jetson-inference #1820: where is file named train_ssd.py ?

bskkim1999, 2024-03-29:
my environment is 
jetpack 4.4 , python 3.6.9, torch1.6.0, torchvision 0.7.0a0+78ed10c

======================

I would like to train my detection dataset.  So, I found train_ssd.py to train after collecting my pictures taken by camera-capture program. below url is what I saw.
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md

But,, there is no train_ssd.py in jetson-inference/python/training/detection/ssd
How can I get train_ssd.py ? I built the project from source not docker

dusty-nv, 2024-03-29:
@bskkim1999 when you cloned the repo, did you use `--recursive` ?

If not, you can use `git submodule update --init`

https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md#cloning-the-repo

bskkim1999, 2024-03-30:
@dusty-nv 
I used "git submodule update --init" not "--recursive".
But I can execute "train_ssd.py" by clone "pytorch-ssd" you uploaded.
https://github.com/dusty-nv/pytorch-ssd/tree/6accaa88845ec135a7d6fe25e9a26afd4698639d
anyway, thank you for your help!!