dusty-nv/jetson-inference #1823: How to open csi camera in a docker container

AnhHazz, 2024-04-03:
Hi @dusty-nv 
I'm using a ros:humble docker container and I want to send camera images through ros2 topic. The problem is this container doesn't have nvgstreamer so I can't use: 
`
std::string gstreamer_pipeline (int capture_width, int capture_height, int display_width, int display_height, int framerate, int flip_method) {
    return "nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)" + std::to_string(capture_width) + ", height=(int)" +
           std::to_string(capture_height) + ", framerate=(fraction)" + std::to_string(framerate) +
           "/1 ! nvvidconv flip-method=" + std::to_string(flip_method) + " ! video/x-raw, width=(int)" + std::to_string(display_width) + ", height=(int)" +
           std::to_string(display_height) + ", format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink";
}
`

`
    std::string pipeline = gstreamer_pipeline(capture_width,
	capture_height,
	display_width,
	display_height,
	framerate,
	flip_method);
    std::cout << "Using pipeline: \n\t" << pipeline << "\n";
`

`
    cv::VideoCapture cap(pipeline, cv::CAP_GSTREAMER);`

Is there any way I can use the csi camera? 
As I know, camera devices always have their name inside /dev folder so we can read from the device. i searched but haven't found where csi camera in that folder. I only found /dev/video0. Is that it? Tks for helping!

dusty-nv, 2024-04-03:
Hi @AnhHazz, the `/dev/video*` devices are for V4L2 interface, instead the Argus CSI daemon communicates over this socket mounted into the container:

https://github.com/dusty-nv/jetson-inference/blob/644f5158a55a7f313f4d43b57ad3ed109fcbf9ab/docker/run.sh#L205

AnhHazz, 2024-04-03:
@dusty-nv Thanks for reply i see, but i still don't know what to do. Can you tell me exactly what steps in order to run the csi camera inside a container? Or how to install nvgstreamer inside the container?

I don't use your docker images so those nvgst commands are not available now

dusty-nv, 2024-04-04:
@AnhHazz I think you need to have gstreamer installed inside the container and be using `--runtime nvidia` to enable the nvidia plugins for gstreamer