dusty-nv/jetson-inference #1825: Problem with testing video viewer function

EmilienRNN, 2024-04-11:
Hello,

I have just discovered this repository and I wanted to try WebRTC for watching my V4L2 camera on browser. I have tried this function :
_video-viewer /dev/video0 webrtc://@:8554/output_ function. This is creating a new window on my Jetson with the camera no problem. But, when i want to watch it through the wab browser, there is the template like in the example but i don't see any video getting displayed.
I have also tried the other functions/applications for testing webrtc (flask, html, server) but it resulted in the same problem.
I am using e-con camera STURDeCAM31_CUOAGX.
Thanks for your help.

dusty-nv, 2024-04-11:
Hi @EmilienRNN, can you try chrome or chromium browser, and disable `chrome://flags#enable-webrtc-hide-local-ips-with-mdns` ?

dusty-nv, 2024-04-11:
@EmilienRNN that is on whatever client device you are running the browser on (could be a jetson).   Just navigate to `chrome://flags` , then search `enable-webrtc-hide-local-ips-with-mdns`, disable it, and restart the browser.   Then navigate to your jetson's port 8554

dusty-nv, 2024-04-11:
With firefox there is a webrtc codec issue I've been unable to figure out, and honestly webrtc be super complicated with all the client support, so for now I recommend just using chrome.  I have tested it on android with chrome and an iPad with Safari

EmilienRNN, 2024-04-21:
Thanks a lot about your help ! Is there a special android version in order to have the compatible chrome application ?

dusty-nv, 2024-04-24:
@EmilienRNN I had tried it previously from Android phone using stock Chrome browser, and it worked.  Also worked on iPad with Safari the last I tried it.