dusty-nv/jetson-inference #1833: Is there a way to manually EOS in python object?

erfansafaie, 2024-04-22:
I have two questions:
1- How we can send EOS event or closing the stream manually in python? Or any object in c++? (without OpenGl window)

2- Is OpenGL window run in main thread? I have a Qt app and I use videoSource and videoOutput objects. When I clicked start recording video in my app it will recording in other thread. Finally, I close OpenGL window, recorded video is saved but OpenGL window not close and freeze! It will close when I close my app. So I think showing window is running on main thread?!

dusty-nv, 2024-04-23:
Hi @erfansafaie, the OpenGL window runs in the same thread as from where you call it from - you can find the source here:

https://github.com/dusty-nv/jetson-utils/blob/af7c667c6367fe6599d8046c213d0cb3b8221ae3/display/glDisplay.cpp

Also to close videoOutput interface, call `videoOutput->Close()` - for GStreamer pipelines, this will gracefully shut them down and send EOS (so the video files/ect should be readable)

erfansafaie, 2024-05-07:
I appreciate your response and help.
Is it work in python same(videoOutput.Close())? I read the binding codes and close function, I think it's not working like destructor in C++.