dusty-nv/jetson-inference #1837: Jetson Inference Docker using Ubuntu 22.04 and TensorRT 10.0.1

Fibo27, 2024-05-10:
Hi @dusty-nv : I am trying to create a docker image that includes Jetson Inference together with ROS2 humble on a x86.

My System: x86, Ubuntu 22.04, CUDA 12.4, CuDNN 9, TensorRT 10.0.1, ROS: humble-desktop

Since your containers do not support the above configuration, I started with the osrf: humble-desktop image and then was able to build an image that included the above CUDA, CuDNN and TensorRT versions together with humble-desktop (I call this Image-1). For the above i used the debian-repo packages for CUDA, CuDNN and TensorRT. As an example, for TensorRT i used: https://docs.nvidia.com/deeplearning/tensorrt/install-guide/index.html#installing

I would now want to create another image (Image-2) that includes Jetson-Inference using the Image-1 as my base image. I do realize that the Ubuntu, CUDA, CuDNN and TensorRT versions used by you in your docker images are of earlier versions.

While building these, I changed the CMakeLists.txt file in your main folder (jetson-inference) and used the same code piece as you have in your Dockerfile: `sed -i 's/nvcaffe_parser/nvparsers/g' CMakeLists.txt`. I also made the same change for the CMakeLists.txt files inside  /tools/trt-bench and /tools/trt-console.

When i build, it stops when it tries to look for "NvCaffeParser.h" in jetson-inference/c/tensorNet.cpp. Since I am using Image-1 as the base for Image-2, I checked inside directory /usr/include/x86_64-linux-gnu and i cannot find NvCaffeParser. Can you please help? It appears that this file doesn't get created during the TensorRT build.

FYI, I have built jetson-inference in my system  x86, Ubuntu 22.04, CUDA 12.1, CuDNN 8.6, TensorRT 8.8 and it works fine. Now I am trying to create a Dockerfile albeit with other versions of CUDA, CuDNN and TensorRT.
Thanks

dusty-nv, 2024-05-10:
Hi @Fibo27, TRT10 had some breaking changes from removing deprecated APIs, and I haven't updated the code for it yet

Fibo27, 2024-05-11:
Thanks @dusty-nv - will look forward to that when you do it. In the meanwhile, I have managed to create a Dockerfile using the base image here: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/Dockerfile.x86_64.ros2_humble
Thanks Again

closed 2024-05-11 by Fibo27