dusty-nv/jetson-inference #1838: Segnet fails to load when using DLA, but PASSED using TensorRT.trtexec

khsafkatamin, 2024-05-10:
Hi @dusty-nv,

I am using a custom segnet model trained following the steps from [Onixaz Pytorch Segmentation](https://www.highvoltagecode.com/post/edge-ai-semantic-segmentation-on-nvidia-jetson). I can run the model using Device GPU without any issues. But When I run with the Device DLA. I face the following issue...

```

[TRT]    =============== Computing costs for 
[TRT]    *************** Autotuning format combination: Half(1572864,524288,1024,1) -> Half(6144,512,32,1) ***************
[TRT]    --------------- Timing Runner: {ForeignNode[/backbone/conv1/Conv.../classifier/classifier.4/Conv]} (DLA)
[TRT]    Skipping tactic 0x0000000000000003 due to exception Assertion context.dlaContext != nullptr failed. 
[TRT]    Fastest Tactic: 0xd15ea5edd15ea5ed Time: inf
[TRT]    *************** Autotuning format combination: Half(1572864,524288,1024,1) -> Half(512,512:16,32,1) ***************
[TRT]    --------------- Timing Runner: {ForeignNode[/backbone/conv1/Conv.../classifier/classifier.4/Conv]} (DLA)
[TRT]    Skipping tactic 0x0000000000000003 due to exception Assertion context.dlaContext != nullptr failed. 
[TRT]    Fastest Tactic: 0xd15ea5edd15ea5ed Time: inf
[TRT]    *************** Autotuning format combination: Half(524288,1:4,1024,1) -> Half(6144,512,32,1) ***************
[TRT]    --------------- Timing Runner: {ForeignNode[/backbone/conv1/Conv.../classifier/classifier.4/Conv]} (DLA)
[TRT]    Skipping tactic 0x0000000000000003 due to exception Assertion context.dlaContext != nullptr failed. 
[TRT]    Fastest Tactic: 0xd15ea5edd15ea5ed Time: inf
[TRT]    *************** Autotuning format combination: Half(524288,1:4,1024,1) -> Half(512,512:16,32,1) ***************
[TRT]    --------------- Timing Runner: {ForeignNode[/backbone/conv1/Conv.../classifier/classifier.4/Conv]} (DLA)
[TRT]    Skipping tactic 0x0000000000000003 due to exception Assertion context.dlaContext != nullptr failed. 
[TRT]    Fastest Tactic: 0xd15ea5edd15ea5ed Time: inf
[TRT]    *************** Autotuning format combination: Half(524288,524288:16,1024,1) -> Half(6144,512,32,1) ***************
[TRT]    --------------- Timing Runner: {ForeignNode[/backbone/conv1/Conv.../classifier/classifier.4/Conv]} (DLA)
[TRT]    Skipping tactic 0x0000000000000003 due to exception Assertion context.dlaContext != nullptr failed. 
[TRT]    Fastest Tactic: 0xd15ea5edd15ea5ed Time: inf
[TRT]    *************** Autotuning format combination: Half(524288,524288:16,1024,1) -> Half(512,512:16,32,1) ***************
[TRT]    --------------- Timing Runner: {ForeignNode[/backbone/conv1/Conv.../classifier/classifier.4/Conv]} (DLA)
[TRT]    Skipping tactic 0x0000000000000003 due to exception Assertion context.dlaContext != nullptr failed. 
[TRT]    Fastest Tactic: 0xd15ea5edd15ea5ed Time: inf
[TRT]    10: [optimizer.cpp::computeCosts::3728] Error Code 10: Internal Error (Could not find any implementation for node {ForeignNode[/backbone/conv1/Conv.../classifier/classifier.4/Conv]}.)
[TRT]    device DLA_0, failed to build CUDA engine
[TRT]    device DLA_0, failed to load fcn_resnet18.onnx
[TRT]    segNet -- failed to load.
segnet:  failed to initialize segNet

```
I tested with `trtexec`. it does not show any error. 

Can you please tell me what is the meaning of these errors? 

- What does it mean to provide a valid calibrator? To run it on DLA, I only changed the device type to DEVICE_DLA [here](https://github.com/dusty-nv/jetson-inference/blob/master/c/segNet.h#L125-L144), do I have to change anything else?
`[TRT]    requested fasted precision for device DLA_0 without providing valid calibrator, disabling INT8`

- Why only the classifier 4 layer running on DLA?

```
[TRT]    ---------- Layers Running on DLA ----------
[TRT]    [DlaLayer] {ForeignNode[/backbone/conv1/Conv.../classifier/classifier.4/Conv]}
[TRT]    ---------- Layers Running on GPU ----------
[TRT]    Trying to load shared library libcublas.so.11
[TRT]    Loaded shared library libcublas.so.11
[TRT]    Using cublas as plugin tactic source
[TRT]    Trying to load shared library libcublasLt.so.11
[TRT]    Loaded shared library libcublasLt.so.11
[TRT]    Using cublasLt as core library tactic source
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +260, GPU +322, now: CPU 660, GPU 5053 (MiB)
[TRT]    Trying to load shared library libcudnn.so.8
[TRT]    Loaded shared library libcudnn.so.8
[TRT]    Using cuDNN as plugin tactic source
[TRT]    Using cuDNN as core library tactic source
[TRT]    [MemUsageChange] Init cuDNN: CPU +82, GPU +125, now: CPU 742, GPU 5178 (MiB)
[TRT]    Global timing cache in use. Profiling results in this builder pass will be stored.
[TRT]    Constructing optimization profile number 0 [1/1].
[TRT]    Reserving memory for host IO tensors. Host: 0 bytes

```

- And what is the meaning of this error? I checked, all the layers are supported by DLA
[... 138 lines cut ...]
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.9028 ms - Host latency: 21.4042 ms (enqueue 0.437756 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.9199 ms - Host latency: 21.4204 ms (enqueue 0.409814 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.882 ms - Host latency: 21.3914 ms (enqueue 0.393835 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.936 ms - Host latency: 21.439 ms (enqueue 0.41676 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.8821 ms - Host latency: 21.3883 ms (enqueue 0.431421 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.9209 ms - Host latency: 21.4322 ms (enqueue 0.465869 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.9178 ms - Host latency: 21.425 ms (enqueue 0.431006 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.9407 ms - Host latency: 21.4461 ms (enqueue 0.395068 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 21.0056 ms - Host latency: 21.5125 ms (enqueue 0.414478 ms)
[05/10/2024-12:25:00] [I] Average on 10 runs - GPU latency: 20.9628 ms - Host latency: 21.4675 ms (enqueue 0.46853 ms)
[05/10/2024-12:25:00] [I] 
[05/10/2024-12:25:00] [I] === Performance summary ===
[05/10/2024-12:25:00] [I] Throughput: 47.467 qps
[05/10/2024-12:25:00] [I] Latency: min = 21.3491 ms, max = 21.6797 ms, mean = 21.4235 ms, median = 21.4022 ms, percentile(90%) = 21.5027 ms, percentile(95%) = 21.571 ms, percentile(99%) = 21.6519 ms
[05/10/2024-12:25:00] [I] Enqueue Time: min = 0.251709 ms, max = 0.654846 ms, mean = 0.43047 ms, median = 0.422363 ms, percentile(90%) = 0.566406 ms, percentile(95%) = 0.595459 ms, percentile(99%) = 0.63623 ms
[05/10/2024-12:25:00] [I] H2D Latency: min = 0.468445 ms, max = 0.602295 ms, mean = 0.495972 ms, median = 0.494629 ms, percentile(90%) = 0.51416 ms, percentile(95%) = 0.517578 ms, percentile(99%) = 0.540039 ms
[05/10/2024-12:25:00] [I] GPU Compute Time: min = 20.861 ms, max = 21.1443 ms, mean = 20.9222 ms, median = 20.908 ms, percentile(90%) = 21.0013 ms, percentile(95%) = 21.0591 ms, percentile(99%) = 21.1084 ms
[05/10/2024-12:25:00] [I] D2H Latency: min = 0.00415039 ms, max = 0.00634766 ms, mean = 0.00532332 ms, median = 0.00524902 ms, percentile(90%) = 0.00561523 ms, percentile(95%) = 0.00579834 ms, percentile(99%) = 0.00610352 ms
[05/10/2024-12:25:00] [I] Total Host Walltime: 3.05476 s
[05/10/2024-12:25:00] [I] Total GPU Compute Time: 3.03372 s
[05/10/2024-12:25:00] [I] Explanations of the performance metrics are printed in the verbose logs.
[05/10/2024-12:25:00] [I] 
&&&& PASSED TensorRT.trtexec [TensorRT v8502] # /usr/src/tensorrt/bin/trtexec --onnx=/home/galaxis/projects/amin/jetson-inference/data/pytorch-segmentation/fcn_resnet18.onnx --fp16 --useDLACore=0 --allowGPUFallback

```

dusty-nv, 2024-05-10:
@khsafkatamin not sure haven't tried those on DLA, it doesn't support all the layers it seems.  You could check DeepStream for other models working with DLA.

khsafkatamin, 2024-05-10:
@dusty-nv Thank you for your prompt response and suggestion. I will look into deepstream. 

But one thing that, I set allowGPUFallback=true, but it still shows the error. do you know anything about that?