dusty-nv/jetson-inference #1841: Implementing a simple pipeline with WebRTC

EmilienRNN, 2024-05-12:
Hello, 
First, I would like to say that it is an incredible work.
I am really struggling to find an example that manages to show an exemple of a gstreamer pipeline having a v4L2 camera and send it through WebRTC (using webrtcbin or whatever). I have seen the examples that you have provided like webrtc html and flask but I would like an example of a pipeline. I am sorry if it is not really the place to ask it but this is the place where I can have chances to get precise answers about my request. 
Thanks for your help.

dusty-nv, 2024-05-12:
Hi @EmilienRNN, you can do it with `video-viewer /dev/video0 webrtc://@:8554/output`

And then to check that it is working, navigate your browser to `http://JETSON_IP:8554`

EmilienRNN, 2024-05-13:
Hello, thanks to your quick answer.
Are you using gstreamer directy to do the webRTC ? Maybe I misunderstood and you didn't use this library at all

dusty-nv, 2024-05-13:
@EmilienRNN yes it is using GStreamer underneath, the code can be found at https://github.com/dusty-nv/jetson-utils/tree/master/codec

EmilienRNN, 2024-05-15:
Thanks for your answers i will try to understand how to implement it. When I'm using video-viewer, it displays the camera video on the jetson and through webRTC. Is it possible to get only the webrtc display and not the normal one ?